Koenkk/zigbee2mqtt #2006: "System Dump"

SensorsIot, 2019-09-20:
## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.6.0 (commit #e26ad2a)
CC253X firmware version: Newest from Link in your description (flashed yesterday evening)

PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
  manufCode: 4476,
  seqNum: 7,
  cmdId: 'tra... deepEqual { frameCntl:
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
  manufCode: 4476,
  seqNum: 7,
  cmdId: 'tra...
    at cb (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:59:24)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:31:17
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/functional.js:76:17)
    at Object.onceWrapper (events.js:286:20)
    at Dissolve.emit (events.js:198:13)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/deprecated/dissolve-chunks/index.js:65:29)
    at Dissolve.emit (events.js:198:13)
    at emitReadable_ (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:456:10)
    at emitReadable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:450:7)
    at readableAddChunk (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:206:11)
===== Buffer =====
{ type: 'Buffer', data: [ 5, 124, 17, 7, 7, 2, 0, 0, 0 ] }
===== Arguments =====
<Buffer 05 7c 11 07 07 02 00 00 00> 5
===== zcl-packet parse =====
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":7,"cmdId":"tradfriArrowSingle","payload":{"value":2}}
===== zcl =====
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":7,"cmdId":"tradfriArrowSingle","payload":{"value":2,"value2":0}}
===============

aidbish, 2019-09-20:
The guy with the Swiss accent

Sorry couldn't resist

Koenkk, 2019-09-20:
Thanks for the report, I've checked this and this case can be ignored.

closed 2019-09-20 by Koenkk

iotappstory, 2019-09-20:
Thanks, Koen. I will use this project in my video on Sunday morning. I love your work.

SensorsIot, 2019-09-26:
Thank you for your answer.
Just for your info: The video is out now: https://youtu.be/nAOIlTCQNnY

Koenkk, 2019-09-26:
@SensorsIot thanks for the video!

pmuntyanu, 2019-11-01:
FYI have the same:

Bug Report
What happened: I hold on/off button on the ikea tradfri remote and the zigbee2mqtt raises an exception

What did you expect to happen: No exception in the stdout

How to reproduce it (minimal and precise):
Click on the on/off button and wait few seconds.

## Debug Info
zigbee2mqtt version:  1.6.0-arm32v6
CC2530 firmware version: CC2530_CC2591_DEFAULT_20190608.zip
platform: raspberry PI

issue stdout:
```
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | PROBLEM DETECTED
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===== Stack =====
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | AssertionError [ERR_ASSERTION]: { frameCntl:
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |    { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   manufCode: 4476,
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   seqNum: 8,
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   cmdId: 'tra... deepEqual { frameCntl:
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |    { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   manufCode: 4476,
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   seqNum: 8,
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |   cmdId: 'tra...
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at cb (/app/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:59:24)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at /app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:31:17
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/functional.js:76:17)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at Object.onceWrapper (events.js:273:13)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at Dissolve.emit (events.js:182:13)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/dissolve-chunks/index.js:65:29)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at Dissolve.emit (events.js:182:13)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at emitReadable_ (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:456:10)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at emitReadable (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:450:7)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    |     at readableAddChunk (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:206:11)
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===== Buffer =====
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | { type: 'Buffer', data: [ 5, 124, 17, 8, 7, 2, 0, 0, 0 ] }
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===== Arguments =====
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | <Buffer 05 7c 11 08 07 02 00 00 00> 5
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===== zcl-packet parse =====
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | {"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":8,"cmdId":"tradfriArrowSingle","payload":{"value":2}}
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===== zcl =====
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | {"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":8,"cmdId":"tradfriArrowSingle","payload":{"value":2,"value2":0}}
ha_zigbee2mqtt.1.lttay3ew4w1d@raspberrypi    | ===============1```