Koenkk/zigbee2mqtt #2008: RTCGQ11LM keeps losing pairing

soloam, 2019-09-20:
Hello, is it just me, or any one else having troubles with the Xiaomi Body Movement Sensor (RTCGQ11LM)? I have several and all of them keep stopping and I have to repair them again (remove and add them again). I'm on zigbee2mqtt 1.5.1 (still need to update to 1.6.0) but I have this problem a long time ago.

What can I do to help spot a eventual problem?

Thank you

soloam, 2019-09-23:
Any one using this sensor with the same problem?

Thank You

soloam, 2019-09-24:
Ok, I did some research, and it seams that some people have problems with devices being kicked out of the network, some of them by OSRAM devices. i don't have OSRAM devices, but I use as repeater The Ikea Repeaters. I only have problems with the RTCGQ11LM devices, and 3 of them are like 3 meters tops from the Coordinator.

middelink, 2019-09-27:
I'm using 4 of these sensors now for about a week and haven't had to rejoin them after the initial setup.

closed 2019-12-03 by stale[bot]