Koenkk/zigbee2mqtt #2009: Hue Switches not working after upgrade from 1.5.1 -> 1.6.0

davekr8, 2019-09-20:
There seems to be double mqtt messages. Some glitch in the code ?

```
Sep 20 20:44:51 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:44:51 PM No converter available for 'action' ()
Sep 20 20:44:51 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:44:51 PM No converter available for 'battery' (100)
Sep 20 20:44:51 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:44:51 PM No converter available for 'linkquality' (36)
Sep 20 20:44:51 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:44:51 PM No converter available for 'counter' (1)
Sep 20 20:44:51 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:44:51 PM Zigbee publish to device 'bulb/ben', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 11
Sep 20 20:44:52 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:44:52 PM MQTT publish: topic 'home/zigbee/bulb/ben', payload '{"state":"OFF","linkquality":18,"brightness":255}'
Sep 20 20:44:52 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:44:52 PM Zigbee publish to device 'bulb/ben', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
Sep 20 20:44:52 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:44:52 PM MQTT publish: topic 'home/zigbee/bulb/ben', payload '{"state":"ON","linkquality":18,"brightness":255}'
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:01 PM MQTT publish: topic 'home/zigbee/hue/switch3', payload '{"counter":1,"linkquality":20,"battery":100,"brightness":255,"action":"on-press","duration":0}'
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:01 PM MQTT publish: topic 'home/zigbee/hue/switch3', payload '{"counter":1,"linkquality":20,"battery":100,"brightness":255,"action":""}'
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'duration' (0)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'action' (on-press)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'battery' (100)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'linkquality' (20)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'counter' (1)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'action' ()
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'battery' (100)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'linkquality' (20)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'counter' (1)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:01 PM Zigbee publish to device 'bulb/ben', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:01 PM MQTT publish: topic 'home/zigbee/bulb/ben', payload '{"state":"ON","linkquality":18,"brightness":255}'
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:01 PM Zigbee publish to device 'bulb/ben', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
Sep 20 20:45:02 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:02 PM MQTT publish: topic 'home/zigbee/bulb/ben', payload '{"state":"ON","linkquality":18,"brightness":255}'
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:03 PM MQTT publish: topic 'home/zigbee/hue/switch3', payload '{"counter":1,"linkquality":33,"battery":100,"brightness":255,"action":"off-press","duration":0}'
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:03 PM MQTT publish: topic 'home/zigbee/hue/switch3', payload '{"counter":1,"linkquality":33,"battery":100,"brightness":255,"action":""}'
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:03 PM No converter available for 'action' ()
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:03 PM No converter available for 'battery' (100)
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:03 PM No converter available for 'linkquality' (33)
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:03 PM No converter available for 'counter' (1)
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:03 PM Zigbee publish to device 'bulb/ben', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 11
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:03 PM MQTT publish: topic 'home/zigbee/bulb/ben', payload '{"state":"OFF","linkquality":18,"brightness":255}'
Sep 20 20:45:03 raspberrypi3 npm[20377]:   zigbee2mqtt:info 2019-9-20 8:45:03 PM Zigbee publish to device 'bulb/ben', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11

```

davekr8, 2019-09-20:
The situation changes and `down-press` seems to be converted, when I set `homeassistant: false`. Strange behaviour.

Koenkk, 2019-09-20:
Is this the same as https://github.com/Koenkk/zigbee2mqtt/issues/1583#issuecomment-496800585?

davekr8, 2019-09-20:
I don't think so. As for the `up-press` is says :

```
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'duration' (0)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'action' (on-press)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'battery' (100)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'linkquality' (20)
Sep 20 20:45:01 raspberrypi3 npm[20377]:   zigbee2mqtt:error 2019-9-20 8:45:01 PM No converter available for 'counter' (1)

```

Koenkk, 2019-09-21:
For me this device works correctly in the dev branch, can you check if it works there?

closed 2019-11-20 by Koenkk