Koenkk/zigbee2mqtt #2010: Dear Koenkk

ninhtiki, 2019-09-21:
When i add a unsupport device, zigbee2mqtt log show:

```
PM No converter available for 'RH3001' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"appVersion":72}}'
  zigbee2mqtt:warn 9/20/2019, 8:03:46 PM No converter available for 'RH3001' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":27,"batteryPercentageRemaining":120}}'
  zigbee2mqtt:info 9/20/2019, 8:03:49 PM Device incoming...
  zigbee2mqtt:info 9/20/2019, 8:03:49 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 9/20/2019, 8:03:52 PM No converter available for 'RH3001' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"appVersion":72}}'
  zigbee2mqtt:warn 9/20/2019, 8:03:52 PM No converter available for 'RH3001' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryVoltage":27,"batteryPercentageRemaining":120}}'
  zigbee2mqtt:warn 9/20/2019, 8:03:52 PM No converter available for 'RH3001' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"appVersion":72}}'

```
How can I handle this problem, Thanks you.

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

kloknibor, 2019-09-21:
See : https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2019-11-27 by stale[bot]

takatukaland, 2020-02-02:
add to devices.js:
	{
        zigbeeModel: ['RH3001'],
        model: 'RH3001',
        vendor: 'Samotech',
        description: 'Magnetic door & window contact sensor',
        supports: 'contact',
        fromZigbee: [fz.ias_contact_status_change],
        toZigbee: [],
    },