Koenkk/zigbee2mqtt #2011: IKEA TRADFRI LED bulb does not get published

digitalyours, 2019-09-21:
# Bug Report
## What happened
I've paired the IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9) successfully. 

`
Sep 21 08:31:15 fhem npm[1346]:   zigbee2mqtt:info 2019-9-21 8:31:15 AM Successfully interviewed '0x000d6ffffe33cf33', device has succesfully been paired
Sep 21 08:31:15 fhem npm[1346]:   zigbee2mqtt:info 2019-9-21 8:31:15 AM Device '0x000d6ffffe33cf33' is supported, identified as: IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)
Sep 21 08:31:15 fhem npm[1346]:   zigbee2mqtt:info 2019-9-21 8:31:15 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6ffffe33cf33","model":"LED1624G9","vendor":"IKEA","description":"TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white","supported":true}}'
`

But it does not get published as a topic on my MQTT Server.

`019-9-21 8:34:59 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-9-21 8:34:59 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"e3d0b97","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
`

## What did you expect to happen
Device gets published as a topic like all other devices. 

## How to reproduce it (minimal and precise)
1. Starting with a clean configuration.yaml / database.db
2. Paring light bulb 
3. See output above

Log file after restart of zigbee2mqtt
`
2019-9-21 8:34:58 AM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-21.08-34-58'
2019-9-21 8:34:58 AM - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-09-21.00-07-51'
2019-9-21 8:34:58 AM - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-9-21 8:34:58 AM - info: Starting zigbee2mqtt version 1.6.0 (commit #e3d0b97)
2019-9-21 8:34:58 AM - info: Starting zigbee-herdsman...
2019-9-21 8:34:58 AM - debug: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extenedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"}$
2019-9-21 8:34:59 AM - info: zigbee-herdsman started
2019-9-21 8:34:59 AM - info: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
2019-9-21 8:34:59 AM - debug: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
2019-9-21 8:34:59 AM - info: Currently 1 devices are joined:
2019-9-21 8:34:59 AM - info: 0x000d6ffffe33cf33 (0x000d6ffffe33cf33): LED1624G9 - IKEA TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (Router)
2019-9-21 8:34:59 AM - warn: `permit_join` set to  `true` in configuration.yaml.
2019-9-21 8:34:59 AM - warn: Allowing new devices to join.
2019-9-21 8:34:59 AM - warn: Set `permit_join` to `false` once you joined all devices.
2019-9-21 8:34:59 AM - info: Zigbee: allowing new devices to join.
2019-9-21 8:34:59 AM - info: Connecting to MQTT server at undefined
2019-9-21 8:34:59 AM - debug: Using MQTT client ID: 'zigbee_pi'
2019-9-21 8:34:59 AM - info: Connected to MQTT server
019-9-21 8:34:59 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-9-21 8:34:59 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"e3d0b97","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
`

## Debug Info
zigbee2mqtt version: latest DEV branch as of Sep 20. (commit #e3d0b97)
CC253X firmware version: 20190608

didiht, 2019-09-21:
What is the result of publishing  **zigbee2mqtt/bridge/config/devices**?

Excerpt of https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

zigbee2mqtt/bridge/log
zigbee2mqtt will output log to this endpoint. Message are always in the form of {"type":"TYPE","message":"MESSAGE"}. Possible message types are:

...
"devices": a list of all devices, this message can be triggered by sending a message to zigbee2mqtt/bridge/config/devices (payload doesn’t matter).

digitalyours, 2019-09-21:
I‘m getting this output:
`log {"type":"devices","message":[{"ieeeAddr":"0x000d6ffffe33cf33","type":"Router","model":"LED1624G9","friendly_name":"0x000d6ffffe33cf33","networkAddress":2137,"manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E14 CWS opal 600lm","hardwareVersion":1,"softwareBuildID":"1.3.002","dateCode":"20170315","lastSeen":1569049203074},{"ieeeAddr":"0x7cb03eaa0a00cf26","type":"Router","model":"AB3257001NJ","friendly_name":"0x7cb03eaa0a00cf26","networkAddress":44152,"manufacturerID":48042,"manufacturerName":"OSRAM","powerSource":"Mains (single phase)","modelID":"Plug 01","hardwareVersion":1,"softwareBuildID":"V1.04.12","dateCode":"20140331DEOS****","lastSeen":1569049383859},{"ieeeAddr":"0x00124b0018e20395","type":"Coordinator"}]}`

didiht, 2019-09-21:
I see one IKEA bulb, one OSRAM device and the coordinator.

Are there really no messages with topic **zigbee2mqtt/0x000d6ffffe33cf33** (IKEA) after publishing e.g. **zigbee2mqtt/0x000d6ffffe33cf33/set** with payload **ON**?

For the OSRAM device Z2M is working as expected?

digitalyours, 2019-09-21:
I don‘t see any messages for 0x000d6ffffe33cf33 and I’m running out of ideas on what to do. (Obviously I also deleted everything, rebooted, etc.)
Every other device is working as expected. Adding it to the network, and it gets immediately announced as a topic.

Innanazmova83, 2019-09-21:
Хз

сб, 21 сент. 2019 г., 16:07 Christoph Rechsteiner <notifications@github.com

didiht, 2019-09-21:
I assume you have other IKEA bulbs which are working fine?

Maybe it's a problem related to this bulb / firmware.

digitalyours, 2019-09-21:
I do have other ikea stuff which runs fine. Maybe it is just this bulb.
One thing I haven’t mentioned, I have upgraded from 1.5.1 where this bulb-type was not supported. There is no hidden cache or anything I could have overlooked? Just making sure...

digitalyours, 2019-09-21:
> Хз сб, 21 сент. 2019 г., 16:07 Christoph Rechsteiner <notifications@github.com
> […](#)

?

didiht, 2019-09-21:
I don't think so, but I am not an expert in this. Maybe Koen can help.

Koenkk, 2019-09-22:
> Device gets published as a topic like all other devices.

What do you expect to be published? Can you provide examples from other devices?

digitalyours, 2019-09-23:
Thx Koenkk for jumping in.
> > Device gets published as a topic like all other devices.
> 
> What do you expect to be published? Can you provide examples from other devices?

Expected behavior would be an announcement of the device as a topic like
zigbee2mqtt/0x000d6ffffe33cf33

All the other devices are immediately available as a topic e.g.
ZIGBEE2MQTT/0x7cb03eaa0a00cf26

Only this light bulb does not get published. I will try to get a second one to figure out if it’s the bulb or the software.

Koenkk, 2019-09-23:
You will only see publishes to `zigbee2mqtt/0x000d6ffffe33cf33` when the bulb changes it state.

You can trigger it by e.g. sending to `zigbee2mqtt/0x000d6ffffe33cf33/set` payload: `OFF`.

closed 2019-11-22 by Koenkk