Koenkk/zigbee2mqtt #2012: zigbee2mqtt.io breaks DNSSEC

electrofloat, 2019-09-21:
Since there's no issue tracking on zigbee2mqtt.io repo I had to create this one here.
I noticed that the website is regularly unavailable, and it seemed random.  Investigation showed that the issue is with the CNAME record at apex.
Such zones are **illegal** according to the RFCs:
 - https://tools.ietf.org/html/rfc1034#section-3.6.2
 - https://tools.ietf.org/html/rfc2181#section-10.1

http://dnsviz.net/d/zigbee2mqtt.io/dnssec/

Can you please get in contact with your DNS provider and make them fix this issue asap?

Thank you!

Koenkk, 2019-09-22:
The current configuration is:

[image]

I've changed it to

[image]

Does that fix this issue?

electrofloat, 2019-09-23:
Yes it does. Although it wasn't your fault in the first place, it is a good fix for the mentioned issue.

Dynadot shouldn't let their customers configure an illegal DNS record at all. Even if you fixed it for yours, any other domains they own could have this exact issue.
So I'm still encouraging you to let them know about this, and hopefully they'll implement some sort of validation or not allow this kind of configuration at all for any of their domains.

Thank you!

Koenkk, 2019-09-23:
I've send them a mail, thanks!