Koenkk/zigbee2mqtt #2013: Cannot pair ikea LED1537R6 GU10 light Cannot get the Node Descriptor of the Device:

kloknibor, 2019-09-21:
# Bug Report
## What happened
I'm trying to get an [ ikea LED1537R6 GU10 light( https://www.zigbee2mqtt.io/devices/LED1537R6.html) but I'm already trying to pair it for a full day. It used to work in the past but I switched to an intel NUC this weekend.

All endpoint devices connect without a problem (see debug info), but these 2 GU10 lights won't pair. Things I tried:

- restart zigbee2mqtt 
- restart system
- reinstall zigbee2mqtt
- press reset button stick
- reflash CC2531 stick
- use a new CC2531 stick
- bring distance between light and coordinator to approx 1-2 meter. 
- Remove devices.db
- run zigbee2mqtt in a docker
- tried to repair 200 times

and tried all these steps in different orders. Still no luck, at one point 1 of the lights did pair, but after reflashing the stick and reinstalling zigbee2mqtt I can't get that one paired anymore as well...

## What did you expect to happen
I would expect to pair as expected like it did in the past.

## How to reproduce it (minimal and precise)
Try to pair [ ikea LED1537R6 GU10 light( https://www.zigbee2mqtt.io/devices/LED1537R6.html) according to instructions, try it with a completely fresh installation and fresh CC2531 stick with a distance of 1-2 meter. I don't think the environment could be better for getting this to pair.

## Debug Info
zigbee2mqtt version:1.6.0
CC253X firmware version:20190608

  zigbee2mqtt:info 9/21/2019, 11:06:07 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"unknown","coordinator":20190608,"log_level":"info","permit_join":true}'
  zigbee2mqtt:error 9/21/2019, 11:06:54 AM Cannot get the Node Descriptor of the Device: 0x000b57fffe4b8bb3 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 9/21/2019, 11:07:14 AM Cannot get the Node Descriptor of the Device: 0x000b57fffe4b8bb3 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 9/21/2019, 11:07:34 AM Cannot get the Node Descriptor of the Device: 0x000b57fffe4b8bb3 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 9/21/2019, 11:07:54 AM Cannot get the Node Descriptor of the Device: 0x000b57fffe4b8bb3 (Error: Timed out after 10000 ms)
  zigbee2mqtt:info 9/21/2019, 11:08:47 AM Successfully reenabled joining
  zigbee2mqtt:info 9/21/2019, 11:09:43 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b00193b093f","type":"Coordinator"},{"ieeeAddr":"0x00158d00029b0ee0","type":"EndDevice","model":"MCCGQ01LM","friendly_name":"0x00158d00029b0ee0","nwkAddr":20256,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","hwVersion":2,"swBuildId":"3000-0001","dateCode":"20150424"},{"ieeeAddr":"0x00158d00028de7c1","type":"EndDevice","model":"MCCGQ01LM","friendly_name":"0x00158d00028de7c1","nwkAddr":20215,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","hwVersion":2,"swBuildId":"3000-0001","dateCode":"20150424"},{"ieeeAddr":"0x00158d0002789837","type":"EndDevice","model":"WSDCGQ11LM","friendly_name":"0x00158d0002789837","nwkAddr":2111,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","hwVersion":30,"swBuildId":"3000-0001","dateCode":"20161129"},{"ieeeAddr":"0x00158d0001e14480","type":"EndDevice","model":"MCCGQ01LM","friendly_name":"0x00158d0001e14480","nwkAddr":63865,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","hwVersion":2,"swBuildId":"3000-0001","dateCode":"20150424"}]}'

Koenkk, 2019-09-21:
Pairing should be improved in the latest dev, can you check if it works there?

dirstel, 2019-09-22:
@koenkk: whatever you did, it helped a lot for (re-)pairing my aqara motion sensors. You nailed it :-)

Koenkk, 2019-09-22:
Great!

kloknibor, 2019-09-24:
I solved it by the following steps : 
- turn of tradfri gate way 
- turn of all other ZigBee devices on other network
- Use a light switch to power the lights on and off instead of screwing it in and out.

The last steps really seems needed to improve the ikea tradfri chance of pairing quite a bit... Otherwise the resetting for some lights can be hard!

Anyhow it works so thanks :)!

closed 2019-09-24 by kloknibor