Koenkk/zigbee2mqtt #2014: Adding Osram Smart+ plug model: AC10691

alwashe, 2019-09-21:
I tried to follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

After adding to ```devices.js``` this:
```
{
    zigbeeModel: ['Plug Z3'],
    model: 'AC10691',
    vendor: 'Osram',
    description: 'Smart+ plug',
    supports: 'on/off',
    fromZigbee: [],
    toZigbee: [],
},
```

Triggering the plug does nothing.

I got this after repairing:
```
Sep 21 14:26:04 zigbee npm[19125]:   zigbee2mqtt:warn 2019-9-21 2:26:04 PM No converter available for 'AC10691' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"lastMessageRssi":-66}}'
Sep 21 14:26:04 zigbee npm[19125]:   zigbee2mqtt:warn 2019-9-21 2:26:04 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

so I added this and named it ```osram_diagnostic_change```:
```
	osram_diagnostic_change: {
		cid: 'haDiagnostic',
		type: 'devChange',
		convert: (model, msg, publish, options) => {
			console.log(msg.data)
		},
	},
```
I am stuck at:
> In case your device is not reporting anything, it could be that this device requires additional configuration. This can be done by adding a configure: section (examples here).

I tried to add the configure section from the Osram smart+ plug model: AB3257001NJ
```
	{
    zigbeeModel: ['Plug Z3'],
    model: 'AC10691',
    vendor: 'Osram',
    description: 'Smart+ plug',
    supports: 'on/off',
    fromZigbee: [],
    toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 3);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
	},
```

I can return the plug easily but wanted to try to contribute.

erikomv, 2019-09-21:
I am looking at solutions as well for this.
Did you try: https://github.com/Koenkk/zigbee-shepherd-converters/issues/572

alwashe, 2019-09-21:
Thank you very much.

Had to add:
```
    'AC10691': [configurations.switch],
```
to: ```zigbee2mqtt/lib/extension/homeassistant.js```

closed 2019-09-21 by alwashe

Koenkk, 2019-09-21:
Can you make a pr to zigbee-shepherd-converters? In this way it can be supported out of the box.

alwashe, 2019-09-21:
yes