Koenkk/zigbee2mqtt #2016: Error while starting zigbee2mqtt 1.6.0

frettijule, 2019-09-21:
# Bug Report
## What happened
After fresh install zigbee2mqtt doesn't start, got error
## What did you expect to happen

## How to reproduce it (minimal and precise)
npm start reproduce the same error after fresh install of zigbee2mqtt or on rm -r node_modules and npm install
## Debug Info
zigbee2mqtt version: 1.6.0
CC2531 firmware version: C2531_DEFAULT_20190608

After starting zigbee2mqtt with npm start I got the error:
zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-9-22 12:34:58 AM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-22.00-34-57'
  zigbee2mqtt:info 2019-9-22 12:34:58 AM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 2019-9-22 12:34:58 AM Starting zigbee-shepherd
  zigbee2mqtt:info 2019-9-22 12:35:00 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: Cluster with key '64527' does not exist)
  zigbee2mqtt:info 2019-9-22 12:36:00 AM Starting zigbee-shepherd
  zigbee2mqtt:error 2019-9-22 12:36:01 AM Error while starting zigbee-shepherd! (Error: dev exists, unregister it first.)
  zigbee2mqtt:error 2019-9-22 12:36:01 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-9-22 12:36:01 AM Failed to start
        {"message":"dev exists, unregister it first.","stack":"Error: dev exists, unregister it first.\n    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/shepherd.js:449:19\n    at Promise.apply (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/q/q.js:1185:26)\n    at Promise.promise.promiseDispatch (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/q/q.js:808:41)\n    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/q/q.js:1411:14\n    at runSingle (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
  zigbee2mqtt:error 2019-9-22 12:36:01 AM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/thomas/.npm/_logs/2019-09-21T22_36_01_804Z-debug.log

ssirkakriss, 2019-09-22:
I had a similar error "Error: Cluster with key"

Then i stumbled upon this: https://github.com/Koenkk/zigbee2mqtt/issues/1999#issuecomment-533306062

It's working now with the dev branch.

Hope it's of help.

frettijule, 2019-09-22:
I find out that the problem isn't zigbee2mqtt version 1.6.0, because i copy my configuration-file and the database from an Raspi with version 1.5.1. There was also an other problem.
If I start now with a clean configuration-file and database.db it's work fine

closed 2019-09-22 by frettijule