Koenkk/zigbee2mqtt #2019: Installation NPM

Luigi-01, 2019-09-22:
Hi,

I have a small server running with Domoticz, Pihole, Nextcloud and Unifi.
This is running now for some time.
Domoticz has a Z-wave stick, P1 meter connection and a Hue bridge connection.
I followed the instruction regarding the installation and run in some problems.
The problems regarding the permissions I have been able to solve. 
The remaining issues I have not been able to solve. (I am not a IT person)
Starting NPM gives the  follow issues:
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional log

Is there some one who can help me.

Koenkk, 2019-09-24:
Do you get any more logging?

closed 2019-11-23 by Koenkk

ru320, 2020-02-02:
Hello Koenkk and all the Rest of the Community.

Ich had the Same Problem.

The reason and the solution is pi-hole.

Pi-hole will Block die DNS for eai_again Server during the Install.
And at the start the Message above will occur.
npm ERR! code ELIFECYCLE .....

Disable pi-hole while the Installation process and it will work fine.

Greetings

Rü