Koenkk/zigbee2mqtt #2020: Feature Request: More properties for Coordinator

dirstel, 2019-09-22:
latest dev-Version
"home/bridge/config/devices" retrives an array of all devices. In my case I'd like to sort devices by friendly_name as I'm using a structured pattern. Sorting can be archived easily by 
`msg.payload.message.sort((a, b) => a.friendly_name.localeCompare(b.friendly_name));
`
But Coordinator has very limited subset of properties (especially lacks friendly_name)
```
ieeeAddr: "0x00..."
type: "Coordinator"
```
Other devices have way more properties
```
ieeeAddr: "0x000..."
type: "Router"
model: "..."
friendly_name: "1_eg/vorgarten/licht/saison"
networkAddress: ###
manufacturerID: ###
manufacturerName: "OSRAM"
powerSource: "Mains (single phase)"
modelID: "Plug 01"
hardwareVersion: 1
softwareBuildID: "V1.04.12"
dateCode: "20140331DEOS****"
lastSeen: null
```
Some more properties would make sense and will make sorting easier:
friendly_name (even if value is "")
softwareBuildID: (as printed in startup)
lastSeen: (even if this returns current timestamp)
...

Koenkk, 2019-09-23:
Added: `networkAddress`, `friendly_name` (always `Coordinator`), `softwareBuildID`, `dateCode`, `lastSeen` (always now). 

Feature is available in the dev brach.

closed 2019-09-23 by Koenkk

dirstel, 2019-09-29:
@Koenkk: though it's somewhat late: modifications are working, returning reasonable values. Thanks a lot.