Koenkk/zigbee2mqtt #2024: Impossible to flash CC2531 (with any method)

AdrienAA, 2019-09-23:
Hello everyone,

I'm new to Zigbee2mqtt and I have **a lot of trouble to flash the stick**.

Here is what I did, and the results :

I bought a CC2531 USB Dongle, a CCDebugger and the wires to connect them.

I installed all the needed softwares and firmwares, then, when I connected everything I had only a RED light. **I changed the orientation of the cables**, checked continuity in everything with a multimeter, but nothing changed.
Only ONCE I had a green light ! So I tried to flash the CC2531, but the software failed (it managed to erase the previous firmware, but not to write the new one).

Then I tried on **another computer**, but I had the same problems : Light won't go GREEN

So I tried all the **alternative methods** (Arduino and Raspberry PI) but it never worked : I had messages like "No chip detected" or "Chip ID = 0ffff". So I started to think that my USB Dongle was broken.

I bought a **new CC2531 USB Dongle** (from another company), but I have exactly the same problems !

Now I'm lost and I just don't understand where the problem could come from...

**What I am sure of :** 

CC2531 are not the problem
Wires are not the problem
Computers are not the problem
CCDebugger is not the problem

Can anybody help me to find a solution please ? I'm getting crazy with this because It doesn't seem logical !

Thank you in advance for you help ! I hope I will soon be able to join the comunity of zigbee2mqtt !

sjorge, 2019-09-23:
Is your stick plugged in when you try to flash? IIRC I had some issues with that when I tried to flash mine... but I forgot if it was because it was plugged in... or not plugged in.

AdrienAA, 2019-09-23:
When I flash I always have the stick plugged, except on Arduino (I tried both technics)

AdrienAA, 2019-09-24:
I will try to change the cables (I changed the USB cable of the CCDebugger but nothing changed and I ordered a new cable to like the USB stick to the debugger.)
Also I tried a 1000 different ways to plug the stick and the debugger in my computer but nothing changes.

I'm getty crazy with this ! ^^ I hope the new cables are the key...

bochelork, 2019-09-24:
Make sure to plug in your CC2531 first into USB. Finally plug in your CC-Debugger into USB.

fredrikgk, 2019-09-24:
@AdrienAA I have seen a couple of similar issues on the TI E2E forum with USB dongles being purchased from Aliexpress. Can you share photos of the dongles you are using? Are any of the dongles you are using purchased from TI?

AdrienAA, 2019-09-25:
I bought my usb dongles from amazon. here are the references of the 2 dongles, the CCDebugger and the cable (none are from TI) : 

https://www.amazon.fr/gp/product/B07PBKVGGP/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

https://www.amazon.fr/gp/product/B07MBZMJB4/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

https://www.amazon.fr/gp/product/B07FP5XHPM/ref=ppx_yo_dt_b_asin_title_o06_s01?ie=UTF8&psc=1

https://www.amazon.fr/gp/product/B07L2SVJH8/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

fredrikgk, 2019-10-01:
I have supported @imdos for the same problem here: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/839998

The dongles you link to @AdrienAA has similar markings, they are made with a very old PCB design revision. I cannot make out the print on the IC, but potentially these can be grey market devices, or even counterfeit. 

I would recommend you to return the units to the seller and purchase new dongles from a more professional distributor.

AdrienAA, 2019-10-02:
Hello !
I bought a new cable to connect the dongle to the CCdebugger some times ago. It just arrived and everything worked ! So apparently the problem was only due to the cable.
Thanks a lot for your help ! I'm still trying to figure out the next steps, but finally I can go forward :)

closed 2019-10-02 by AdrienAA