Koenkk/zigbee2mqtt #2026: New Ikea symfonisk Remote Controller

plbartek89, 2019-09-24:
Hello , new ikea Symfonisk Remote Controller is out. 
[PIC](https://images.ifun.de/wp-content/uploads/2019/09/sonos-schalter.jpg)
It supports : 
1xClick , 2xclick , 3xClick and Rotate.
Any plans to implement this one ?
I have two of them, if I can help somehow just report.

Koenkk, 2019-09-24:
Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html , if yiu need help lets us know.

kolbigh, 2019-10-07:
Hello,
i have one of these devices too.

The output from zigbee2mqtt is:
>  zigbee2mqtt:info 10/5/2019, 2:25:57 PM New device 'SYMFONISK Sound Controller' with address 0x14b457fffe663a12 connected!
>  zigbee2mqtt:info 10/5/2019, 2:25:57 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x14b457fffe663a12","meta":{"modelID":"SYMFONISK Sound Controller"}}'
>  zigbee2mqtt:warn 10/5/2019, 2:25:57 PM Device with modelID 'SYMFONISK Sound Controller' is not supported.
>  zigbee2mqtt:warn 10/5/2019, 2:25:57 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
>  zigbee2mqtt:warn 10/5/2019, 2:25:59 PM Device with modelID 'SYMFONISK Sound Controller' is not supported.
>  zigbee2mqtt:warn 10/5/2019, 2:25:59 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Following https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html and inserting a new section into `node_modules/zigbee-shepherd-converters/devices.js` like:
>{
>        zigbeeModel: ['SYMFONISK Sound Controller'],
>        model: 'E1744',
>        description: 'SYMFONISK Sound Controller',
>        supports: 'lets have a look',
>        vendor: 'IKEA',
>        fromZigbee: [],
>        toZigbee: [],
>    },

restarting zigbee2mqtt and trying to push the device does not show any output.
I also tried to append the `configure:` part (like it is listed in the web page, for an IAS device), but that  does not change anything.

Sorry for that amateurish aproach, but i have not much experience in that domain, so i would say i need some help here...

Koenkk, 2019-10-08:
Closing this in favor of https://github.com/Koenkk/zigbee-shepherd-converters/issues/658

closed 2019-10-08 by Koenkk

muuuh, 2019-11-06:
I'd like to use it with Homeassistant. Could you please give me a hint?