Koenkk/zigbee2mqtt #2028: question: Friendly Name as Paths (cant find docs)

rm2kdev, 2019-09-24:
This is just a quick question i couldn't find it on the docs (did read, extensively)

Does friendly_name support paths i.e
can i call my hue lights

friendly_name: Hue/Lights/Kitchen/Roof
friendly_name: Hue/Lights/Kitchen/Lamp
friendly_name: Hue/Lights/Bedroom/Roof
friendly_name: Hue/Lights/Hallway/Roof

and device like

friendly_name: Hue/Dimmers/Bedroom
friendly_name: Hue/Dimmers/Kitchen
friendly_name: Hue/Dimmers/Lounge
friendly_name: Sensors/Temp/Kitchen

carldebilly, 2019-09-24:
It's not supported yet by Zigbee2MqttAssistant.. https://github.com/yllibed/Zigbee2MqttAssistant/issues/26

But I think I'll fix it soon if it's officially supported by Zigbee2Mqtt.

Koenkk, 2019-09-24:
`/` within the friendly name is allowed and tested.

closed 2019-09-24 by Koenkk