Koenkk/zigbee2mqtt #2029: Remove Device not working

3DIRK, 2019-09-24:
# Bug Report
## What happened
IKEA Tradfri Light is not reachable anymore, so I tried to remove the light from the network. Removing throws an error. (And "force remove" did not start either.)
## What did you expect to happen
Device shall be remove
## How to reproduce it (minimal and precise)
Issue command to remove device

Logfile: https://pastebin.com/t3hHEUPx
## Debug Info
zigbee2mqtt version: latest-dev
CC253X firmware version: latest non routing

sjorge, 2019-09-24:
I noticed that force remove doesn't happen.
Remove, device doesn't respond, nothing happens

Koenkk, 2019-09-25:
'Force remove' has been removed as it was giving false expectations.

Note that the coordinator/zigbee2mqtt cannot remove a device, it can only request a device to remove itself from the network (this is how removing in zigbee works). If the device doesn't remove itself; it just cannot be removed from the network.

3DIRK, 2019-09-26:
Thanks for the reply, @Koenkk 

There is no need anymore to remove the device from database in case of an unsucessful removal? Don't we get a bunch of stale devices in the database?

Koenkk, 2019-09-26:
In case of an unsuccessful removal the device is not removed from the network thus it shouldnt be removed from the db.

Note that somes devices also remove itself from the network when you factory reset them. In this case they will send an event and zigbee-herdsman will remove them from the database.

sjorge, 2019-09-26:
There do seem to be some edge cases, say the DB is corrupt and you need
to repair everything... 

Some device then automatically show up in the db in a incomplete state,
resulting in an incomplete DB. They device then seem unable to
succesfully be repaired because they are in the db but not really. So
the only way to solve that is to change the id and channel of hte
netwerk. 

(That is how I fixed my corrupt db, which was a pain. Force remove +
repair used to work more or less, I first tried blacklisting but then
they obviously  cant be repaired at all.)

closed 2019-09-27 by 3DIRK

Koenkk, 2019-09-27:
@sjorge I think with zigbee-herdsman (zigbee2mqtt dev branch) the database won't get corrupt anymore (the whole database thing has been simplified and we are not using nedb anymore).

In case you want start from scratch, just change the pan_id and zigbee-herdsman will automatically cleanup the db.

lancelon, 2020-12-27:
Quick question - am having various problems pairing new devices. I reflashed my CC2531 and suddenly an IKEA blind which I'd unsuccessfully tried to pair about 8 months earlier appeared. Never seen it before in Zigbee2MQTT. 
It says the interview is not complete, though, and the device has an exclamation mark and does not work. It's also showing as unsupported.
I can't remove it nor do anything with it really. I'm guessing this device still counts as part of my limit of 15 direct children?