Koenkk/zigbee2mqtt #2030: Osram AC03647 wrong colors

jasperro, 2019-09-24:
I have been having some strange color issues with my OSRAM AC03647 bulb. The colors are set wrongly by mqtt command and home assistant.
`mosquitto_pub -h 192.168.1.16 -t "zigbee2mqtt/jasperpl/set" -m "{"color":{"r":255,"g":255,"b":255},"color_mode":1}"`
When r: 255, g: 255 and b: 255, the color is set to white as expected.
When the color is set to (255,0,0) with max brightness the expected color would be red, but instead the color is set to magenta. When the brightness is set to 50%, it does become red.
When the color is set to (255,255,0) yellow is expected but the color is set to lime. (255,127,0) however does produce yellow.
When the color is set to (0,0,255), pure blue isn't the color the light is set to. The light gets set to blue when you use (0,0,0)
Green works perfectly.
It is not immediately obvious to me why this happens, does anyone have an idea? Maybe an issue with the RGB to xy conversion?.

zigbee2mqtt version: 1.6.0 (dev)
CC253X firmware version: Z-Stack_Home_1.2

closed 2019-11-30 by stale[bot]

grostim, 2019-12-10:
I have the same issue.
Color are "shifted".

Also, i do not get any "feedback" mqtt message when i apply a setting or change the bulb state. Therefore, Hass is not working properly with these bulbs.

mani9876, 2020-01-09:
I have exactly the same issue with my new OSRAM AC03647 bulb.
Another issue is, that the transition time only applies if I e.g. set the brightness from 100 to 0, if I set the brightness vom 0 to 100, it immediately sets the brightness to 100%, regardless of the transition time value.
Is it possible to reopen this issue?

jasperro, 2020-01-10:
The colors have been improved in the dev branch. See issue https://github.com/Koenkk/zigbee-herdsman-converters/issues/575. How to switch to dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html .

As for the transition issue, "osram_set_transition": 0.1 in your device config should do the trick.