Koenkk/zigbee2mqtt #2032: Ikea tradfri bulbs/spots drop from network

neographikal, 2019-09-24:
# Bug Report
## What happened
After a while I can't turn on Ikea Tradfri lamps (GU10 spots and several bulbs). They seem to be offline from zigbee2mqtt's perpective. Powercycling the bulb fixes the problem temporarily. CC2531 stick is close by all the bulbs and on USB extension cable, Xiaomi sensors that are much farther away can reach the stick just fine.

https://pastebin.com/cKRKMrUj

## What did you expect to happen
Lamps keep working

## How to reproduce it (minimal and precise)
Wait a few hours

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

oxelamp, 2019-09-25:
Can you try coordinator version 20190619?
Worked wonders for my mainly tradfri setup.

neographikal, 2019-09-25:
You mean the source routing firmware?

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

neographikal, 2019-09-26:
One night later: same problems. Network is gone, Xiaomi stuff still works.

https://pastebin.com/znTBbxzC

Koenkk, 2019-09-26:
> Xiaomi stuff still works.

Can you still control your non TRADFRI devices?

neographikal, 2019-09-26:
The Xioami stuff are only sensors (door contact sensors, buttons, motion detection, temp sensors). Those seem to work. The source routing firmware did force some of them to go via a Tradfri bulb initially.

Networkmap now: https://pastebin.com/Zd5HLvsr

It does see the Tradfri bulbs as online but can't control them and the mesh network is gone. Yesterday evening before I went to sleep it was looking much better.

Koenkk, 2019-09-27:
Do you see anything in the zigbee2mqtt logging when power cycling the bulbs?

neographikal, 2019-09-28:
https://pastebin.com/96whPKk6

Pluggin it in after the last

`Sep 28 11:05:51 neo-server zigbee2mqtt[412]: #033[33m  zigbee2mqtt:warn#033[39m 9/28/2019, 11:05:51 AM Failed network lqi scan for device: '0x086bd7fffe320785' with error: 'Error: Timed out after 10000 ms'`

Those ID's in the lqi scan lines correspond with the Ikea Tradfri bulbs. Did it a second time with two GU10 spots that were lost in limbo:

https://pastebin.com/1XPJfhYH

Same result.

Koenkk, 2019-09-28:
I looks that they are rejoining again; somehow they are kicked of the network. Do you have any other zigbee hubs running (e.g. TRADFRI gateway)

To dive deeper into this issue, could you sniff the network, start at a point where you can still turn the bulbs on/off until the point where you can't (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html).

neographikal, 2019-09-28:
Hmm I do have an aqara gateway still on, but these networks should be able to Co exist right? My neighbors could also have Philips Hue or something else. Will try to sniff later this week.

neographikal, 2019-09-29:
Strangely enough unplugging the Aqara gateway did help (at least for now, will monitor the network). But I do not understand why, multiple networks should be able to co-exist right?

neographikal, 2019-09-30:
Aqara gateway is not in use, but the lamps dropped of the zigbee network again. I'll try to do some sniffing.

anzecesar, 2019-10-01:
Hi, I'm having the same issue.
First I had the default firmware, some lights further away from controller were having networking issues. Similar as described here. 
 
I flashed with source_routing. It seems better initially, but overtime some lights become unresponsive. I have a lot of Tradfri gu10 spots with some e27s. All Tradfri, ~30 lights total. The misbehaving lights are always further away from the controller.

I have no other zigbee hubs online.

I have xiaomi switches and 4 temperature sensors. Other than pairing difficulties I have no issues with the switches.

I'm also seeing this often: 

```
zigbee2mqtt:info 10/1/2019, 7:23:12 PM Zigbee publish to device '0x086bd7fffe06b9c8', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 1
  zigbee2mqtt:error 10/1/2019, 7:23:18 PM Zigbee publish to device '0x086bd7fffe06b9c8', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
```

Sending the signal multiple times eventually works.

I'll try to get a hold of another usb stick to capture some traffic

neographikal, 2019-10-02:
Distance does not seem to affect it here. It is definitely better with this firmware but still not entirely stable. Will try to capture traffic this weekend.

neographikal, 2019-10-02:
[zigbee2mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/3682945/zigbee2mqtt.txt)

Lamps dropped of the network today again. Didn't have the sniffing stick ready, I did spot a crash of zigbee2mqtt however. See attached logfile @ time 18:50. It crashes and restarts itself. Don't know if this related 1:1 to my tradfri problem but it is not expected behaviour.

Koenkk, 2019-10-03:
@neographikal It seems that the stick crashes because all of the network scans. 

Could you disable the automatic network scan? (maybe this also solves this problem?)

neographikal, 2019-10-03:
will try, thanks. Where can I find this option?

https://www.zigbee2mqtt.io/configuration/configuration.html

Doesn't mention it it seems.

Koenkk, 2019-10-03:
It's not done by zigbee2mqtt itself but done by some external application, do you e.g. have the zigbee2mqtt node red addon or home assistant zigbee2mqtt network map addon installed?

neographikal, 2019-10-03:
Okay, thanks. I have zigbee2mqttAssistant running

root@neo-server:/home/neo# docker stop zigbee2mqttassistant

Let's see what that does :)

bochelork, 2019-10-03:
Can you sniff your network? I just noticed that network address changes are not handled well. This is what I did to get my Osram device back to my network.
- Sniff you network
- Switch off the tradfri and switch it back on
- See if you get a device announcement in your wireshark data
- Manually change your decimal network address in database.db
- Restart zigbee2mqtt.

Koenkk, 2019-10-03:
Will try to implement a fix for this tomorrow (https://github.com/Koenkk/zigbee2mqtt/issues/2017#issuecomment-538142703)

manup, 2019-10-04:
The problem might be in the light firmware itself. We have various reports for IKEA GU10 and E27 dimmable and white spectrum bulbs dropping off. They become only responsive again after a power-cycle. This happens even with the IKEA gateway itself https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-536604489

Firmware versions at least up to 1.2.214 seems to be affected. No reports yet for newer firmware 2.3.007 (white spectrum) lights and new hardware revisions.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

Koenkk, 2019-10-04:
@manup https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-523470291 could it be that the lights change the networkAddress without announcing the new one? Otherwise it would be strange that group commands still work? A re-power would trigger a device announce resulting in the gateway receiving the new network address.

manup, 2019-10-04:
It varies, a changed nwk address is handled in our controller on all incoming commands. We also send out NWK address request broadcasts for unresponsive devices which would detect the change.

In my setup it happened only twice that a E27 dimmable lights got stuck, after it was running for months without issues. I fired up the sniffer to see if the light is doing at least something, but there was total silence, not even the regular NWK link status commands every 15 seconds. So it seems in this case the firmware was stuck or crashed all together. I think I've tried even Touchlink but it didn't respond.

I haven't found a way to reproduce the error yet, perhaps bringing it in a large network with multiple hops and many broadcasts could trigger the fault.

Koenkk, 2019-10-04:
Just released a fix for https://github.com/Koenkk/zigbee2mqtt/issues/2017 in the dev branch, let me know if this also fixes this issue.

neographikal, 2019-10-04:
Updated to latest docker version, will monitor for problems. Disabling zigbee2mqtt-assistant didn't help btw.

JBS5, 2019-10-05:
> @manup [dresden-elektronik/deconz-rest-plugin#1261 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-523470291) could it be that the lights change the networkAddress without announcing the new one? Otherwise it would be strange that group commands still work? A re-power would trigger a device announce resulting in the gateway receiving the new network address.

About the working group command in deCONZ: All of the times sooner of later also a group command failed and have to powercycle the light.

neographikal, 2019-10-05:
Unfortunately last nights fix did not resolv the dropping of the Tradfri lamps from the network. Will try to sniff the traffic later this weekend.

mecrip, 2019-10-10:
I've got the same issue starting from 1.6.

 I downgraded but looks like that something is crashed in the stick because the issue is still happening. Using latest source routing firmware.

mecrip, 2019-10-10:
In my case it will also bring down the whole network after some time ,so other devices (like xiaomi) became not reachable until system reboot.

neographikal, 2019-10-10:
Didn't get the sniffing working yesterday. Will try again.

neographikal, 2019-10-15:
I need some help with setting up the sniffer. Flashed the stick with the zboss firmware, downloaded wireshark, started the zboss gui, pointed it to Wireshark, added the keys andddd no traffic in Wireshark.

Zboss does do something, the LED on the stick goes from flashing to solid green when starting Wireshark but that's about it. No traffic shows up. Tried to capture to pcap file in zboss gui, but that stays empty as well.

Running on Win10 X64

bochelork, 2019-10-15:
Sounds correct to me. That's how I sniff as well.

neographikal, 2019-10-15:
It's collecting data now. Hopefully the lamps will drop off the network this night.

neographikal, 2019-10-16:
Of course everything worked this morning. Still, pcap might contain relevant info. Attached captured traffic.
[Everything OKAY.zip](https://github.com/Koenkk/zigbee2mqtt/files/3732828/Everything.OKAY.zip)

mecrip, 2019-10-19:
My issue disappeared when i switched the cc2531 with a new one. Maybe it was crashed oor something.. but even if reflashed it did not work properly.
The newer adapter works flawlessly since a week.

neographikal, 2019-10-21:
It was stable for 4 days here, but when I got home lamps weren't working again. Makes capturing the problem difficult but will try anyhow...

neographikal, 2019-10-24:
Finally have a capture from a night where the lamps stop working! When I wake up none except one lamp was working. Zigbee2mqtt was still working though.

[Lamps gone from network.zip](https://github.com/Koenkk/zigbee2mqtt/files/3768852/Lamps.gone.from.network.zip)

neographikal, 2019-10-27:
There is definitely something going on with the Tradfri stuff. I bought an Ikea hub to test and see which firmware versions my lamps are: latest for all of them. Pairing, no fuss at all, works like a charm. Because I changed the CC2531 stick to another one I had to repair everything. 

Xiaomi Aqara stuff: pairs instantly, no matter how far away from the coordinator they are. Hold the button for 4 secs and presto, joins the network, no fuss. The Ikea stuff doesn't want to comply:

```
Oct 27 14:01:26 neo-server zigbee2mqtt[1302]: #033[32mzigbee2mqtt:info #033[39m 2019-10-27T13:01:26: Starting interview of 'Tradfri lamp bay window 2'
Oct 27 14:01:26 neo-server zigbee2mqtt[1302]: #033[32mzigbee2mqtt:info #033[39m 2019-10-27T13:01:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Tradfri lamp bay window 2"}}'
Oct 27 14:01:46 neo-server zigbee2mqtt[1302]: #033[31mzigbee2mqtt:error#033[39m 2019-10-27T13:01:46: Failed to interview 'Tradfri lamp bay window 2', device has not successfully been paired
Oct 27 14:01:46 neo-server zigbee2mqtt[1302]: #033[32mzigbee2mqtt:info #033[39m 2019-10-27T13:01:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"Tradfri lamp bay window 2"}}'
```

Not just one of 'em, but most of them. Really difficult to get them on the network while all the simple battery powered Xiaomi stuff works fine. Eventuelly it works, but it's not going really smooth. I'm now at:

Version of Zigbee2Mqtt: 1.6.0 (latest dev of today).
Coordinator version: 20190619 (source).

What can I do to provide debug info to resolve this? Or I could send over / bring by some lamps :)

Koenkk, 2019-10-27:
As you have the IKEA hub now, it would be interesting to see how stable it is when having the bulbs paired to the IKEA hub.

neographikal, 2019-10-27:
will try that as well, but can you make something of the traces I made? I can't pair the bulbs with zigbee2mqtt at all while the Xiaomi stuff works flawless

Koenkk, 2019-10-27:
Is the Ikea hub still turned on? You should power down the complete ikea hub network (hub and bulbs)

LukeHandle, 2019-10-27:
@neographikal I had pairing issues with the tradfri GU10 bulb. It stayed in the flashing state and was not being picked up by z2m (took 5+ attempts. Unsure if holding very near helped our not). The LED controller and E27 were painless though.

Initially (before changing the z2m security key - not sure if related) I only had the GU10 connected and it sometimes came on by itself. Hasn't happened in the last week since setting up again.

Can't confirm firmware versions, but I can dig up the models if that helps.

neographikal, 2019-10-27:
Hmm the GU10's aren't giving me trouble, Powered everything down and started to pair everything 1 by 1. Paired the first three E27's succesfully, now I got two that won't. And they have worked correctly.

It is all over the place, no pattern to be recognized, but surely not stable..

neographikal, 2019-10-27:
Ok, now it's getting even weirder; as a last resort I flashed the Z-stack 3.0.x firmware for the CC2531. All the light bulbs pair **instantly**, no problems at all......

No clue how stable the network is going to be but the difference with the other version is remarkable.....

Koenkk, 2019-10-28:
@neographikal I've ran my production network a few months on CC2531 Z-Stack 3 and consider it stable.

neographikal, 2019-10-28:
Thanks, I'll monitor the stability. Using the current daily build of zigbee2mqtt as well. First results are encouraging! Maybe update the docs so that people will start using zstack 3.0?

Koenkk, 2019-10-28:
@neographikal let's first see if it keeps working now.

neographikal, 2019-10-28:
The answer is no. In the morning, everything was still fine. Logs do not provide a lot of useful info:

Oct 28 18:35:08 neo-server zigbee2mqtt[1302]: #033[31mzigbee2mqtt:error#033[39m 2019-10-28T17:35:08: Publish 'set' 'state' to 'Tradfri lamp bay window 2' failed: 'Error: Data request failed with error: 'No network route' (205)'

Lamps just drop off the network again. What can I do to help debug this?

Koenkk, 2019-10-29:
Please try https://github.com/Koenkk/zigbee2mqtt/issues/2032#issuecomment-546702323

neographikal, 2019-10-29:
Already did that, no other hubs present, disconnected all the Ikea bulbs and paired them 1 by 1. No difference.

Koenkk, 2019-10-29:
So when the bulbs are paired to the ikea hub they also end up being uncontrollable?

neographikal, 2019-10-29:
Haven't used the ikea hub extensively but can try. Can't imagine that won't work tbh, this stuff gets sold to the masses, if it was this shitty Ikea would get everything back :) But will try anyway.

Koenkk, 2019-10-29:
@neographikal while looking at your wireshark log I found an interesting thing. 

When things work:
[image]

When things stop working:
[image]

The strangest thing is that after the on/off command, which the bulb doesn't respond to, it still send a link status update, meaning that it is still in the network.

neographikal, 2019-10-29:
Pfoee so this is really a bug in the Tradfri bulbs? I've got the problem with the E14 non color / non WB version, E27 same version ánd the GU10's..... That would be really really shitty.

Koenkk, 2019-10-30:
The only way to find that out is to test them with the tradfri hub.

aherbjornsen, 2019-10-30:
I've experienced identical problems with bulbs falling out, in particular "GU10 WS" bulbs.

I updated my GU10s and some "E14 WS" and "E27 WS" I have with the latest test build (2.3.007) a couple of weeks ago.

Since then I've had one single GU10 drop out, which is a huge improvement.

Koenkk, 2019-10-30:
Afaik tradfri bulbs with the 1.x firmware are not updated anymore (in case you have that)

neographikal, 2019-10-30:
I don't have WS numbers on them, I have a bunch of LED1623G12's and LED1649CS's and some GU10 spots.

SalexSun, 2020-01-02:
Just wanted to contribute my findings regarding these bulbs. I have similar problems with the Ikea Tradfri LED1650R5 bulbs. All these bulbs can suddenly stop responding to commands from zigbee2mqtt and at some point they usually start responding again.

I'm running Hassio with the zigbee2mqtt addon installed. In total there are about 40-50 devices where the majority are routers. Recently I moved from CC2531 to CC26X2R1 (very good!) hoping it would also fix the problems with the IKEA bulbs. The IKEA bulbs are located outsize, approximately 15m from the coordinator, but no more than 5m from other routers in the network. 

In total I have 12 of these bulbs connected to zigbee2mqtt. I have configured 2 zigbee groups for these bulbs. One groups contains 4 bulbs and the other contains 8 bulbs. Controlling the groups has **never** been problematic, only when controlling individual bulbs problems occur. 

So for those still having problems with these bulbs. When configuring them as for a zigbee group they seem to always respond to commands

groenmarsmannetje, 2020-01-02:
I also have similar issues with the IKEA Tradfri GU10 bulbs. After a while some do not report and respond anymore this happens quite randomly. Even the bulbs that are near the coordinator might drop off. This started when I added the last eight bulbs. And I have more then more then 50 devices  now and more then half of them are also routers. I use a CC2531 stick with source routing firmware and 2 additional sticks as router. Luckily the bulbs keep responding if you use them as groups so all my automations and remote still work with these grouped bulbs.

I am also considering a CC26X2R1 as coordinator now. Please can you confirm if it is more stable with that device.

Otherwise I might consider to split the mesh network into 2. But I am not sure yet if and how I can set that up on a single raspberry pi with Hassio.

[2 more replies]

closed 2020-03-10 by stale[bot]