Koenkk/zigbee2mqtt #2036: cc2530+cp2102 and GBAN (cc2530+RFX2401) not recognized?

bartve66, 2019-09-25:
I am trying a new setup:
RPI 3 B+
hass.io with mosquitto mqtt broker and Zigbee2mqtt Hass.io Add-on
Fresh install, newest versions. I am trying to use 2 different coordinators, alternating a cc2530+cp2102 and a GBAN (cc2530+RFX2401), with the latest firmware.

When I start the Zigbee2mqtt Hass.io Add-on it seems each of the coordinators is not recognized?
This is the log of the Zigbee2mqtt Hass.io Add-on:

```
2019-09-25T14:05:18: PM2 log: App [npm:0] starting in -fork mode-
2019-09-25T14:05:18: PM2 log: App [npm:0] online
> zigbee2mqtt@1.6.0 start /zigbee2mqtt-1.6.0
> node index.js
  zigbee2mqtt:info 9/25/2019, 2:05:23 PM Logging to directory: '/share/zigbee2mqtt/log/2019-09-25.14-05-21'
  zigbee2mqtt:info 9/25/2019, 2:05:23 PM Starting zigbee2mqtt version 1.6.0 (commit #unknown)
  zigbee2mqtt:info 9/25/2019, 2:05:23 PM Starting zigbee-shepherd
events.js:174
      throw er; // Unhandled 'error' event
      ^
Error: Port is not open
    at SerialPort.close (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:425:29)
    at Znp.<anonymous> (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/znp/znp.js:107:37)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/znp/znp.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/znp/znp.js:3:12)
    at SerialPort.serialPort.open (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/znp/znp.js:103:45)
    at SerialPort._error (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:197:14)
    at binding.open.then.err (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:241:12)
Emitted 'error' event at:
    at errorOrDestroy (internal/streams/destroy.js:107:12)
    at SerialPort.onerror (_stream_readable.js:733:7)
    at SerialPort.emit (events.js:198:13)
    at SerialPort._error (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:199:10)
    at process.nextTick (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/@serialport/stream/stream.js:204:31)
    at process._tickCallback (internal/process/next_tick.js:61:11)
npm
 ERR! code ELIFECYCLE
npm ERR!
 errno 1
npm
 ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR!
 A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-25T12_05_23_311Z-debug.log
2019-09-25T14:05:24: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
```

This beyond my skill level, does anyone have an idea?

james-fry, 2019-09-26:
Try with different USB port for me it’s /dev/ttyUSB0 and make sure that rtscts is false

{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "pan_id": 6754,
    "log_level": "debug",
    "channel": 11,
    "rtscts": false,
    "network_key": [private
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {}
}

bartve66, 2019-09-27:
Thanx, James. I tried some things in line with your suggestions and it works now ... but I am not really sure it works the way it should be.

This i what i did:

```
ls -la /dev/tty*
crw-rw-rw-    1 root     root        5,   0 Sep 26 22:39 /dev/tty
crw-rw----    1 root     audio     204,  64 Sep 26 22:39 /dev/ttyAMA0
crw-rw----    1 root     audio     188,   0 Sep 26 22:39 /dev/ttyUSB0
```
Next:
```
dmesg | grep tty
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait cgroup_enable=memory 8                 250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x400                 00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000837] console [tty1] enabled
[    0.812800] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[    2.500230] usb 1-1.2: cp210x converter now attached to ttyUSB0
[...]

```

So I tried /dev/ttyUSB0 inthe zigbee2mqtt add-on config, and that worked. But why does the GBAN coordinator not show up in the ls -la /dev/tty* output?

Some other information, the output of hass.io>system>hardware in the hass.io web interface:
```
serial:
/dev/ttyAMA0
/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
/dev/ttyUSB0
```

james-fry, 2019-09-29:
Glad you got it working

closed 2019-12-05 by stale[bot]