Koenkk/zigbee2mqtt #2037: Unable to calibrate temperature on Aqara WSDCGQ11LM

gimlichael, 2019-09-25:
# Bug Report
## What happened
My Aqara WSDCGQ11LM Temperature Humidity Sensor sends wrong temperature by 1 celcius. I tried to op-in for tweaking of the temperature, but it does not seem to work (configuration.yaml).

## What did you expect to happen
I'll expect, that values i decide to adjust will work as documented (although documentation is not 100% clear).

## How to reproduce it (minimal and precise)
In the configuration.yaml i have tried these options in accordance with https://www.zigbee2mqtt.io/configuration/device_specific_configuration.html and https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html (both variants does not work):

devices:
  '0x00158d0003cd0347':
    friendly_name: '2-Hygro-Temp-Fyrrum'
    retain: true
    temperature_calibration: -1

devices:
  '0x00158d0003cd0347':
    friendly_name: '2-Hygro-Temp-Fyrrum'
    retain: false
    device_options:
        temperature_calibration: -1

## Debug Info
zigbee2mqtt version:  1.6.0 (commit #e26ad2a)
CC253X firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2019-09-26:
Your configuration is correct, this was a bug. 

Fixed in the dev branch, thanks!

closed 2019-09-26 by Koenkk

gimlichael, 2019-09-26:
Thank you for the quick-fix.

I just did a git pull pointing to the dev branch.
After npm start, i still don't see the calibrated value in either Domoticz or in the log.

Can you maybe point to what I am doing wrong?
This is my config:

```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://*intentionally-masked*:1883'
serial:
  port: COM3
devices:
  '0x00158d0003cd0347':
    friendly_name: '2-Hygro-Temp-Fyrrum'
    retain: false
    temperature_calibration: -1
```

magcode, 2019-10-06:
Same here, I don't see the calibrated value

```
git rev-parse HEAD
e26ad2aca2eeb0aa366982481eb921ee603fb6d1
```

`info: ztemp1 (0x00158d0002f946c5): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)`

```
   '0x00158d0002f946c5':
    friendly_name: ztemp1
    retain: false
    temperature_calibration: -0.8
```

entropyst72, 2020-01-06:
Same here, with or without "device_options:"

> '0x00158d000110030c':
>   friendly_name: '0x00158d000110030c'
>   temperature_precision: 1
>   temperature_calibration": -2

(temperature_precision option works).

magcode, 2020-01-06:
works for me meanwhile. Got the latest release?

onlymejosh, 2020-04-22:
I'm running into the same issue, any ideas I am on `911cd58ef27e071d20f48249fd16d53b89ac888d`

daha76, 2020-06-02:
I experiencing the same issue pressure calibration for Aqara sensor doesn't work. Still raw value from sensor is reported.

max0x19, 2020-12-24:
Hello.
I have the same problem with the Osram Motion AC01353010G.
Below a part of configuration.yaml:

devices:
'0x00158d00052dcfgh':
	friendly_name: aqaraweather_ext
'0x000d6f000f5e4569':
	friendly_name: 'osram_motion'
	device_options:
	temperature_calibration: 2

The **temperature_calibration** doesn't work.
Could you help me, please?
Regards

matejdro, 2021-03-22:
I also can't get this to work:

```
devices:
  '0x00158d000678e798':
    friendly_name: Temp Sensor
    temperature_calibration: -2
```

@magcode sorry for pinging, could you post your config please?

matejdro, 2021-03-22:
I did wait (number was changing, but calibration was not applied).

h-tro, 2021-04-08:
### same issue
device: Xiaomi/Aqara MCCGQ11LM und WSDCGQ11LM
command: zigbee2mqtt/AqaraDW-01/set
{"temperature_calibration": -2}
log: Zigbee2MQTT:error 2021-04-08 22:37:23: No converter available for 'temperature_calibration' (-2)

kwolniak, 2021-04-29:
@Koenkk why it's not posiible to calibrate humidity?