Koenkk/zigbee2mqtt #2038: Support for Paulmann RGB Remote 500.67

bochelork, 2019-09-25:
I'm trying to add Paulmann RGB Remote 500.67 [link](https://nl.paulmann.com/binnenverlichting/smart-home/zigbee/smarthome-zigbee-afstandsbediening-gent-met-wandhouder/50067)
Some commands are straight forward (White Temp, +/-) but some command are not shown by z2m. Sniffer data shows activity but I don't know how to receive them in zigbee2mqtt. 
Especially the 'Enhance move to Hue and Saturation' (color wheel) would be a very nice to have.
[paulmann remote control.zip](https://github.com/Koenkk/zigbee2mqtt/files/3654953/paulmann.remote.control.zip)

I attached sniffer data. I tried to execute the commands by time: Network address = 0x1562
```
0-5s press -
5-10s press +
10-15s standbye
15-20s move from red via purple to blue on color wheel
20-25s move from  red via yellow to green on color wheel
25-30s Press W
30-35s move from cold-warm to warm white
35-40 press 1
40-45 press 2
45-50 press 3
50-55 press P
```

Koenkk, 2019-09-27:
- Could you share your network key? Otherwise I cannot decrypt it
- Can you switch to the dev branch and start zigbee2mqtt with `DEBUG=* npm start`, do you see any logging when pressing the buttons?

bochelork, 2019-09-27:
Changing color:
[Pastebin
](https://pastebin.com/qQN0hrp2)
I can see:
Skipping command 'enhancedMoveToHueAndSaturation' because it is missing from the lookup

and 

zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":3,"clusterid":768,"srcaddr":5474,"s...

Groupid follows the last number I pressed. 

Commands in the remote which I see are:
```
colorLoopSet
commandMoveToColorTemp
enhancedMoveToHueAndSaturation
commandMove
commandStop

```

Koenkk, 2019-09-27:
I've added the missing events, with the latest dev branch you should now see a missing converters warning.

bochelork, 2019-09-27:
I do see the messages and I'm getting data. Trying to update the
herdsman-convertors. But having other big problems atm. All my xiaomi wall
switches are offline.

probutus, 2020-05-05:
Sorry for doing thread-necromancy but I have the paulmann remote and I would be very happy if I could pair it; unfortunately it does not show up when I activate pairing. Is there a way to connect it to zigbee2mqtt? (My Ikea button is working flawlessly). Alternatively, do you have a colour wheel remote control which is working with zigbee2mqtt?

Heronimonimo, 2020-05-07:
I would be very interested in using this remote because of the colour wheel as well.

Koenkk, 2020-05-07:
Looks that @bochelork got quite far in adding support for this device.

@probutus @Heronimonimo please try to pair the device with zigbee2mqtt. 
- If this fails first check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
- If after that it still fails sniff the traffic while pairing the device: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

probutus, 2020-05-07:
I checked the howto.
* Permit join is enabled
* Zigbee2mqtt is linked to one Ikea tradfri button

* Paulmann remote was linked to two ikea bulbs (white) and working

I did the following:
1) reset the remote, this caused the Ikea lamps to join the zigbee2mqtt network
2) tried to pair the remote (pressing "1" >3s, pressing "2" >3s, pressing "3" > 3s), no log output shown
3) paired the remote with one bulb again (pressing "1" >3sec) pairing worked. I could switch the lamp
4) reset the remote again, Ikea lamps again joined the zigbee2mqtt network
5) cursed and opened a beer

I collected all the steps in the debug log below (except "5").
[zigbee.txt](https://github.com/Koenkk/zigbee2mqtt/files/4595116/zigbee.txt)

I also ordered a second cc2531 usb dongle to do the zigbee sniffing, this should arrive on saturday. If there is anything more I can try please let me know

Cheers,
Alex

`

probutus, 2020-05-09:
Second cc2531 arrived. I did a similar session like above

1) pushed button of already paired tradfri button to check logging is working
2)pushed sequentially all 4 buttons on the remote to start pairing 
3) paired with Ikea-Lamps, and able to switch them off

Please let me know if I can log more

BR,

Alex

[zigbee.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4603428/zigbee.pcapng.gz)
[zigbee_herdsmen.log](https://github.com/Koenkk/zigbee2mqtt/files/4603429/zigbee_herdsmen.log)

Koenkk, 2020-05-09:
I don't see any joining attempts of the remote in this sniff.

Could it be that the remote only works on certain channels? Can you try sniffing the traffic on channel 15?

probutus, 2020-05-09:
here comes the log. The only output I see from the remote is this Touchlink scan request

[sniff_ch15.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4603450/sniff_ch15.pcapng.gz)

probutus, 2020-05-09:
I checked; It's operating on channel 20 (like my hue bridge). Attached a log.
I did the following
1) toggle the coupled IKEA lamp on->off->on
2) pushed the pairing buttons sequentially 1, 2 3 P
Again, there are these ZLL Touchlink scan requests
[sniff_ch20.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4603475/sniff_ch20.pcapng.gz)

Koenkk, 2020-05-09:
You probably need to turn off whole your hue network, otherwise the device pairs to hue instead of zigbee2mqtt.  (point 3 of https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all). On which channel is zigbee2mqtt running?

probutus, 2020-05-09:
zigbee2mqtt is on channel 11

probutus, 2020-05-09:
Thanks a lot for your help, that's much appreciated!!
[sniff_ch11.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4603815/sniff_ch11.pcapng.gz)
[sniff_ch11.txt](https://github.com/Koenkk/zigbee2mqtt/files/4603816/sniff_ch11.txt)

I switched off the complete hue system incl. all lamps. I only kept the tradfri button.

Koenkk, 2020-05-11:
I don't see the remote attempting to pair yet. So how to correctly pair it first has to be figured out. @bochelork how did you manage to pair it? (it seems you've successfully done this?)

probutus, 2020-05-15:
is there anything I can do further? Alternatively, is there a chance to get the philips living colors remote supported?

Koenkk, 2020-05-15:
Hopefully someone from this issue can help how to pair it.

Regarding the living colors remote, are you sure this is a zigbee device? Are you able to pair it?

probutus, 2020-05-15:
According to the manual I found it is using the ZLL protocol. I can pair it with Ikea and Philips lamps.

bochelork, 2020-05-15:
I was able to pair the device,  but don't recall how.  I will check tonight or this weekend and let you know.

bochelork, 2020-05-15:
Ok, I paired the remote, this is how:
- Press W-key for 3 seconds
- Press Power On/Off Key for 5 seconds.

Now an unsupported devices joined the network. So we have to play with the devices.js. Maybe we can add support together. 

I copied Muller Licht remote to the paulmann section:
```
    {
        zigbeeModel: ['RemoteControl '],
        model: '500.67',
        vendor: 'Paulmann',
        description: 'RGB Remote control',
        supports: 'toggle, brightness, other buttons are not supported yet!',
        fromZigbee: [
            fz.tint404011_on, fz.tint404011_off, fz.cmdToggle, fz.tint404011_brightness_updown_click,
            fz.tint404011_move_to_color_temp, fz.tint404011_move_to_color, fz.tint404011_scene,
            fz.tint404011_brightness_updown_release, fz.tint404011_brightness_updown_hold,
        ],
        toZigbee: [],
    },

```
and at least there are some results. But the colour functions are not supported yet. Anyone who wants to dig in to this?
```

igbee2mqtt:debug 2020-05-15 22:29:50: Received Zigbee message from '0x00158d0002447fd9', type 'commandColorLoopSet', cluster 'lightingColorCtrl', data '{"bits":15,"bytee":0,"action":0,"direction":0,"time":0,"starthue":0}' from endpoint 1 with groupID 1
zigbee2mqtt:debug 2020-05-15 22:29:50: No converter available for '500.67' with cluster 'lightingColorCtrl' and type 'commandColorLoopSet' and data '{"bits":15,"bytee":0,"action":0,"direction":0,"time":0,"starthue":0}'
zigbee2mqtt:debug 2020-05-15 22:29:50: Received Zigbee message from '0x00158d0002447fd9', type 'commandEnhancedMoveToHueAndSaturation', cluster 'lightingColorCtrl', data '{"enhancehue":62965,"saturation":254,"transtime":1}' from endpoint 1 with groupID 1
zigbee2mqtt:debug 2020-05-15 22:29:50: No converter available for '500.67' with cluster 'lightingColorCtrl' and type 'commandEnhancedMoveToHueAndSaturation' and data '{"enhancehue":62965,"saturation":254,"transtime":1}'
zigbee2mqtt:debug 2020-05-15 22:29:56: Received Zigbee message from '0x00158d0002447fd9', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4509,"imageType":30736,"fileVersion":386146324}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-05-15 22:29:56: No converter available for '500.67' with cluster 'genOta' and type 'commandQueryNextImageRequest' and data '{"fieldControl":0,"manufacturerCode":4509,"imageType":30736,"fileVersion":386146324}'
```

probutus, 2020-05-16:
Wow!! The pairing instructions were the missing link! I paired the remote and logged via wireshark; I also attached the debug log from zigbee2mqtt
After pairing was done I pushed the buttons "W", then the white bar, then the color wheel, then "+", "-" and OFF

Unfortunately due to the amount of output the beginning of the textual debug log got cut; the wireshark trace is complete
[pairing_remote_logfile.txt](https://github.com/Koenkk/zigbee2mqtt/files/4638050/pairing_remote_logfile.txt)
[pairing_remote.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/4638051/pairing_remote.pcapng.gz)

probutus, 2020-05-16:
I updated your entry to avoid being spammed with the Ota-Messages in the log:
`    {
        zigbeeModel: ['RemoteControl '],
        model: '500.67',
        vendor: 'Paulmann',
        description: 'RGB Remote control',
        supports: 'toggle, brightness, other buttons are not supported yet!',
        fromZigbee: [
            fz.tint404011_on, fz.tint404011_off, fz.cmdToggle, fz.tint404011_brightness_updown_click,
            fz.tint404011_move_to_color_temp, fz.tint404011_move_to_color, fz.tint404011_scene,
            fz.tint404011_brightness_updown_release, fz.tint404011_brightness_updown_hold,
            fz.ignore_genOta,
        ],
        toZigbee: [],
    },

I now receive the data from the colourwheel via debug log:
zigbee2mqtt:debug 2020-05-16 11:42:17: No converter available for '500.67' with cluster 'lightingColorCtrl' and type 'commandEnhancedMoveToHueAndSaturation' and data '{"enhancehue":46003,"saturation":254,"transtime":1}'
zigbee2mqtt:debug 2020-05-16 11:42:17: No converter available for '500.67' with cluster 'lightingColorCtrl' and type 'commandColorLoopSet' and data '{"bits":15,"bytee":0,"action":0,"direction":0,"time":0,"starthue":0}'
zigbee2mqtt:debug 2020-05-16 11:42:17: No converter available for '500.67' with cluster 'lightingColorCtrl' and type 'commandEnhancedMoveToHueAndSaturation' and data '{"enhancehue":25700,"saturation":254,"transtime":1}'

If I understand right, we can not use this value directly to create an absolute rgb value out of it (we only have hs and not v). So my idea would be to output the values directly via mqtt and have the instance controlling the lamp (in my case openhab2) taking the current "v" value and update the color. does this make sense?`

bochelork, 2020-05-16:
The debug-message also tells the group ID, this is the number (1-3) which is selected before giving a command. We need this number as well in de mqtt output.

zigbee2mqtt:debug 2020-05-16 14:45:16: Received Zigbee message from '0x00158d0002447fd9', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":286,"transtime":1}' from endpoint 1 with **groupID 3**

probutus, 2020-05-16:
@bochelork: Thanks for the hint :) I was wondering how to use the three buttons.
I managed to get the data out of Zigbee2mqtt:
My new devices.js (extract):
    {
        zigbeeModel: ['RemoteControl '],
        model: '500.67',
        vendor: 'Paulmann',
        description: 'RGB Remote control',
        supports: 'toggle, brightness, other buttons are not supported yet!',
        fromZigbee: [
            fz.tint404011_on, fz.tint404011_off, fz.cmdToggle, fz.tint404011_brightness_updown_click,
            fz.tint404011_move_to_color_temp, fz.tint404011_move_to_color, fz.tint404011_scene,
            fz.tint404011_brightness_updown_release, fz.tint404011_brightness_updown_hold,
            fz.ignore_genOta, fz.command_move_hue_and_sat, fz.command_color_loop_set,
        ],
        toZigbee: [],
    },`
`
I also added the following entries to to the fromZigbee.js:
    command_color_loop_set: {
        cluster: 'lightingColorCtrl',
        type: 'commandColorLoopSet',
        convert: (model, msg, publish, options, meta) => {
            const payload = {
                    action: 'circle_hold',
                    bits: msg.data.bits,
                    bytee: msg.data.bytee,
                    action_val: msg.data.action,
                    direction: msg.data.direction,
                    time: msg.data.time,
                    starthue: msg.data.starthue,
            };
            if (msg.groupID) payload.action_group = msg.groupID;
            return payload;
        },
    },
    command_move_hue_and_sat: {
        cluster: 'lightingColorCtrl',
        type: 'commandEnhancedMoveToHueAndSaturation',
        convert: (model, msg, publish, options, meta) => {
            const payload = {
                    action: 'circle_hold',
                    enhancehue: msg.data.enhancehue,
                    saturation: msg.data.saturation,
                    transition_time: msg.data.transtime,
            };
            if (msg.groupID) payload.action_group = msg.groupID;
            return payload;
        },
    },
I attached the files aswell
The MQTT output looks like this:
zigbee2mqtt:info  2020-05-16 16:02:09: MQTT publish: topic 'zigbee2mqtt/paulmann_remote', payload '{"linkquality":94,"rate":80,"enhancehue":61423,"saturation":254,"bits":15,"bytee":0,"direction":0,"time":0,"starthue":0,"action":"circle_hold","transition_time":1,"action_group":1}'

I still have the following questions
1) the genOta messages are not filtered although I added the fz.genOta to the properties
2) I would expect to receive two different messages "enhancedMoveToHueAndSaturation" and "colorLoopSet" but somehow both zigbee messages show up in one big message
[files.zip](https://github.com/Koenkk/zigbee2mqtt/files/4638503/files.zip)

probutus, 2020-05-16:
UPDATE:
I have a first version which handles all buttons on the remote control including group-ID's as attribute in the mqtt-message (action_group: 0..3).
@Koenkk , @bochelork: review comments are very welcome (especially the adding of the action_group to the existing entries in fromZigbee.js)

Having the action group as an attribute in the mqtt message is imho not optimal since we always have to parse the message before linking it to a specific
[files2.zip](https://github.com/Koenkk/zigbee2mqtt/files/4639024/files2.zip)
 target device. Is it possible to modify the mqtt topic based on the action_group?

bochelork, 2020-05-16:
Nice work @probutus! Finally putting the remote in my scripts!
 
I'm just not happy yet with the color wheel. I get quite some false hits with "action":"circle_hold" after pressing other buttons. Not sure if it's the remote or the protocol. And... what to do with the enhancehue value? How can I convert it to a value I can publish to my lamps?

Finally it's quite a flood off commands from the remote. If the light is blinking you should better wait with pressing buttons for a new command....

bochelork, 2020-05-16:
@probutus 

> 3. How did you figure out how to pair the device? I searched the whole internet and could not find the hint. Is this some secret druid knowledge?

Had to dig in some manuals:
https://cdn.paulmann.com/media/pdf/d5/33/da/SMF_Montageanleitung_V17_180615_GB.pdf
Remote is described starting from page 80.

probutus, 2020-05-16:
> Nice work @probutus! Finally putting the remote in my scripts!
>
> I'm just not happy yet with the color wheel. I get quite some false hits with "action":"circle_hold" after pressing other buttons. Not [...]

I have a new version which puts the enhancedhue-value in a "standard" format between 0 and 360 and the saturation to a percentage value. With this, I can now control my lamp via the color wheel.

[fromZigbee.zip](https://github.com/Koenkk/zigbee2mqtt/files/4639232/fromZigbee.zip)

However, I still see some strange behaviour when the remote is waking up. It re-sends the last colour wheel command independently on what button you were pushing. I feel like this is a "feature" of zigbee2mqtt when a wakeup comes it refreshes the last state. 
Also, I figured out that the colour temperature delivers values between 153 and 370. I currently want to convert the number in my openhab2 ruleset and get funny error messages like
"Rule 'Receive paulmann rc item and react': Could not cast 299 to java.lang.Number; line 20, column 27, length 34"

probutus, 2020-05-16:
I found the reason for the strange behaviour: The parsing of the loop message caused the strange flickering. I removed it and now everything looks smoother. files attached
Review comments welcome
[files4.zip](https://github.com/Koenkk/zigbee2mqtt/files/4639373/files4.zip)

bochelork, 2020-05-17:
Thanks @probutus. It's working better now. My lamps don't accept hue, so I have to convert it to RGB, but it seems okay now.

probutus, 2020-05-17:
@bochelork : That's great news. Now we need to convince @Koenkk to put our source in the mainline. I found out now that the value 360 is not a vaild hue value. I fixed this in the latest version.
[files_final.zip](https://github.com/Koenkk/zigbee2mqtt/files/4640203/files_final.zip)

I still have two topics where I would need feedback from @Koenkk :+1: 
1) the genOta messages are not filtered although I added the fz.genOta to the properties
2) I would be very happy if there is a way to modify the mqtt topic name based on the groupID. This would allow me a much higher automation and no need for dedicated rules to dispatch between different clients

Koenkk, 2020-05-17:
Based on this I added an initial implementation here. The main changes are the use of the already existing ZCL compliant converters. You can find the changes here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1240/files

**Before** trying this first update to the latest zigbee2mqtt dev branch, I found some issue with the command color loop set parsing. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

probutus, 2020-05-17:
I tested it and it is working; I had to adapt my rules file a little (attached a mqtt log below)

I have some question left regarding the group_id: Is there a chance to modify the mqtt topic based on the action group? In this case I can directly map the topics to different devices.

Thank you all for your help!!

```
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/log', ... (91 bytes))
{"type":"device_announced","message":"announce","meta":{"friendly_name":"paulmann_remote"}}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":52,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (160 bytes))
{"linkquality":31,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":1028,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":86,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (161 bytes))
{"linkquality":92,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":43947,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":84,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (161 bytes))
{"linkquality":84,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":22873,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":81,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (161 bytes))
{"linkquality":78,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":19532,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":81,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (161 bytes))
{"linkquality":81,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":21588,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":70,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (160 bytes))
{"linkquality":73,"action":"enhanced_move_to_hue_and_saturation","action_enhanced_hue":6168,"action_saturation":254,"action_transition_time":1,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (84 bytes))
{"linkquality":68,"action":"brightness_move_down","action_rate":80,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (62 bytes))
{"linkquality":68,"action":"brightness_stop","action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (84 bytes))
{"linkquality":68,"action":"brightness_move_down","action_rate":80,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (62 bytes))
{"linkquality":70,"action":"brightness_stop","action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (84 bytes))
{"linkquality":68,"action":"brightness_move_down","action_rate":80,"action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (62 bytes))
{"linkquality":68,"action":"brightness_stop","action_group":3}
[... cut ...]
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (62 bytes))
{"linkquality":39,"action":"brightness_stop","action_group":3}
Client mosq-8icUaEmERcJ2eOubby received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/paulmann_remote', ... (242 bytes))
{"linkquality":57,"action":"color_loop_set","action_update_flags":{"action":true,"direction":true,"time":true,"start_hue":true},"action_action":"deactivate","action_direction":"decrement","action_time":0,"action_start_hue":0,"action_group":3}

```

Koenkk, 2020-05-18:
Good to hear that it works!

> I have some question left regarding the group_id: Is there a chance to modify the mqtt topic based on the action group?

This is not possible, everything is published to `zigbee2mqtt/[FRIENDLY_NAME]`  which keeps things consistent and simple. In your scripts just check for the `action_group` property.

Just to confirm, can I merge the current code in the mainline?

probutus, 2020-05-18:
:+1: Confirmed. Change is working, please commit to mainline:

@bochelork With this commit, you need to re-calculate the hue and saturation value on your own:

hue = extendedhue * 360 / 65536 % 360
saturation = extendedsaturation * 200 / 256 % 100

Koenkk, 2020-05-18:
Does it make sense to move this calculation inside the converter?

bochelork, 2020-05-18:
Think that's better, not having to think about conversion in user programs. Is it right not all lamps support a command in hue? I tried with my osram but I received a message that 'hue' was not valid. So I had to convert to RGB which worked fine.

probutus, 2020-05-18:
> Does it make sense to move this calculation inside the converter?

there are two contradicting goals here:
1) with the extended hue and saturation we are able to show more colours
2) Compatibility to the existing systems who operate on the smaller values
I am open for both ways but I have the feeling that newer devices will use the extended values (since it is also written like that in the Zigbee standard).
Why not offer both values in the mqtt message "extendedhue",  "extendedsat" , "hue" and "sat". This also eliminates the problem of having to support two incompatible worlds. zigbee2mqtt takes the values which are offered and calculates the other ones?

Koenkk, 2020-05-18:
Adding both is fine. Would it also make sense to add an x/y value? (as this is used within zigbee2mqtt to control bulb colors). If yes, what formula do you use?

probutus, 2020-05-18:
I do not know about the x/y conversion. I understood that this is an CIE colour room representation. Maybe @bochelork can comment on this?

bochelork, 2020-05-18:
My scripts are all in Python, there are build-in colour conversion functions.  About EnhancedHue, isn't that a bit like shooting with a cannon on a mosquito? What will be the resolution of the colour wheel?

probutus, 2020-05-18:
@bochelork : I got your point. I thought about this aswell:
The remote only gives hue and saturation this means we can not generate a rgb or x/y value out of it since the brightness is missing. For the remote it doesn't make sense to have 24bit color representation but there will be other devices in the future using this representation. So overall it makes sense to agree on one representation which is shared among all clients and this is also distributed for all classes involving brightness and hue and this one representation should have the maximum resolution possible. (atm 24bit).

Koenkk, 2020-05-19:
Merged the initial PR, implementation will be available in the latest dev branch in a few hours.

probutus, 2020-05-19:
Very good!! Thanks a lot! Is there a chance that someone updates the documentation (especially regarding the way in how to pair the remote would be worth mentioning)

Koenkk, 2020-05-20:
For the docs, please edit the following file: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/500.67.md

Regarding the `saturation`, according to the ZCL spec this is the same as the non enhanced variant of this command. (so this is not a enhanced saturation, only the hue is enhanced)

Heronimonimo, 2020-05-20:
Great!

Made a pull for adition of pairing instruction and explanation on hue and saturation: https://github.com/Koenkk/zigbee2mqtt.io/pull/296

probutus, 2020-05-20:
> For the docs, please edit the following file: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/500.67.md
>
> Regarding the `saturation`, according to the ZCL spec this is the same as [...]

Ah, ok. Missed that one.. The saturation which is coming from the remote is 0..255 and the saturation used by "my" application is in the range of "0..100". So I concluded there is also an enhanced saturation available. Is then the one with 0..254 the correct one?

Heronimonimo, 2020-05-20:
Maybe a saturation_pct with a range from 0 to 100 would be the correct fix for that.

sjorge, 2020-05-20:
> Maybe a saturation_pct with a range from 0 to 100 would be the correct fix for that.

To be conssitant we'd probably need to add it to all remotes/bulbs that publish the normal saturation 0-255 range too then.

Koenkk, 2020-05-20:
@sjorge saturation should indeed by in 0-255 range, updated the code.

probutus, 2020-05-20:
tested the latest master 1.13.1 -> Works as expected
Thank you all very much for your help and support!! I will have some beer for you!

Koenkk, 2020-05-21:
Great, thanks all! I assume this can be closed now.

closed 2020-05-21 by Koenkk