Koenkk/zigbee2mqtt #2039: bug: JavaScript Type error

cbrandlehner, 2019-09-26:
# Bug Report
**Yes**
## What happened
I upgraded to version 1.6.0 and found errors in the log. (see below)
## What did you expect to happen
Looks like the code is trying to access a JavaScript variable that is undefined. Code should use a typeof check first.

## How to reproduce it (minimal and precise)
Upgrade to version 1.6.0 and have zigbee device that supports energy information.
In my case it is: https://www.zigbee2mqtt.io/devices/Z809AF.html

## Debug Info
zigbee2mqtt version: 1.6.0 (commit #ca481d7)
CC253X firmware version: 20190608

# LOG output:
```
debug 2019-09-26T07:10:01: Received Zigbee message from 'Z2' of type 'attributeReport' with data '{"instantaneousDemand":0}' from endpoint 1 with groupID 0
error 2019-09-26T07:10:01: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'seMetering' of undefined
    at Object.convert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:1246:35)
    at converters.forEach (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:146:41)
    at Array.forEach (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:144:20)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:317:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:55:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:430:22))
```

Koenkk, 2019-09-27:
Should be fixed now, can you check with the latest dev branch?

cbrandlehner, 2019-09-29:
checked with dev branch - OK now.

closed 2019-09-29 by Koenkk