Koenkk/zigbee2mqtt #2041: Unable to install on armv6

t1b3r1um, 2019-09-26:
Howdy!

I've been trying to install Zigbee2MQTT on my Raspberry Pi (old model), and am hitting an error when attempting to install. 

NPM = 6.9.0
Node= 10.16.3

Soon after the install starts, it will log a warning that zigbee-herdsman and typescript failed the md5 check, and will attempt to re-download.

The install will eventually:

pi@raspberrypi:/opt/zigbee2mqtt $ npm install
npm WARN tarball tarball data for zigbee-herdsman@0.5.7 (sha512-5Ak/kPb5CC+smRU3n1v5g/GEGEMKHQoBrZKmaA1EFfTYx0hMC47055U/Ajatu1Urdd4sXFzSyxwHBMX0eaytGg==) seems to be corrupted. Trying one more time.
WARN tarball tarball data for zigbee-herdsman@0.5.7 [            ......] | extranpm WARN tarball tarball data for typescript@3.5.3 (sha512-ACzBtm/PhXBDId6a6sDJfroT2pOWt/oOnk4/dElG5G33ZL776N3Y6/6bKZJBFpd+b05F3Ct9qDjMeJmRWtE2/g==) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for zigbee-herdsman@0.5.7 (sha512-5Ak/kPb5CC+smRU3n1v5g/GEGEMKHQoBrZKmaA1EFfTYx0hMC47055U/Ajatu1Urdd4sXFzSyxwHBMX0eaytGg==) seems to be corrupted. Trying one more time.
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/takeUntil.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/typescript-36151077/lib/protocol.d.ts'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/takeWhile.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/takeWhile.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/takeWhile.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/tap.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/tap.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/tap.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/testing/TestMessage.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/testing/TestMessage.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/testing/TestMessage.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/testing/TestScheduler.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/testing/TestScheduler.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/testing/TestScheduler.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/add/operator/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operator/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operators/throttle.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/add/operator/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operator/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operators/throttleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/add/observable/throw.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/observable/throw.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/observable/throwError.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/observable/throwError.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/observable/throwError.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/throwIfEmpty.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/throwIfEmpty.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/throwIfEmpty.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operators/throwIfEmpty.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/add/operator/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operator/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operators/timeInterval.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sample.d.ts'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm2015/internal/operators/timeout.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sample.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/_esm5/internal/operators/timeout.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sample.js.map'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/add/operator/timeout.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sampleTime.d.ts'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/internal/operators/timeout.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sampleTime.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b/operator/timeout.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/rxjs-990150cd/internal/operators/sampleTime.js.map'
[... cut ...]
pi@raspberrypi:/opt/zigbee2mqtt/node_modules $ ls
pi@raspberrypi:/opt/zigbee2mqtt/node_modules $ cd ..
pi@raspberrypi:/opt/zigbee2mqtt $ nano package.json
pi@raspberrypi:/opt/zigbee2mqtt $ nano package.json
pi@raspberrypi:/opt/zigbee2mqtt $ npm --version
6.9.0
pi@raspberrypi:/opt/zigbee2mqtt $ node --version
v10.16.3

I'm not very proficient in linux, so I haven't done too much troubleshooting. I did have to install node.js using this guide:

https://warlord0blog.wordpress.com/2018/06/27/node-js-v8-on-raspberry-pi-zero/

Koenkk, 2019-09-27:
Looks like an npm issue, can you try it again with `rm -rf node_modules && npm install`.

subdancer, 2019-09-28:
hello, i'm havin the same problem on a fresh install on raspbian buster lite.
tried your command 
rm -rf node_modules && npm install
but the resulting warnings and logs are the same as posted above.
the npm version is below:

`node --version
v10.16.3
npm --version
6.11.3
`

`npm WARN tarball tarball data for zigbee-herdsman@0.5.7 (sha512-5Ak/kPb5CC+smRU3n1v5g/GEGEMKHQoBrZKmaA1EFfTYx0hMC47055U/Ajatu1Urdd4sXFzSyxwHBMX0eaytGg==) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for zigbee-herdsman@0.5.7 (sha512-5Ak/kPb5CC+smRU3n1v5g/GEGEMKHQoBrZKmaA1EFfTYx0hMC47055U/Ajatu1Urdd4sXFzSyxwHBMX0eaytGg==) seems to be corrupted. Trying one more time.
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee2mqtt/node_modules/.staging/fsevents-2bc44009/node_modules/tar/index.js'
npm WARN tar ENOENT: no such file or directory, lstat '/opt/zigbee2mqtt/node_modules/.staging/fsevents-2bc44009/node_modules/tar/lib'
.....
`

john2exonets, 2019-09-28:
+1 Same issue.  Using a tried & true Pi3 running raspbian stretch.

raphaelg, 2019-09-28:
Same issue here as well, tried to update and now nothing works anymore

Koenkk, 2019-09-28:
I could reproduce this issue on a Pi 1, I'm not sure why it occurs or if I can fix it. 

However you can workaround this by executing:

```bash
rm -rf node_modules && npm install --no-shrinkwrap
```

zoral566, 2019-10-01:
Dev branch build:
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-10-01T09:44:21: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-01.10-44-17'
zigbee2mqtt:info  2019-10-01T09:44:22: Starting zigbee2mqtt version 1.6.0 (commit #92fcf5a)
zigbee2mqtt:info  2019-10-01T09:44:22: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-10-01T09:44:30: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-01T09:44:30: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:427:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
zigbee2mqtt:error 2019-10-01T09:44:30: Failed to start zigbee
zigbee2mqtt:error 2019-10-01T09:44:30: Exiting...
zigbee2mqtt:error 2019-10-01T09:44:30: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:427:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-01T09_44_30_267Z-debug.log

pi@raspberrypi:/opt/zigbee2mqtt $ npm -v
6.1.0
pi@raspberrypi:/opt/zigbee2mqtt $ node -v
v10.5.0

piit79, 2019-10-10:
Thanks for the workaround, @Koenkk . I didn't even remove `node_modules`, I just did
```
npm install --no-shrinkwrap
```
after the first failure and it worked fine.

zoral566, 2019-10-10:
> Thanks for the workaround, @Koenkk . I didn't even remove `node_modules`, I just did
> 
> ```
> npm install --no-shrinkwrap
> ```
> 
> after the first failure and it worked fine.

I do not understand why it fails. Recent installation raspberry pi zero:
pi@raspberrypi:/opt $ sudo git clone https://github.com/koenkk/zigbee2mqtt.git -b dev
Cloning into 'zigbee2mqtt'...
remote: Enumerating objects: 7246, done.
remote: Total 7246 (delta 0), reused 0 (delta 0), pack-reused 7246
Receiving objects: 100% (7246/7246), 10.75 MiB | 1.10 MiB/s, done.
Resolving deltas: 100% (5082/5082), done.
pi@raspberrypi:/opt $ node -v
v10.15.0
pi@raspberrypi:/opt $ sudo chown -R pi:pi /opt/zigbee2mqtt
pi@raspberrypi:/opt $ cd zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm install --no-shrinkwrap
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 679 packages from 477 contributors and audited 877112 packages in 574.318s
found 0 vulnerabilities

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2019-10-10T14:02:51: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-10-10.16-02-41'
zigbee2mqtt:info  2019-10-10T14:02:53: Starting zigbee2mqtt version 1.6.0 (commit #ba92f73)
zigbee2mqtt:info  2019-10-10T14:02:53: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-10-10T14:03:00: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-10T14:03:00: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
zigbee2mqtt:error 2019-10-10T14:03:00: Failed to start zigbee
zigbee2mqtt:error 2019-10-10T14:03:00: Exiting...
zigbee2mqtt:error 2019-10-10T14:03:00: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-10-10T14_03_00_828Z-debug.log

piit79, 2019-10-10:
@zoral566 Your problem is unrelated to this issue - your `npm install` completed successfully.

Looks like a MQTT connection issue.

Koenkk, 2019-10-10:
@zoral566; @piit79 is right; please check https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

zoral566, 2019-10-10:
> @zoral566; @piit79 is right; please check https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

thank you for taking care of me. 
CC2530 raspberry serial port:
pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/ttyAMA0 
crw-rw---- 1 root dialout 204, 64 okt   10 16:41 /dev/ttyAMA0

pi@raspberrypi:/opt/zigbee2mqtt $ ls -l /dev/serial/by-id
ls: '/dev/serial/by-id' nem érhető el: No such file or directory
pi@raspberrypi:/opt/zigbee2mqtt $ test -w /dev/serial1 && echo success || echo failure
success
pi@raspberrypi:/opt/zigbee2mqtt $ cat data/configuration.yaml 
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyAMA0
advanced:
  baudrate: 115200
  rtscts: false
pi@raspberrypi:/opt/zigbee2mqtt $ sudo service mosquitto status
● mosquitto.service - Mosquitto MQTT v3.1/v3.1.1 Broker
   Loaded: loaded (/lib/systemd/system/mosquitto.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-10-10 16:14:10 CEST; 29min ago
     Docs: man:mosquitto.conf(5)
           man:mosquitto(8)
 Main PID: 3473 (mosquitto)
   Memory: 468.0K
   CGroup: /system.slice/mosquitto.service
           └─3473 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

okt 10 16:14:09 raspberrypi systemd[1]: Starting Mosquitto MQTT v3.1/v3.1.1 Broker...
okt 10 16:14:10 raspberrypi systemd[1]: Started Mosquitto MQTT v3.1/v3.1.1 Broker.

t1b3r1um, 2019-10-23:
The fix for me was to download and manually install herdsman@0.5.7 prior to running the install for zigbee2mqtt. There was another package (typewriter?) that also had to be manually installed. 

In-case someone finds this in the future :)

closed 2019-10-23 by t1b3r1um