Koenkk/zigbee2mqtt #2044: linkquality is bad even with antenna

guitar9, 2019-09-27:
Hello, 
I have bought a CC2531 with antenna. But the linkquality is bad. I am 7 meters away from the stick with 2 walls between and the linkquality is 31.
Somebody can say me the connecting to the antenna is correct?
[image]
[image]

zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608

guitar9, 2019-09-28:
I tried to soldering with another cc2531 like in this tutorial. https://hassiohelp.eu/2019/03/02/zigbee-cc2531-antenna/

But it is the same. linkquality 31 :( 
Maybe it has something to do with my wlan router? Should I try to change the channel of the cc2531?

guitar9, 2019-09-28:
1.5 meters away from the stick behind a chair a get linkquality 47. And without antenna no connection. This is shit :dagger:

guitar9, 2019-09-28:
Is it possible that my internet router sends on the same frequency and destroys my zigbee network? I have a speedport W724V

Koenkk, 2019-09-28:
Did you already try https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html?

Note that the antenna on the CC2531 is actually quite good already (due to it's design).

guitar9, 2019-09-28:
yes I have a usb extension cable and I tried to change to channel

guitar9, 2019-10-02:
I solved with using a cc2531 flashed as routers

closed 2019-10-02 by guitar9