Koenkk/zigbee2mqtt #2045: Osram Smart Plug not pairing

tangopium, 2019-09-28:
Hi at all,

I'm trying to pair my Osram Smart+ plug (AB3257001NJ) but it fails. I'm running the latest version of zigbee2mqtt (dev branch) with the newest firmware version (19.06.2019) for the CC2531.

Please see the log here: https://pastebin.com/6aSnLqne

I guess the most relevant line from the log is the following:

`Interview failed for '0x7cb03eaa0a00286b with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms' +5m`

Is this a bug in zigbee2mqtt or does the device has an issue (I just bought it on Amazon)?

Thanks for your help!

tangopium, 2019-09-28:
After trying it for about 5 times in row it finally worked. May this be a timeout issue? Does the device needs longer to reply?

Koenkk, 2019-09-28:
It could be, let's see if other people also run into this issue for this device or if it's just bad luck.

mabi, 2019-10-01:
I have three of them, working flawlessly and always pairing at first try.

TilmanK, 2019-10-15:
> It could be, let's see if other people also run into this issue for this device or if it's just bad luck.

I have the same issue and I'm fiddling around with it for some days now. I'm also running the latest dev branch (0527b89) and I tried all available firmwares for the CC2531 including the 3.0 version.

Pairing works only, when I move the devices really close to the stick. (half a meter between stick and plug). Once paired (the have a link strength between 70 and 130) I can switch them On and Off via MQTT and get an appropriate response within a millisecond or so.

My Problem starts once I move them to the places where I need them (1 and 2 floors up), I won't get anything back from them anymore. The plug one floor up (2,5 meter away from the Coordinator) is receiving messages I send on MQTT (and is actually switched on/off), but will not answer:

´{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x7cb03eaa0a024857' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"0x7cb03eaa0a024857"}}´

The plug on the second floor isn't reachable an all anymore:
´{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x7cb03eaa0a013056' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"0x7cb03eaa0a013056"}}´

I don't see anything useful in the logs:
https://pastebin.com/uNia8ewT

Any help is highly appreciated!

Koenkk, 2019-10-15:
Are other devices working well? You might want to read https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html

bochelork, 2019-10-17:
I've Osram ralated pairing problems as well. Not sure if it's related. 

```
zigbee2mqtt:info  2019-10-17T18:24:26: Starting interview of 'RGBWOSRAM'
zigbee2mqtt:error 2019-10-17T18:24:46: Failed to interview 'RGBWOSRAM', device has not successfully been paired

```
Tried it over 10 times, always the same result, with different CC2531's.

Sniffer data:
[pairing osram.zip](https://github.com/Koenkk/zigbee2mqtt/files/3740761/pairing.osram.zip)

bochelork, 2019-10-17:
Okay, I have it paired now, but I had to clear my database. Without any devices paired, the Osram strip paired fine. After that I modified my original db file with the data from the successful pairing.
Now I can control my led strip. Not sure if all devices are still in the air. Am I on the limits of my CC2531?

Koenkk, 2019-10-18:
@bochelork you are not on the limits; otherwise it wouldn't pair. Note that even though interview fails; it can still work properly.

lolorc, 2019-10-18:
I'm also have osram plugs, 5 of them, have re-paired them this week, it worked flawlessly, and it has always been straightforward with those devices, even with version <= 1.6

TilmanK, 2019-10-31:
So, I finally found the time to play around again with those plugs.

To sum things up:
1. One plug is broken and does not pair correctly.
2. I managed to get the zigbee network stable with that tutorial Koenkk posted for now at least for those Osram plugs.

Thanks,

Tilman

closed 2020-01-06 by stale[bot]

st-wo, 2020-11-27:
Hi,

I am also having trouble to pair the OSRAM plugs. Zigbee2mqtt does not even react. No prompt in the log. Should I hold the OSRAM button after the click sound? Should the LED react in pairing mode (like flashing)?

Any ideas?

I am running openhabian in Raspberry Pi Zero W with CC2531.