Koenkk/zigbee2mqtt #2046: Errors starting new install of zigbee2mqtt@1.6.0 on Rpi3 B+

hocomadvies, 2019-09-28:
# Bug Report
## What happened
I followed the instructions Flashing the usb stick and installing zigbee2mqtt in the Rpi. Than after starting:
 pi@raspberrypi:/opt/zigbee2mqtt $ npm start

I get some errors and than the program exit. So it is not working. 
Can't figure out why not. To me it seems to be an error in the node.js ? could not find any info about this error. 

Here is the debug log:
https://pastebin.com/ZMjVgr1C

## Debug Info

zigbee2mqtt version: Latest 
CC253X firmware version: Latest
Using a CC2531 with antenna from here: https://nl.aliexpress.com/item/32982628286.html?spm=a2g0s.9042311.0.0.71a84c4dqUm7lV 
How can I determine if I got the CC2531 and not the CC2540? (what is even the difference?).
The green led is off when it is in the Rpi? Is that normal behaviour? Has been on for a while after I did a reboot.

Koenkk, 2019-09-28:
Can you post the output which you get after `npm start`?

hocomadvies, 2019-09-29:
`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.6.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 9/28/2019, 2:34:11 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-28.14-34-09'
  zigbee2mqtt:info 9/28/2019, 2:34:11 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 9/28/2019, 2:34:11 PM Starting zigbee-shepherd
  zigbee2mqtt:info 9/28/2019, 2:34:34 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
  zigbee2mqtt:info 9/28/2019, 2:35:34 PM Starting zigbee-shepherd
  zigbee2mqtt:error 9/28/2019, 2:35:52 PM Error while starting zigbee-shepherd! (Error: SRSP - SYS - osalNvRead after 6000ms)
  zigbee2mqtt:error 9/28/2019, 2:35:52 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 9/28/2019, 2:35:52 PM Failed to start
        {"message":"SRSP - SYS - osalNvRead after 6000ms","stack":"Error: SRSP - SYS - osalNvRead after 6000ms\n    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:227:24)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 9/28/2019, 2:35:52 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-28T13_36_27_981Z-debug.log
`

Koenkk, 2019-09-29:
All communication with the stick fails. Can you verify that you indeed have a cc2531 and not a cc2540? This is written on the chip.

hocomadvies, 2019-09-29:
> All communication with the stick fails. Can you verify that you indeed have a cc2531 and not a cc2540? This is written on the chip (very small though)

On the board is written CC25** on the chip it says CC2540 Do I need a different hex file for flashing this stick? Or do I need a new stick?

hocomadvies, 2019-09-29:
Ok, now I found more info on the stick and I understand it is cc2540 bluetooth dongle not zigbee. So need to find an other cc2531 with antenna :) Nice.

closed 2019-09-29 by hocomadvies