Koenkk/zigbee2mqtt #2047: Error while starting zigbee-shepherd - docker version 1.6.0

hajo62, 2019-09-28:
I made an update of my docker install to **1.6.0** as described on your pages.
When I now start zigbee2mqtt, I see the following error message. 
After about a minute the container restarts with the same result. 
I do see, sensor-value in my homeassistant:

```
docker-compose up zigbee2mqtt
mosquitto is up-to-date
Recreating zigbee2mqtt ... done
Attaching to zigbee2mqtt
zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      | 
zigbee2mqtt      | > zigbee2mqtt@1.6.0 start /app
zigbee2mqtt      | > node index.js
zigbee2mqtt      | 
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 1:14:39 PM Logging to directory: '/app/data/log/2019-09-28.13-14-37'
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 1:14:39 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 1:14:39 PM Starting zigbee-shepherd
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 1:14:53 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
```

Koenkk, 2019-09-28:
- Are you sure the serial path is correct?
- Can you provide the logging when setting the following environment variable in the docker container `DEBUG=zigbee-herdsman*`?

hajo62, 2019-09-28:
- serial path: Yes, because I do see values of my sensores

- That's the output of the docker-compose up command (Hope, that's what you are asking for):
```
docker-compose up zigbee2mqtt
mosquitto is up-to-date
Recreating zigbee2mqtt ... done
Attaching to zigbee2mqtt
zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      | 
zigbee2mqtt      | > zigbee2mqtt@1.6.0 start /app
zigbee2mqtt      | > node index.js
zigbee2mqtt      | 
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 4:20:39 PM Logging to directory: '/app/data/log/2019-09-28.16-20-38'
zigbee2mqtt      |   zigbee2mqtt:debug 9/28/2019, 4:20:39 PM Removing old log directory '/app/data/log/2019-09-28.16-06-27'
zigbee2mqtt      |   zigbee2mqtt:debug 9/28/2019, 4:20:40 PM Loaded state from file /app/data/state.json
zigbee2mqtt      |   zigbee2mqtt:debug 9/28/2019, 4:20:40 PM Saving state to file /app/data/state.json
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 4:20:40 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 4:20:40 PM Starting zigbee-shepherd
zigbee2mqtt      |   zigbee2mqtt:debug 9/28/2019, 4:20:40 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
zigbee2mqtt      | 2019-09-28T16:20:40.152Z zigbee-herdsman:znp:log Opening with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00193648CA-if00 and {"baudRate":115200,"rtscts":true}
zigbee2mqtt      | 2019-09-28T16:20:40.180Z zigbee-herdsman:znp:log Serialport opened
zigbee2mqtt      | 2019-09-28T16:20:40.182Z zigbee-herdsman:znp:log Writing skip bootloader payload
zigbee2mqtt      | 2019-09-28T16:20:40.185Z zigbee-herdsman:unpi:writer --> buffer [239]
zigbee2mqtt      | 2019-09-28T16:20:41.208Z zigbee-herdsman:znp:SREQ --> SYS - version - {}
zigbee2mqtt      | 2019-09-28T16:20:41.221Z zigbee-herdsman:unpi:writer --> frame [254,0,33,2,35]
zigbee2mqtt      | 2019-09-28T16:20:47.223Z zigbee-herdsman:znp:timeout SRSP - SYS - version after 6000ms
zigbee2mqtt      | 2019-09-28T16:20:47.247Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":[85]}
zigbee2mqtt      | 2019-09-28T16:20:47.254Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,0,15,0,37]
zigbee2mqtt      | 2019-09-28T16:20:53.259Z zigbee-herdsman:znp:timeout SRSP - SYS - osalNvRead after 6000ms
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 4:20:53 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
zigbee2mqtt      | 2019-09-28T16:20:53.285Z zigbee-herdsman:znp:log Serialport closed
zigbee2mqtt      |   zigbee2mqtt:info 9/28/2019, 4:21:53 PM Starting zigbee-shepherd
zigbee2mqtt      | 2019-09-28T16:21:53.302Z zigbee-herdsman:znp:log Opening with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00193648CA-if00 and {"baudRate":115200,"rtscts":true}
zigbee2mqtt      | 2019-09-28T16:21:53.314Z zigbee-herdsman:znp:log Serialport opened
zigbee2mqtt      | 2019-09-28T16:21:53.314Z zigbee-herdsman:znp:log Writing skip bootloader payload
zigbee2mqtt      | 2019-09-28T16:21:53.315Z zigbee-herdsman:unpi:writer --> buffer [239]
zigbee2mqtt      | 2019-09-28T16:21:54.318Z zigbee-herdsman:znp:SREQ --> SYS - version - {}
zigbee2mqtt      | 2019-09-28T16:21:54.320Z zigbee-herdsman:unpi:writer --> frame [254,0,33,2,35]
zigbee2mqtt      | 2019-09-28T16:22:00.326Z zigbee-herdsman:znp:timeout SRSP - SYS - version after 6000ms
zigbee2mqtt      | 2019-09-28T16:22:00.331Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":[85]}
zigbee2mqtt      | 2019-09-28T16:22:00.333Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,0,15,0,37]
zigbee2mqtt      | 2019-09-28T16:22:06.338Z zigbee-herdsman:znp:timeout SRSP - SYS - osalNvRead after 6000ms
zigbee2mqtt      |   zigbee2mqtt:error 9/28/2019, 4:22:06 PM Error while starting zigbee-shepherd! (Error: SRSP - SYS - osalNvRead after 6000ms)
zigbee2mqtt      |   zigbee2mqtt:error 9/28/2019, 4:22:06 PM Press the reset button on the stick (the one closest to the USB) and start again
zigbee2mqtt      |   zigbee2mqtt:error 9/28/2019, 4:22:06 PM Failed to start
zigbee2mqtt      | 	{"message":"SRSP - SYS - osalNvRead after 6000ms","stack":"Error: SRSP - SYS - osalNvRead after 6000ms\n    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/znp/znp.js:227:24)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
zigbee2mqtt      |   zigbee2mqtt:error 9/28/2019, 4:22:06 PM Exiting...
zigbee2mqtt      | npm ERR! code ELIFECYCLE
zigbee2mqtt      | npm ERR! errno 1
zigbee2mqtt      | npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
zigbee2mqtt      | npm ERR! Exit status 1
zigbee2mqtt      | npm ERR! 
zigbee2mqtt      | npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
zigbee2mqtt      | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt      | 
zigbee2mqtt      | npm ERR! A complete log of this run can be found in:
zigbee2mqtt      | npm ERR!     /root/.npm/_logs/2019-09-28T16_22_06_543Z-debug.log
```

[... cut ...]
9/28/2019, 4:20:40 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
9/28/2019, 4:20:53 PM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
9/28/2019, 4:21:53 PM - info: Starting zigbee-shepherd
9/28/2019, 4:22:06 PM - error: Error while starting zigbee-shepherd! (Error: SRSP - SYS - osalNvRead after 6000ms)
```

Koenkk, 2019-09-28:
All communication with the stick fails, the values you see are cached ones.

Can you try to reflash the stick?

hajo62, 2019-09-28:
Hhm, cached value; there is a history in HA with different value.
[image]

hajo62, 2019-09-29:
Do you really think it is helpful to reflash the stick, although the history shows changing values?

negmos, 2019-09-30:
I have exactly the same problem with exactly the same HW/SW setup

Koenkk, 2019-09-30:
@hajo62 yes, I don't know where these values come from, but based on your logs of https://github.com/Koenkk/zigbee2mqtt/issues/2047#issuecomment-536204289 I can only conclude that zigbee2mqtt doesn't start at all -> no data from zigbee can be received.

Before we dive further, I would suggest re flashing the firmware and check if it starts.

hajo62, 2019-09-30:
I did the reflash, but unfortunately the containers still die.
Flash Output:
```
$ sudo ./cc-tool -e -w /private/tmp/CC2531_DEFAULT_20190608/CC2531ZNP-Prod.hex 
Password:
  Programmer: CC Debugger
  Target: CC2531
  Erasing flash...
  Completed       
  Writing flash (241 KB)...
  Completed (17.95 s.)
```
How would I check, that the new version is correct flashed?

Here the log:
```
$ docker-compose up zigbee2mqtt
mosquitto is up-to-date
Starting zigbee2mqtt ... done
Attaching to zigbee2mqtt
zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      | 
zigbee2mqtt      | > zigbee2mqtt@1.6.0 start /app
zigbee2mqtt      | > node index.js
zigbee2mqtt      | 
zigbee2mqtt      |   zigbee2mqtt:info 9/30/2019, 3:58:22 PM Logging to directory: '/app/data/log/2019-09-30.15-58-21'
zigbee2mqtt      |   zigbee2mqtt:debug 9/30/2019, 3:58:22 PM Removing old log directory '/app/data/log/2019-09-30.15-31-51'
zigbee2mqtt      |   zigbee2mqtt:debug 9/30/2019, 3:58:22 PM Loaded state from file /app/data/state.json
zigbee2mqtt      |   zigbee2mqtt:debug 9/30/2019, 3:58:22 PM Saving state to file /app/data/state.json
zigbee2mqtt      |   zigbee2mqtt:info 9/30/2019, 3:58:22 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
zigbee2mqtt      |   zigbee2mqtt:info 9/30/2019, 3:58:22 PM Starting zigbee-shepherd
zigbee2mqtt      |   zigbee2mqtt:debug 9/30/2019, 3:58:22 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
zigbee2mqtt      | 2019-09-30T15:58:22.974Z zigbee-herdsman:znp:log Opening with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00193648CA-if00 and {"baudRate":115200,"rtscts":true}
zigbee2mqtt      | 2019-09-30T15:58:23.011Z zigbee-herdsman:znp:log Serialport opened
zigbee2mqtt      | 2019-09-30T15:58:23.013Z zigbee-herdsman:znp:log Writing skip bootloader payload
zigbee2mqtt      | 2019-09-30T15:58:23.017Z zigbee-herdsman:unpi:writer --> buffer [239]
zigbee2mqtt      | 2019-09-30T15:58:24.041Z zigbee-herdsman:znp:SREQ --> SYS - version - {}
zigbee2mqtt      | 2019-09-30T15:58:24.055Z zigbee-herdsman:unpi:writer --> frame [254,0,33,2,35]
zigbee2mqtt      | 2019-09-30T15:58:30.052Z zigbee-herdsman:znp:timeout SRSP - SYS - version after 6000ms
zigbee2mqtt      | 2019-09-30T15:58:30.078Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":[85]}
zigbee2mqtt      | 2019-09-30T15:58:30.086Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,0,15,0,37]
zigbee2mqtt      | 2019-09-30T15:58:36.091Z zigbee-herdsman:znp:timeout SRSP - SYS - osalNvRead after 6000ms
zigbee2mqtt      |   zigbee2mqtt:info 9/30/2019, 3:58:36 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: SRSP - SYS - osalNvRead after 6000ms)
zigbee2mqtt      | 2019-09-30T15:58:36.121Z zigbee-herdsman:znp:log Serialport closed
zigbee2mqtt      | 2019-09-30T15:59:36.137Z zigbee-herdsman:znp:log Opening with /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00193648CA-if00 and {"baudRate":115200,"rtscts":true}
zigbee2mqtt      |   zigbee2mqtt:info 9/30/2019, 3:59:36 PM Starting zigbee-shepherd
zigbee2mqtt      | 2019-09-30T15:59:36.151Z zigbee-herdsman:znp:log Serialport opened
zigbee2mqtt      | 2019-09-30T15:59:36.152Z zigbee-herdsman:znp:log Writing skip bootloader payload
zigbee2mqtt      | 2019-09-30T15:59:36.152Z zigbee-herdsman:unpi:writer --> buffer [239]
zigbee2mqtt      | 2019-09-30T15:59:37.156Z zigbee-herdsman:znp:SREQ --> SYS - version - {}
zigbee2mqtt      | 2019-09-30T15:59:37.158Z zigbee-herdsman:unpi:writer --> frame [254,0,33,2,35]
zigbee2mqtt      | 2019-09-30T15:59:39.980Z zigbee-herdsman:unpi:parser <-- [254,28,68,129,0,0,5,4,202,101,1,1,0,36,0,52,33,18,0,0,8,24,180,10,0,0,33,245,19,202,101,29,143]
zigbee2mqtt      | 2019-09-30T15:59:39.982Z zigbee-herdsman:unpi:parser --- parseNext [254,28,68,129,0,0,5,4,202,101,1,1,0,36,0,52,33,18,0,0,8,24,180,10,0,0,33,245,19,202,101,29,143]
zigbee2mqtt      | 2019-09-30T15:59:39.986Z zigbee-herdsman:unpi:parser --> parsed 28 - 2 - 4 - 129 - [�e$4�
zigbee2mqtt      | !��e] - 143
zigbee2mqtt      | 2019-09-30T15:59:40.000Z zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1029,"srcaddr":26058,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":1188148,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,180,10,0,0,33,245,19]}}
zigbee2mqtt      | 2019-09-30T15:59:40.005Z zigbee-herdsman:unpi:parser --- parseNext []
zigbee2mqtt      | 2019-09-30T15:59:43.161Z zigbee-herdsman:znp:timeout SRSP - SYS - version after 6000ms
zigbee2mqtt      | 2019-09-30T15:59:43.166Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":[85]}
zigbee2mqtt      | 2019-09-30T15:59:43.168Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,0,15,0,37]
zigbee2mqtt      | 2019-09-30T15:59:43.193Z zigbee-herdsman:unpi:parser <-- [254,3,97,8,0,1,85,62]
zigbee2mqtt      | 2019-09-30T15:59:43.194Z zigbee-herdsman:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
zigbee2mqtt      | 2019-09-30T15:59:43.196Z zigbee-herdsman:unpi:parser --> parsed 3 - 3 - 1 - 8 - [U] - 62
[... cut ...]
zigbee2mqtt      | npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
zigbee2mqtt      | npm ERR! Exit status 1
zigbee2mqtt      | npm ERR! 
zigbee2mqtt      | npm ERR! Failed at the zigbee2mqtt@1.6.0 start script.
zigbee2mqtt      | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt      | 
zigbee2mqtt      | npm ERR! A complete log of this run can be found in:
zigbee2mqtt      | npm ERR!     /root/.npm/_logs/2019-09-30T15_59_49_325Z-debug.log
```

hajo62, 2019-09-30:
User error

closed 2019-09-30 by hajo62

Koenkk, 2019-09-30:
@hajo62 what was the error? (perhaps other users can also benefit from this).

negmos, 2019-09-30:
For me, i reflashed the device and now it works.. i will monitor for some days and see if it will happen again.

hajo62, 2019-09-30:
see https://github.com/Koenkk/zigbee2mqtt/issues/2047#issuecomment-536632038

After the reboot, the old installation (not docker) auto-started the service; the service ran before the docker container could start.