Koenkk/zigbee2mqtt #2050: update.sh never worked for me

itajackass, 2019-09-29:
On every update of Z2M i can't use sudo ./update.sh because i get a lot of errors.

Maybe the code from:

echo "Installing dependencies..."
rm -rf node_modules
npm install

must be:

Koenkk, 2019-09-29:
What errors do you get?

itajackass, 2019-09-29:
I'm sorry i forgot to save log. I remember I get error about some dependencies to install on some folders. But with sudo before the 2 commands posted above, all is ok

Kryzek, 2019-09-29:
I’ll guess it will be ok after sudo chown -R pi:pi /opt/zigbee2mqtt - replace pi with your own username.

Koenkk, 2019-09-29:
@Kryzek yes you are right. The installation instructions already contain this command, in addition a comment has been added that `npm install` should be executed as `pi` user. (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#2-installing). 

I will close this for now, if there are any other problems regarding this, please post it here together with logs.

closed 2019-09-29 by Koenkk