Koenkk/zigbee2mqtt #2056: GE Quirk Smart Switch (PTAPT-WH02) not working

itsfrosty, 2019-09-30:
# Bug Report
## What happened
GE Quirk Smart Switch (PTAPT-WH02) is not supported. I tried to add my own zigbee-shepherd-converters but not able to make it work. Also tried configure from other devices as recommended in the guide. I own smart plug from same company and was able to make it work but no luck on switch :(. 

## What did you expect to happen
- Device pairs successfully but turning the button on/off doesn't print anything on the console
- Sending off/on from mqtt prints things on console but the actual device doesn't turn off/on

## How to reproduce it (minimal and precise)
My device converter: https://pastebin.com/5Ux0zhgF (Smart plug works fine but smart switch doesn't)
Log from pairing the device: https://pastebin.com/3ajxC9LD
Log from sending On/Off from mqtt: https://pastebin.com/593HTz4C
Nothing happens when pressing On/Off on the switch

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-10-01:
Can you share the `database.db` file?

itsfrosty, 2019-10-01:
Thanks for looking into it. Here you go:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/3678777/database.db.zip)

Koenkk, 2019-10-03:
Can you try with the **latest zigbee2mqtt dev branch** and the following config:

```js
{
    zigbeeModel: ['Smart Switch'],
    model: 'PTAPT-WH02',
    vendor: 'GE',
    description: 'Quirky smart switch',
    supports: 'on/off',
    fromZigbee: [fz.state],
    toZigbee: [tz.on_off],
    endpoint: (device) => {
        return {'default': 2};
    },
},
```

itsfrosty, 2019-10-05:
Thank you this worked! Should we add this and the one for smart plug to the devices file?
Here is the final configuration which worked: https://pastebin.com/s7Jf9pVH

itsfrosty, 2019-10-05:
Thanks! This partially worked. I am now able to turn the device ON/OFF by sending commands from mqtt. But turning the actual device ON/OFF doesn't print anything in the debug mode (DEBUG=zigbee-herd* npm start) and also doesn't send anything to mqtt. 

Here is the db file:
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/3692898/database.db.zip)

Koenkk, 2019-10-05:
Great, I assume that you mean that nothing happens when you press the button on the device itself.  This should be solved with the following definition:

```js
{
        zigbeeModel: ['Smart Switch'],
        model: 'PTAPT-WH02',
        vendor: 'GE',
        description: 'Quirky smart switch',
        supports: 'on/off',
        fromZigbee: [fz.state],
        toZigbee: [tz.on_off],
        endpoint: (device) => {
            return {'default': 2};
        },
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(2);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
    },
```

itsfrosty, 2019-10-06:
This worked! Able to turn the device on/off using mqtt and using the button on actual device also reflects back into mqtt.

Do you want to add it to devices.js file?

Thanks for your help.

Koenkk, 2019-10-07:
Done, will be supported in the dev branch soon, thanks!