Koenkk/zigbee2mqtt #2057: [Feature Request] Validate / re-set state

syphernl, 2019-09-30:
Sometimes devices are told to do something (e.g. turn on, close etc) but this message may not reach them. It happened to me this morning where 3 blinds were supposed to open at the same time but only 2 did. Telling the 3rd one to open again worked fine, so it was probably a temporary hiccup.

Is there a way Zigbee2mqtt can re-send a particular device action in case the device is not in a changed state for a given period of time (e.g. 30 seconds)?

Koenkk, 2019-10-01:
- Does this happens a lot or just rarely?
- When this fails, does zigbee2mqtt log an error? (if so please provide the log of this)
 - If this is the case, we can implement some retry mechanism

syphernl, 2019-10-02:
> * Does this happens a lot or just rarely?
It happens from time to time.
> * When this fails, does zigbee2mqtt log an error? (if so please provide the log of this)
Actually this morning:
```
info  2019-10-01T05:16:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'phone_charger' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"phone_charger"}}'

error 2019-10-02T05:50:10: Publish 'set' 'state' to 'blinds_slaapkamer_zijraam' failed: 'Error: Data request failed with error: 'No network route' (205)'
```

If I send the request again, the device go to the intended state. 

>   * If this is the case, we can implement some retry mechanism

Koenkk, 2019-10-02:
What firmware are you currently running? In case you are running the default one I recommend switching to the source routing one (may solve these problems https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing).

syphernl, 2019-10-02:
I'm using the default firmware indeed. What are the pros/cons of switching to this one instead? 

I assume this requires re-pairing of all devices?

Koenkk, 2019-10-02:
The source routing firmware's caches the routes but allows for less direct children (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator).

Switching to this firmware doesn't require re-pairing. (same for switching back)

syphernl, 2019-10-11:
> Switching to this firmware doesn't require re-pairing. (same for switching back)
Hmm, that unfortunately doesn't appear to be the case:

```
zigbee2mqtt:info  2019-10-11T07:29:03: Logging to directory: '/srv/zigbee2mqtt-dev/data/log/2019-10-11.09-29-03'
zigbee2mqtt:info  2019-10-11T07:29:03: Starting zigbee2mqtt version 1.6.0 (commit #b8154e3)
zigbee2mqtt:info  2019-10-11T07:29:03: Starting zigbee-herdsman...
zigbee2mqtt:info  2019-10-11T07:29:11: zigbee-herdsman started
zigbee2mqtt:info  2019-10-11T07:29:11: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:info  2019-10-11T07:29:11: Currently 0 devices are joined:
```

Koenkk, 2019-10-12:
> Seems that the database.db was cleaned out, not sure why. Putting the old one back brings back the devices (pfew!)

Thanks for letting me know, this was a critical zigbee-herdsman bug which cleared the database upon reflash of the stick.

closed 2019-12-19 by stale[bot]