Koenkk/zigbee2mqtt #2058: Adding support for Stelpro "Ki" STZB402+

carldebilly, 2019-10-01:
# Support for Stelpro "Ki" STZB402+
I'm trying to add support for Stelpro's _Ki_ thermostat and I think I need help to make it working properly.

## `devices.js` file

``` javascript
    {
        zigbeeModel: ['STZB402+','STZB402'],
        model: 'STZB402',
        vendor: 'Stelpro',
        description: 'Ki, line-voltage thermostat',
        supports: 'temperature ',
        fromZigbee: [
            fz.thermostat_att_report,
            fz.thermostat_dev_change
        ],
        toZigbee: [
            tz.thermostat_local_temperature,
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_keypad_lockout,
            tz.thermostat_system_mode,
        ],
        configure: async (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 25);
            const actions = [
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genGroups', coordinator, cb),
                (cb) => device.bind('genTime', coordinator, cb),
                (cb) => device.bind('hvacThermostat', coordinator, cb),
                (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('hvacThermostat', 'localTemp', 300, 3600, 0, cb),
                (cb) => device.report('hvacThermostat', 'occupiedHeatingSetpoint', 0, 1800, 1, cb),
            ];
            execute(device, actions, callback);
        },
    },
```
## `lib/resources/homeassistant.js` file
I also added this to `homeassistant.js` file:

``` javascript
    'STZB402': [configurations.thermostat, configurations.sensor_temperature],
```

## Log file error
(I don't know to which problem it's related)
```
PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 7
    at boundsError (internal/buffer.js:49:9)
    at Buffer.readUInt16LE (internal/buffer.js:128:5)
    at BuffaloZcl.readUInt16 (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:70:35)
    at BuffaloZcl.read (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:267:25)
    at BuffaloZcl.read (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:263:26)
    at Function.parsePayloadGlobal (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:198:53)
    at Function.parsePayload (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:150:25)
    at Function.fromBuffer (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:129:30)
    at cb (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:54:46)
    at /zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:27:17
===== Buffer =====
{ type: 'Buffer', data: [ 24, 5, 7, 0, 0, 18, 0 ] }
===== Arguments =====
<Buffer 18 05 07 00 00 12 00> 513
===== zcl-packet parse =====

d
===== zcl =====

d
===============
```

## MQTT Payload
```
{
  "local_temperature":22,
  "linkquality":2,
  "last_seen":"2019-10-01T00:03:21.800Z", 
  "local_temperature_calibration":0,
  "pi_heating_demand":0,
  "occupied_heating_setpoint":10,
  "unoccupied_heating_setpoint":17,
  "occupied_cooling_setpoint":35,
  "remote_sensing":2,
  "control_sequence_of_operation":"heating only",
  "system_mode":"heat"
}
```
## What is working
1. `local_temperature` information is correct
1. `occuped_heating_setpoint` information is working correctly and can be successfully through MQTT.
1. `linqquality` _homeassistant_ sensor component reported correctly.

## What is not working
1. No `climate` component reported to _homessistant_
1. `temperature` component reported to _homessistant_ not working (state is empty)
1. Ability for the thermostat to report its _current operation_ status (heating / not heating) - cluster/attribute `0x0201 / 0x0008`.
1. Ability to display "outdoor temperature" - this feature is implemented by Stelpro itself (for SmartThings) there: <https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L334>
1. Ability to lock keypad using zigbee cluster/attribute `0x204 / 0x0001`

## Details
Zigbee2Mqtt v`1.6.0` (installed through _hass.io_ add-on)

## Related issues
* #497 My first tentative last year to support Stelpro
* #810 @goldriver2 did a support for a similar product, Stelpro Orleans baseboard
* #1650 @MrNickll  adding support for Stelpro Maestro, a very very similar product (newer with more features)

# Question
Any idea how I can fix the _not working_ section? Should I open distinct issues for each of them?

MrNickll, 2019-10-01:
@carldebilly Just making sure, you're using the dev branch? If not I would switch to it seeing as it's now using zigbee-herdsman and I believe zigbee-shepherd will be obsolete eventually. That would prevent you from doing the work twice.
If you look at https://github.com/Koenkk/zigbee2mqtt/issues/1678#issuecomment-532953744, there's a bit of a difference in the way the configure section is written after 1.6 dev. I believe it has to do with how zigbee-herdsman is expecting the reporting data.

For your problems and non working features, I think you can tackle them all on this same issue, but for the sake of simplicity, I'd try to fix them one by one.

So your error reports an incorrect offset. Hard to tell which of the other issues referenced it's related to, but that -327 value does look like an incorrect offset of something.

1. In your "not working - temperature", what do you expect to see?
2. I didn't try to get this working on SMT402 yet.
3. Look into the v1.6.0 devices.js and tozigbee.js. There's a Sinope TH1123ZB line voltage thermostat that's now supported and some code has been written for an outdoor temp display. It's not the same cluster ID/attribute but I think the principle should be similar for Stelpro, just find your cluster and attribute. I haven't got around to it yet due to lack of time. My main concern is more how will I send this data from Home Assistant to my thermostat.
4. I've added the same tz.thermostat_keypad_lockout, but haven't tried to send it a command for that yet.
5. See above. 
Also, your MQTT payload reports:
```
  "occupied_heating_setpoint":10,
  "unoccupied_heating_setpoint":17,
  "occupied_cooling_setpoint":35,
```
This is a line voltage thermostat. Why would it report a cooling setpoint? I'm also unsure whether it should use an unoccupied heating setpoint as this implies presence detection and again, not sure Stelpro thermostats have this feature.

carldebilly, 2019-10-01:
Hi @MrNickll,

No, I'm using `1.6.0`, not the _dev_ version.  I asked Koenkk few days ago if I should code it on zigbee-herdsman or zigbee-shepherd and he told me it doesn't matter.  So for simplicity of development, I choose to just modify the code directly in the HASS.IO container just to try.  I have a lab I can setup for that, but I didn't try it.

I saw your comment, but you're using an `async` _configure_, and since I can't find any documentation about how it's working, I decided to stay with what I have. There's also a `meta:{configureKey:1}` I don't understand.

The -327 is only returned when the device is set to off, otherwise it's returning the right value, so what I need here is a way to change the result when the device is off (once again it's because I can find any doc on how to do this).

1. I don't know... what is the correct value to return when the thermostat is off? an empty string? a null value?
2. It's supported by the ST's handler...  once I understand how it works, I'll do it for this handler.
3. Thanks I'll try that! Did you make it working with yours?
4. Idem
5. The cooling setpoint seems hard-coded in the converter. For the "occupied" vs "unoccupied", it seems it's used by Stelpro for "normal" and "eco" mode. When switching to "eco" mode, it's reported as "unoccupied". So that is working correctly, just wrongly named.

## Additional precisions
* I also have a Stelpro Maestro installed + somme Stelpro Ki _zwave_. Once it works with the "Ki Zigbee", I'm pretty sure I can make the Maestro working.
  Meastro thermostat is very similar to Ki (the white one... the black model is a zigbee coordinator) with only few exceptions, like ability to report temperature. Most of the "cool" features like opened window detection are local to the termostat and not accessible from the zigbee network.
* I have a contact at Stelpro willing to help me in this business, but I need to ask the right question.

Koenkk, 2019-10-01:
@carldebilly perhaps I understood your question wrong, especially for this case it is handier to use zigbee-herdsman/zigbee-herdsman-converters (zigbee2mqtt dev branch) as it has a lot of configure stuff.

carldebilly, 2019-11-01:
@Koenkk @MrNickll
I switched to Herdsman to continue my tests.  I'm using the Zigbee 3.0 firmware - I suppose it's not causing problems to develop?

I tried to reproduce what I did previously with `zigbee-shepherd` to `zigbee-herdsman`... and nothing work at all!

## What I tried

<details>
  <summary>Converter declaration in `converters.js`</summary>

``` javascript
    {
        zigbeeModel: ['STZB402+','STZB402'],
        model: 'STZB402',
        vendor: 'Stelpro',
        description: 'Ki, line-voltage thermostat',
        supports: 'temperature ',
        fromZigbee: [
            fz.thermostat_att_report,
            fz.thermostat_dev_change,
        ],
        toZigbee: [
            tz.thermostat_local_temperature,
            tz.thermostat_occupancy,
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint,
            tz.thermostat_temperature_display_mode,
            tz.thermostat_keypad_lockout,
            tz.thermostat_system_mode,
            tz.thermostat_system_mode,
            tz.thermostat_running_state,
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(25);
            const binds = [
                'genBasic', 'genIdentify', 'genGroups', 'hvacThermostat', 'hvacUserInterfaceCfg',
                'msTemperatureMeasurement',
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(endpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await configureReporting.thermostatPIHeatingDemand(endpoint);
        },
    },
```
</details>

<details>
  <summary>Log file</summary>

```
info  2019-11-01T04:33:21: THERM 4005 cuisine (0xf8f005fffff78d7c): STZB402 - Stelpro Ki, line-voltage thermostat (Router)
[...]
debug 2019-11-01T04:33:47: Received Zigbee message from 'THERM 4005 cuisine', type 'attributeReport', cluster 'hvacThermostat', data '{"systemMode":4}' from endpoint 25 with groupID 0
error 2019-11-01T04:33:47: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/app/lib/extension/deviceReceive.js:83:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:82:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/app/lib/zigbee.js:56:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:405:22)
    at Generator.next (<anonymous>))
debug 2019-11-01T04:33:47: Received Zigbee message from 'THERM 4005 cuisine', type 'attributeReport', cluster 'hvacThermostat', data '{"16412":4}' from endpoint 25 with groupID 0
error 2019-11-01T04:33:47: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/app/lib/extension/deviceReceive.js:83:42)
    at Array.filter (<anonymous>)
[... 90 lines cut ...]
                    "endpointID": 1
                }
            ]
        }
    },
    "appVersion": 104,
    "stackVersion": 4,
    "hwVersion": 1,
    "dateCode": "20170628 00062\u0000",
    "zclVersion": 1,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    }
}
```
</details>

* Nothing works now..
* No MQTT topics published for this device (as it was working with my first try in this issue description)
* Right now I'm only trying to make is working as a dumb thermostat... we'll see later for advanced features.

## Related stuff
* I noticed there's a new support for another thermostat device which is very similar (the company doing it - _Sinopé_ - is only few kilometers from _Stelpro_, so it's probably a startup of old employees): https://github.com/Koenkk/zigbee-herdsman-converters/pull/656
* _Stelpro_'s Maestro https://github.com/Koenkk/zigbee2mqtt/issues/1678#issuecomment-532953744 (already referenced in previous comments)

carldebilly, 2019-11-02:
## Update!
I managed to get the converter to _stop crashing_...

<details>
<summary>converter code</summary>

``` javascript
    {
        zigbeeModel: ['STZB402+','STZB402'],
        model: 'STZB402',
        vendor: 'Stelpro',
        description: 'Ki, line-voltage thermostat',
        supports: 'local temp, keypad lockout, mode, state, backlight, outdoor temp',
        fromZigbee: [
            fz.thermostat_att_report
        ],
        toZigbee: [
            tz.thermostat_local_temperature,
            tz.thermostat_occupancy,
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint,
            tz.thermostat_temperature_display_mode,
            tz.thermostat_keypad_lockout,
            tz.thermostat_system_mode,
            tz.thermostat_system_m,
            tz.thermostat_running_state,
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(25);
            const binds = [
                'genBasic',
                'genIdentify',
                'genGroups',
                'hvacThermostat',
                'hvacUserInterfaceCfg',
                'msTemperatureMeasurement',
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(endpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await configureReporting.thermostatPIHeatingDemand(endpoint);
        },
    },
```
</details>

Not everything work yet (almost nothing), but I'm still getting the following error message, which is very similar to what I got with Zigbee Shepherd...
```
No converter available for 'STZB402' with cluster 'hvacThermostat' and type 'raw' and data '{"type":"Buffer","data":[24,3,7,0,0,18,0]}'
```

Could it be related to [Stelpro's Manufacturer specific Set Heat mode](https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L481)?

I can see the source of the problem is in the `zclFrame.js`/`ZclFrame.parsePayloadGlobal` on the following line:
``` javascript
 entry[parameter.name] = buffalo.read(typeStr, options);
```

<details>
<summary>The value of "offset" is out of range. It must be >= 0 and <= 5. Received 7  [OPEN FOR STACKTRACE]</summary>

stack trace:
```
"RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 7
    at boundsError (internal/buffer.js:54:9)
    at Buffer.readUInt16LE (internal/buffer.js:135:5)
    at BuffaloZcl.readUInt16 (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\buffalo\buffalo.js:70:35)
    at BuffaloZcl.read (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\buffalo\buffalo.js:267:25)
    at BuffaloZcl.read (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\buffaloZcl.js:263:26)
    at Function.parsePayloadGlobal (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js:211:57)
    at Function.parsePayload (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js:159:25)
    at Function.fromBuffer (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\zclFrame.js:138:30)
    at ZStackAdapter.onZnpRecieved (c:\src\zigbee2mqtt\node_modules\zigbee-herdsman\dist\adapter\z-stack\adapter\zStackAdapter.js:414:51)
    at Znp.emit (events.js:182:13)"

options.typeStr = 'uint16'
parameter.type=33
paramter.name='attrId'
```
</details>

Koenkk, 2019-11-03:
> Could it be related to [Stelpro's Manufacturer specific Set Heat mode](https://github.com/stelpro/Ki-ZigBee-Thermostat/blob/master/devicetypes/stelpro/stelpro-ki-zigbee-thermostat.src/stelpro-ki-zigbee-thermostat.groovy#L481)?

Could you sniff this? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

carldebilly, 2019-12-22:
Hi again, I managed to find time to sniff the thermostat. I paired it with SmartThings. Here's my logs:
[Stelpro.zip](https://github.com/Koenkk/zigbee2mqtt/files/3992056/Stelpro.zip)

There's 3 sessions in this...
1. The first one is the pairing process with SmartThings.  The device zigbee address is 6BF4 after pairing.
2. The second one is me playing with the heating setpoint from both the app and the thermostat itself
3. On the third, I setup the display of the external temperature on the thermostat.

Here's the temperature display command:
[image]

carldebilly, 2019-12-22:
One more thing...

## Heating Demand
Here's the "operating status" where the thermostat is reporting when it's heating and not...

[Stelpro Session 4.zip](https://github.com/Koenkk/zigbee2mqtt/files/3992090/Stelpro.Session.4.zip)

[image]
[image]
[image]
[image]

carldebilly, 2019-12-26:
I found the reason why the following configuration isn't working:
``` javascript
await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
```
It works, I can see the payload coming to the thermostat and coming back, but the payload will be converted as a `raw` payload because it's crashing in the `ZclFrame.parseGlobalPayload`. After investigation, it looks like the definition of the _Command_ #7 is not correctly defined in `zigbee-herdsman`. It works when the payload is all zeros (which seems to be the case for common reporting configuration), but in this case it's another `attrId`, so it's overflowing the buffer at reading it.

I'll issue a PR in https://github.com/Koenkk/zigbee-herdsman to fix this problem.

carldebilly, 2019-12-26:
@Koenkk I suspect the failing test is incorrect.  Also I'm wordering if the parsing should be `"flat"` instead of `"repetitive"`...

carldebilly, 2019-12-28:
@MrNickll 
https://github.com/Koenkk/zigbee2mqtt/pull/2628 has been created :-)

Still a draft PR and needs changed to both herdsman & herdsman-converters becore completing this one.

carldebilly, 2020-01-08:
@MrNickll @goldriver2
I just tested with `latest-dev` docker image and it works!  Still not available on latest HASS.IO "edge" addon, but I confirmed it's working properly :-).

closed 2020-01-08 by carldebilly

MrNickll, 2020-01-08:
@carldebilly do you get all the functions? I'm running zigbee2mqtt standalone on its own raspberry pi and updated to latest dev yesterday. Pairing is much quicker with your integration :) 
However I'm not seeing humidity reported. Rest seems to work including outdoor temp display.

carldebilly, 2020-01-08:
@MrNickll I tested most of my implementation using the Stelpro Ki which doesn't have this sensor. I thought it was working, but I could be wrong.

Do you like the keypad locking? ;-)

bobzer, 2020-01-08:
Hi i'm trying to test this but don't know how to because i'm using hassio and the zigbee2mqtt addon

carldebilly, 2020-01-08:
@bobzer I have the same setup, but I installed the _portainer_ add-on and used it to launch a Z2M container just to test.

MrNickll, 2020-01-08:
@carldebilly I haven't tried the keypad locking. I'll check it out tonight.
For humidity: https://github.com/Koenkk/zigbee-herdsman-converters/pull/886
Added device bind for 'msRelativehumidity'. After asking yesterday I checked my database.db and found cluster 1029 was missing from the pairing. Added this to devices.js for SMT402, re-paired and it started reporting humidity!