Koenkk/zigbee2mqtt #2059: Unbinding IKEA ON/OFF switch from Xiaomi socket

pieter, 2019-10-01:
I think I may just be doing something wrong, but the documentation isn't really helping me.

I added an [Ikea On/Off switch](https://www.zigbee2mqtt.io/devices/E1743.html) as first device to Zigbee2MQTT. This worked fine, and am now using the device for automations in Home Assistant.

I then added a [Xiaomi ZNCZ02LM](https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html) which also works fine through Home Assistant.

However, now when I press the Ikea button, the Xiaomi device also turns on / off, which I don't want. Documentation seems to suggest this may be due to binding (but shouldn't binding be explicitly done?). Trying to unbind the two devices results in a time-out:

```
10/1/2019, 2:46:22 PM - error: Failed to unbind  'ikea_kitchen_button' - 6 - (Error: Timed out after 10000 ms)
10/1/2019, 2:46:22 PM - error: Failed to unbind cluster '6' from '0x000d6ffffec93121' to '0x00158d0001fa7eac' (Error: Timed out after 10000 ms)
```
Though I'm not sure this is actually the problem. It seems unrelated to my automations, since the toggling also happens when I turn off Zigbee2Mqtt. 

I tried putting the Xiaomi socket in its own group, thinking that might help, like this:

```
groups:
  '12':
    friendly_name: group_12
    retain: false
    devices:
      - '0x00158d0001fa7eac'
```

but this didn't change anything.

Any tips?

zigbee2mqtt version: Zigbee2mqtt 1.6.0
CC253X firmware version: 20190608

bochelork, 2019-10-03:
I had the same thing. I think it's not zigbee2mqtt related. The switch is paired directly to the socket (even when it was about 10 meters away). My solution was to reset the xiaomi sockets (if I am right long press the button, maybe 4 or maybe even 10 seconds, long time ago) and remove them from the mains when pairing the Ikea switch.

pieter, 2019-10-04:
@bochelork you’re right, unplugging the controller still causes this to happen so they must be talking to each other. Do you remember the procedure to avoid this? I tried pairing the switch first, or the socket first, but they still pair up, even when removing the socket while pairing the button.

@koenkk any ideas? I thought I could avoid this by putting the button and the socket in a different group but that doesn’t seem to do anything.

AS137430, 2019-10-05:
Facing the same issue. But it didn't happen right away. I had 2 of the Xiaomi ZNCZ02LM which were working well and with Home Assistant. I added the Ikea E1743 months later and when trying to sort out the HA automation, suddenly the Xiaomi sockets started to turn on/off in sync with the E1743 1 day after successfuly pairing of the E1743. In the log, the initial message from the E1743 is gone and only the Xiaomi on/off messages are visible.

Open to suggestions on how to solve this.

Koenkk, 2019-10-07:
Fixed in latest dev branch; by default this device is in group 0 and by default the switch controls it. Solved by removing the plug from group 0 by default.

closed 2019-10-07 by Koenkk

TBobsin, 2019-12-17:
Hello,
i have installed Starting zigbee2mqtt version 1.8.0-dev (commit #960f9e4)
but this Problem still exist. 
The Tradfri E1743 - IKEA TRADFRI ON/OFF switch and E1524 - IKEA TRADFRI  remote are toggling the QBKG12LM - Xiaomi Aqara double key wired wall switch.

leonka93, 2020-01-05:
Hello,
I also have the problem that the ikea switch commands me an xiaomi relay.
how do i unbind hassio?
Thank you
Michele.

Skynetnexus, 2021-03-04:
Hi, I too am experiencing TRADFRI on/off switch mistakenly triggering  Xiaomi ZNCZ02LM

Switch:-

{"id":37,"type":"EndDevice","ieeeAddr":"0x842e14fffe1xxx","nwkAddr":8180,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":6,"groupID":901,"type":"group"},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b95","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1614854253491}

Plug :-

{"id":33,"type":"Router","ieeeAddr":"0x00158d00023xxxxx","nwkAddr":44757,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,3,6,16,5,10,1,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":26,"5":55,"8":4954,"9":2051,"100":0,"149":639.4227294921875,"152":0,"154":0},"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-20-2019"}},"genOnOff":{"attributes":{"61440":4289647872,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[12,4],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-20-2019","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1614891595230}

Orochica, 2022-05-29:
Hi I also have the same issue.Any solution to this problem?

TheJulianJES, 2023-01-27:
Just for reference for anyone still having this issue with Xiaomi EU plugs (MAEU01 or MMEU01).
It's a firmware issue on newer versions and you need to downgrade it: https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072