Koenkk/zigbee2mqtt #2060: Mqtt cannot coonect to CC2531 module.

catalin31, 2019-10-01:
Hi,

So, I got a corrupted SD and because I had no back-up I had to buil all up again (not a very big pain since I have just couple of switches and some sensors) 
But when I tried to add back the CC2531 into the configuration.yaml I get this error:

2019-09-30 22:49:00 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/ttyACM0 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 190, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 107, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 124, in async_initialize
    await self.application_controller.startup(auto_form=True)
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 113, in startup
    await self.initialize()
  File "/usr/local/lib/python3.7/site-packages/bellows/zigbee/application.py", line 62, in initialize
    await e.reset()
  File "/usr/local/lib/python3.7/site-packages/bellows/ezsp.py", line 55, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.7/site-packages/bellows/uart.py", line 208, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()

I did enable some more debug logs in the configuration 
logger:
  default: critical
  logs:
    homeassistant.components: info
    homeassistant.components.zha: debug
    bellows: debug
    zigpy: debug

And I get this:
2019-10-01 19:52:24 INFO (MainThread) [homeassistant.components.switch] Setting up switch.mqtt
2019-10-01 19:52:24 INFO (MainThread) [homeassistant.components.switch] Setting up switch.mqtt
2019-10-01 19:52:25 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.updater
2019-10-01 19:52:25 DEBUG (MainThread) [zigpy.appdb] Loading application state from /config/zigbee.db
2019-10-01 19:52:26 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2019-10-01 19:52:26 DEBUG (bellows.thread_0) [bellows.uart] Resetting ASH
2019-10-01 19:52:26 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'1ac038bc7e'
2019-10-01 19:52:26 DEBUG (MainThread) [zigpy.ota] Initialize OTA providers
2019-10-01 19:52:26 DEBUG (MainThread) [zigpy.ota.provider] OTA image directory '/config/zigpy_ota/' does not exist
2019-10-01 19:52:26 INFO (MainThread) [homeassistant.components.notify] Setting up notify.mobile_app
2019-10-01 19:52:28 INFO (MainThread) [homeassistant.components.hue] Invalid authentication for 192.168.1.31, requesting link.
2019-10-01 19:52:32 INFO (MainThread) [homeassistant.components.automation] Initialized trigger test off at time
2019-10-01 19:52:32 INFO (MainThread) [homeassistant.components.automation] Initialized trigger tunr on Front door light at sunset
2019-10-01 19:52:32 INFO (MainThread) [homeassistant.components.automation] Initialized trigger test on at time
2019-10-01 19:52:32 INFO (MainThread) [homeassistant.components.automation] Initialized trigger turn off Front door light at sunrise
2019-10-01 19:52:45 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.cloud
2019-10-01 19:55:05 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-01 19:55:05 DEBUG (MainThread) [bellows.ezsp] /dev/ttyACM0 connection lost unexpectedly: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-01 19:55:05 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart

To be mentioned:
- I only registered 3 devices with the zigbee2mqtt, all paired correctly
- no errors on the zigbee2mqtt side, it connects to the mqtt, it has the devices learned
zigbee2mqtt:info 10/1/2019, 7:55:37 PM Connecting to MQTT server at mqtt://core-mosquitto
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM zigbee-shepherd ready
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM Connected to MQTT server
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00020b75ec', payload '{"occupancy":true,"linkquality":99,"battery":100,"voltage":3035}'
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00027b087c', payload '{"battery":100,"voltage":3025,"water_leak":false,"linkquality":97}'
  zigbee2mqtt:info 10/1/2019, 7:55:38 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00023a6abd', payload '{"water_leak":true,"linkquality":99,"battery":100,"voltage":3005}'

- CC2531 is flashed with the latest firware (_20190619)
- all are on the latest version - zigbee2mqtt is at 1.6, and Mosquitto broker is at 5.1

Any ideas what am I doing wrong? Or what else should I change/check?

Koenkk, 2019-10-01:
`2019-09-30 22:49:00 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry /dev/ttyACM0 for zha` are you using ZHA or zigbee2mqtt; note that these are 2 different solutions.

catalin31, 2019-10-01:
I am using zigbee2mqtt addon and the entry in the configuration.yaml is:
zha:
  usb_path: /dev/ttyACM0
  database_path: /config/zigbee.db

same as it was before I got the corrupted SD

Koenkk, 2019-10-01:
According to your logging you also seem to use ZHA (https://www.home-assistant.io/components/zha/). Please remove this configuration section from your home assistant `configuration.yaml`.

catalin31, 2019-10-01:
ok, I did remove that part and now this is my entire configuration.yaml:
mqtt:
  broker: core-mosquitto
  port: 1883
  username: xx
  password: xx

switch: !include switch.yaml
group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml

logger:
  default: critical
  logs:
    homeassistant.components: info
    homeassistant.components.zha: debug
    bellows: debug
    zigpy: debug

restarted hass and I still get this error:
2019-10-01 20:12:58 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1913375312] Received invalid command: zha/devices
2019-10-01 20:14:06 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-01 20:14:06 DEBUG (MainThread) [bellows.ezsp] /dev/ttyACM0 connection lost unexpectedly: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-10-01 20:14:06 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart

catalin31, 2019-10-01:
I understood the difference, not sure why before it worked just fine with both and I was able to add entities to the overview from the ones learned via the addon.
so I got rid of all the config, using just the zigbee2mqtt now, I have my 3 devices there, it seems connected to mqtt, but I don't know how to add now those to the overview (I cannot find them in the unused entities, and even if I add them manually in the configuration.yaml:
devices:
  '0x00158d00020b75ec':
    friendly_name: 'my_occupancy_sensor'
    retain: true
groups:
  '1':
    friendly_name: group_1
    retain: true
there is nothing I can add to overview...

I understand that this is me doing something wrong, but I just can't figure out what (super sure is something doh), can I get a hint on what is wrong please?

Koenkk, 2019-10-03:
I'm not sure what overview you mean, but it looks that this is an issue from https://github.com/danielwelch/hassio-zigbee2mqtt/issues? If so, please create one there and close this one.

catalin31, 2019-10-04:
yes, I use hassio on PI3. I will get help on the other topic, thanks!

closed 2019-10-04 by catalin31