Koenkk/zigbee2mqtt #2064: Cannot start Zigbee2mqtt

MarcE85, 2019-10-02:
zigbee2mqtt version: 1.6.0.0
CC2531 firmware version: latest

Since my last update I'm not able to start the zigbee2mqtt service.
I already tried to replug the stick and also pressed the reset button, but nothing worked.

DEBUG log:
[debug start error log](https://pastebin.com/92dGG9m7)

Regards Marc

way2many, 2019-10-03:
Same here on latest dev docker (which changed name?)

Koenkk, 2019-10-04:
Please check https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

If that doesn't work try reflashing the stick.

FouFouCode, 2019-11-27:
I've got the same crash, checked https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html, reflashed twice the stick, unplugged the device, sometimes pushed the nearest button close to USB port, but still not able to start zigbee2mqtt with a CC2531.
Logs can be found at https://pastebin.com/Lp8JZdAc
Any help will be appreciated ;-)
Regards,

Koenkk, 2019-11-27:
For the CC2531, you shouldn't set "rtscts":true in the `configuration.yaml`.

FouFouCode, 2019-11-27:
It's already set to false. Here's my configuration.yaml:
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: xxx
  password: xxx
serial:
  port: /dev/ttyACM0
advanced:
  rtscts: false

FouFouCode, 2019-11-28:
I found the (stupid) issue: I did receive a CC2540 instead of a CC2531. And, you know what: this is not working as expected...
My mistake.
Sorry for the noise!

Koenkk, 2019-11-28:
Thanks, then I will close this as the OP didn't respond.

closed 2019-11-28 by Koenkk