Koenkk/zigbee2mqtt #2066: has cc2652r better range?

guitar9, 2019-10-02:
Has somebody tried cc2652r?
Has it a better range then cc2531?

lolorc, 2019-10-02:
I've been an early adopter, I'm pretty sure it still has bugs.
The most important bug to me was I had my z2m systemd service file to stop the daemon, use uhubctl to power cycle my cc2652r, then restart it every 7h. This problem is gone for me with the latest unofficiall firmware. This evening I decided to move my important devices to cc2652r+herdsman/dev. (some of them were still using a more stable network with cc2530+cc2591)
fingers crossed !
My fear for now is, will I have to repair my devices if I power cycle or upgrade the firmware of my cc2652r. Time will tell :)

concerning range, it's definitely better !
I guess I'm going to give two cc2531, one cc2530+cc2591 and one cc2538+cc2592 to some friends sooner or later :-)

Koenkk, 2019-10-03:
Yes the antenna on the CC2652R development board (the only thing you can buy for now) is better.

However the main reason to switch to the CC2652R is the improved performance and memory (allowing for more devices).

Note that this device is officially not supported by zigbee2mqtt yet, hopefully soon: https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-535437377

I will close this for now, lets keep CC2652R discussion here: #1429

closed 2019-10-03 by Koenkk