Koenkk/zigbee2mqtt #2068: Adding support for new Philips Hue White (Bluetooth)

jordy55, 2019-10-02:
## What happened
Was trying to add (Support) for new philips hue lamp.
I downloaded the following file and put it under share/zigbee2mqtt/.
Then I put the following in the configuration of the file share/zigbee2mqtt/devices.js :

```
{
        zigbeeModel: ['LWA001'],
        model: '???',
        vendor: 'Philips',
        description: 'Hue White bulb E27',
        extend: hue.light_onoff_brightness,
    },
```
But i don't know the model number. I get the following code in my database.js file:
`{"id":8,"type":"Router","ieeeAddr":"0x0017880106b3d53b","nwkAddr":33792,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWA001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190122","swBuildId":"1.53.3_r27175","zclVersion":2,"interviewCompleted":false,"meta":{},"_id":"eaLsZo39Obf3R5uF"}`

## What did you expect to happen
Was hoping it worked right away. But it is a new lightbulb.

## How to reproduce it (minimal and precise)
Resetting the bulb and connect it again.

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190608

quarkpw, 2019-10-02:
Have the same bulb, I added to existing, which makes it work, but maybe separate config is better.
    {
        zigbeeModel: ['LWB006', 'LWB014', 'LWA001'],
        model: '9290011370',
        vendor: 'Philips',
        description: 'Hue white A60 bulb E27',
        extend: hue.light_onoff_brightness,
    },

Koenkk, 2019-10-03:
Added support for this, will be supported in the dev branch in a few hours, thanks both!

closed 2019-10-03 by Koenkk

jvdb-git, 2019-10-06:
@Koenkk first of all: thanks for all the great work. 

Secondly, I'd like to add that the Philips Hue GU10 White ambiance has been updated as well. The new model is called 'LTG002':
`{
        zigbeeModel: ['LTW013','LTG002'],
        model: '8718696598283',
        vendor: 'Philips',
        description: 'Hue white ambiance GU10',
        extend: hue.light_onoff_brightness_colortemp,
    },`

Works like a charm.

jvdb-git, 2019-10-08:
> @jvdb-git <https://github.com/jvdb-git> does this model also supported
> bluetooth?

Yes, it does indeed.

jordy55, 2019-10-08:
Just to confirm, it works for me!
Had te re-install zigbee2mqtt but after that it connects perfect!
Yes, it's a bleutooth bulb so you can reset the bulb trough the app without any bridge.
After that it is going into connect mode

Koenkk, 2019-10-09:
@jvdb-git also added, will be supported in dev branch soon.