Koenkk/zigbee2mqtt #2070: Z2M High-Availability (semi-cluster)

jasimancas, 2019-10-03:
Hello,

I have a complete system running on Docker (HOST-A) (Home Assistant, MQTT and Z2M (latest-dev)), I have another backup machine with everything installed also on docker (HOST-B).

With rsync I synchronize the folders of the 3 containers every 10 minutes in the direction HOST-A -> HOST-B.

This morning, when I updated the operating system of HOST-A (ClearLinux) did not turn on, I turned on the HOST-B and started to give Z2M this error:

> 0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.6.0~prestart: zigbee2mqtt@1.6.0
6 info lifecycle zigbee2mqtt@1.6.0~start: zigbee2mqtt@1.6.0
7 verbose lifecycle zigbee2mqtt@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.6.0~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.6.0~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.6.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.6.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.6.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.6.0
15 verbose cwd /app
16 verbose Linux 4.15.0-65-generic
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v10.16.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.6.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

I have restarted the machine, and when I restart the container I see that it does not give an error, but nothing works, it appears:

> zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0xd0cf5efffe1a4e11' of group 'lampara_salon_2',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0xd0cf5efffe18a7b8' of group 'lampara_salon_2',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0x90fd9ffffed86c8b' of group 'lampara_salon_2',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0xd0cf5efffe1a4672' of group 'lampara_salon_1',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0x90fd9ffffeee1f55' of group 'lampara_salon_2',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0x90fd9ffffed89e34' of group 'lampara_salon_1',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0x90fd9ffffed86db1' of group 'lampara_salon_1',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0xd0cf5efffe1f4c63' of group 'lampara_salon_1',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0x90fd9ffffef807de' of group 'espejo_bano',
zigbee2mqtt:error 2019-10-03T08:46:07: Cannot find '0xd0cf5efffe05b1b4' of group 'espejo_bano'

I go to the directory and I see that the database file occupies 1k, it is actually empty, I stop the container, copy and paste the Z2M data directory from last night's backup, overwrite, start the container and everything working.

Any idea why those mistakes? Why is the database deleted?

LukeHandle, 2019-10-21:
Are you sure your `rysnc` is working correctly (eg. with a recursive flag)? If the file was only 1k, does that mean z2m deleted it on first run (unlikely!), or was not copied correctly by your automation?

If you hash/stat both files, do the timestamps line up/both recent?

closed 2019-10-26 by jasimancas

fliespl, 2023-01-29:
@jasimancas do you mind sharing your cluster solution?