Koenkk/zigbee2mqtt #2071: Read-only configuration file

juaningan, 2019-10-03:
Hi, I'm trying succesfully to build zigbee2mqtt for Nix ecosystem (Nixpkgs, NixOS...). Goes well but Im have a problem with configuration file.

To implement configuration options as Nix Modules the configuration file has to be read-only (stored in /nix/store and linked from ZIGBEE2MQTT_DATA dir). Problem: when app detects new device try to write in configuration.yaml and fails.

I could workaround with two files: configuration.yaml (read-only) and devices.yaml (read-write) with some trade-offs but also fails right now.

Can you put me on track of some solution? I can implement it and PR but don't know which way do you prefer.

juaningan, 2019-10-03:
by the way, Im not sure that write in an user configuration file will be a good practice. Why do you do? To ease users to modify devices?

Koenkk, 2019-10-03:
- This is done to add new devices to the configuration.yaml; having the users to do this manually would be a cumbersome process.
- You can specify devices in a separate file (https://www.zigbee2mqtt.io/configuration/configuration.html#specifying-devices-and-groups-in-a-separate-file). If that doesn't work; please check with the latest dev branch; if it still doesn't work; please report here with logs and instructions on how to reproduce.

juaningan, 2019-10-03:
Ok Ive check with dev branch and the situation improves:

* I got same error but now application cont:inues and write new device in devices.yaml:

```
(node:16362) UnhandledPromiseRejectionWarning: Error: EACCES: permission denied, open '/home/user/.zigbee2mqtt/configuration.yaml'
    at Object.openSync (fs.js:438:3)
    at Object.writeFileSync (fs.js:1189:35)
    at Object.write (/home/user/zigbee2mqtt/lib/util/yaml.js:29:8)
    at write (/home/user/zigbee2mqtt/lib/util/settings.js:256:10)
    at Object.addDevice (/home/user/zigbee2mqtt/lib/util/settings.js:421:5)
    at Controller.onZigbeeEvent (/home/user/zigbee2mqtt/lib/controller.js:174:40)
    at Zigbee.emit (events.js:189:13)
    at Controller.herdsman.on (/home/user/zigbee2mqtt/lib/zigbee.js:55:52)
    at Controller.emit (events.js:189:13)
    at Controller.<anonymous> (/home/user/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:454:22)
```

* steps to reproduce (using devices: devices.yaml):
```
$ export ZIGBEE2MQTT_DATA=~/.zigbee2mqtt/
$ chmod 0400 $ZIGBEE2MQTT_DATA/configuration.yaml
$ chmod 0600 $ZIGBEE2MQTT_DATA/devices.yaml
$ npm start
```

* Also, if both file are read-only app keeps running. So great news because if one Nix user decides to manage they devices with Nix app works.

Koenkk, 2019-10-03:
latest dev will now not write to `configuration.yaml` anymore when there are no changes; I think this was the cause of your issue. Can you confirm that it works OK now?

juaningan, 2019-10-03:
Nice! It's working well. The main concern is not solved yet but i guess I can close the issue and come back later. Thanks so much!

closed 2019-10-03 by juaningan

makefu, 2020-01-30:
Hey @juaningan would you mind sharing the expression you created for packaging zigbee2mqtt ?

juaningan, 2020-01-30:
Hi, i generate expressions with node2nix, and then overrides some stuff:

https://gist.github.com/juaningan/d96764c6ea4f3205c39b1c92d3ca8091

You have to import z2m.nix in your configuration.nix. Try it and share. Because I'm not sure if do some other workaround.

makefu, 2020-01-31:
@juaningan thanks for sharing! Without the override i got the error i reported https://github.com/svanderburg/node2nix/issues/171 . Now i can finally continue with the deployment,
Cheers!

JustinLex, 2023-02-26:
Updated link for the external configuration documentation since the old link died: 

https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files

This is a super useful feature for storing configuration.yaml in a ConfigMap on Kubernetes!

Example ConfigMap:
```
apiVersion: v1
kind: ConfigMap
metadata:
  name: configuration
data:
  configuration.yaml: |
    mqtt:
      base_topic: zigbee2mqtt
      server: mqtt://10.2.16.2
    serial:
      port: /dev/ttyACM0
    homeassistant: true
    permit_join: true
    devices: devices.yaml
    groups: groups.yaml
```

(Note that you should probably store your configuration in a Secret instead, if you're using MQTT authentication or [setting a network key](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#extract-config-to-separate-files).