Koenkk/zigbee2mqtt #2072: Did someone develop ZigBee device using NXP chips ?

Fabiancrg, 2019-10-03:
Hi,

This is not a bug or request feature; I am a Z2M user, currently working only with commercial components and I am currently trying to start some ZigBee development.
I was wondering why Z2M project is using only Ti CC253x or CC265x, it looks like it's very expensive to develop things for these chips as IAR Embedded Workbench is very expensive for hobbyist (I see it’s around $2000).

I know that IAR comes also with a free Kickstart license but limited to 4KB code, not sure if it is possible to do something with this limitation.
Did someone try the NXP JN516x chips? I see they are used by Xiaomi/Aqara and all support material, software and SDK are available for free.

If someone could share their experience it might help me decide which path to choose.

Thank you

Koenkk, 2019-10-03:
CC265X is developed using Code composer studio (free)

Fabiancrg, 2019-10-03:
@Koenkk Thank you for the information, I did not know there was a free dev tool for this board.
I've order the dev board and will give a try.

closed 2019-12-03 by Koenkk