Koenkk/zigbee2mqtt #2077: Ikea Remote: Error on toggle hold

sti0, 2019-10-04:
# Bug Report
## What happened
After pressing long the toggle button of the ikea remote an error appeared in zigbee2mqtt protocol:

  ```
zigbee2mqtt:info 10/4/2019, 12:17:46 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":57,"action":"toggle"}'
  zigbee2mqtt:info 10/4/2019, 12:17:46 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":57,"action":""}'
  zigbee2mqtt:info 10/4/2019, 12:17:49 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":55,"action":"toggle_hold"}'
  zigbee2mqtt:info 10/4/2019, 12:17:49 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":55,"action":""}'
PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
  manufCode: 4476,
  seqNum: 41,
  cmdId: 'tr... deepEqual { frameCntl:
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },
  manufCode: 4476,
  seqNum: 41,
  cmdId: 'tr...
    at cb (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:59:24)
    at /zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:31:17
    at Dissolve.<anonymous> (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/functional.js:76:17)
    at Object.onceWrapper (events.js:286:20)
    at Dissolve.emit (events.js:198:13)
    at Dissolve.<anonymous> (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/dist/deprecated/dissolve-chunks/index.js:65:29)
    at Dissolve.emit (events.js:198:13)
    at emitReadable_ (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:456:10)
    at emitReadable (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:450:7)
    at readableAddChunk (/zigbee2mqtt-1.6.0/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:206:11)
===== Buffer =====
{ type: 'Buffer', data: [ 5, 124, 17, 41, 7, 2, 0, 0, 0 ] }
===== Arguments =====
<Buffer 05 7c 11 29 07 02 00 00 00> 5
===== zcl-packet parse =====
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":41,"cmdId":"tradfriArrowSingle","payload":{"value":2}}
===== zcl =====
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":41,"cmdId":"tradfriArrowSingle","payload":{"value":2,"value2":0}}
===============
```

## What did you expect to happen

First, the error should not appear. Second, like the other buttons of the remote the first message should be a "toggle_hold" followed by a "toggle_release" after releasing it. The command "toggle" only should only be send after pressing it short.

For example pressing arrow right:
```
  zigbee2mqtt:info 10/4/2019, 12:23:09 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":84,"action":"arrow_right_click"}'
  zigbee2mqtt:info 10/4/2019, 12:23:09 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":84,"action":""}'
```

and holding and releasing it on a second try:
```
  zigbee2mqtt:info 10/4/2019, 12:22:54 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":"arrow_right_hold"}'
  zigbee2mqtt:info 10/4/2019, 12:22:54 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":""}'
  zigbee2mqtt:info 10/4/2019, 12:22:56 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":78,"action":"arrow_right_release","duration":2.055}'
  zigbee2mqtt:info 10/4/2019, 12:22:56 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":78,"action":""}'
```

## How to reproduce it (minimal and precise)

Long pressing toggle button.

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190425

Koenkk, 2019-10-04:
Duplicate of #2006

sti0, 2019-10-04:
@Koenkk 
this issue can't be ignored. Please read the "What did you expect to happen" section. In my opinion there are a false message handling. If I hold the toggle button it always sends a "toggle" message. Means an automation for the "toggle" button always triggers even I hold it! Furthermore there is no toggle_release message which means that I can't use this event at the moment because the error occur.

Koenkk, 2019-10-04:
Didn't read your second part, sorry.

Koenkk, 2019-10-05:
I agree that the current integration is wrong; did some debugging and this is what I found

- When pressing the button it immediately sends a Zigbee command -> translated to `toggle` 
- After 3 seconds it sends another Zigbee command -> currently ignored
- 0.5 seconds after this it sends another Zigbee command -> translated to `toggle_hold`
- When released the button doesn't send anything -> meaning we cannot provide a `release`.

The question is; what would be a useful integration? Where should the commands be translated to?

sti0, 2019-10-06:
Thanks for digging into this. Sounds like it is not possible to use it like described in the issue, right?

This is because every time it sends a "toggle" even if we hold the button....Only way would be to wait these 3 seconds if the button is hold but that would delay every "toggle" press. In my opinion we should leave it as it was. 

Maybe this behavior should be mentioned in the docs.

1. Toggle press sends immediately a toggle press
2. Toggle hold sends immediatley a toggle press followed by a toggle_hold after 3,5 seconds
3. Toggle release sends no message

Koenkk, 2019-10-07:
Done, assuming this can be closed now.

closed 2019-10-07 by Koenkk

SeLLeRoNe, 2021-04-06:
Sorry to jump on this old issue, but I am wondering, can't z2m send an "empty action" right after the "toggle" and "toggle_hold"?
This will prevent the toggle to be re-toggled if mesages are retained.

I think something like this was already put in place for some Aqara buttons.

That would help with automations and restarts (and retain), will not help with the "hold issue" but at least some other things will be less annoying :)

wifiuk, 2022-10-23:
deconz is able to do the toggle_hold without sending a toggle, might be worth looking how that is done.

razvancatinas, 2025-10-06:
As I understand from the comments above, there should be a "toggle_hold" action message if press for 3.5 seconds. This seems not to be working; it only sends "toggle" action no matter how log is pressed (the center button). Any advise? Thanks!