Koenkk/zigbee2mqtt #2078: Ikea Remote: Missing duration after holding brightness

sti0, 2019-10-04:
# Bug Report
## What happened
After holding and releasing the brightness buttons (up or down) I recieve a release message but the duration is missing.

 ```
 zigbee2mqtt:info 10/4/2019, 12:28:02 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":"brightness_down_hold"}'
  zigbee2mqtt:info 10/4/2019, 12:28:02 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":""}'
  zigbee2mqtt:info 10/4/2019, 12:28:03 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":"brightness_down_release"}'
  zigbee2mqtt:info 10/4/2019, 12:28:03 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":""}'
```

## What did you expect to happen

Getting the duration like holding and releasing the arrow buttons
```
zigbee2mqtt:info 10/4/2019, 12:22:54 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":"arrow_right_hold"}'
  zigbee2mqtt:info 10/4/2019, 12:22:54 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":76,"action":""}'
  zigbee2mqtt:info 10/4/2019, 12:22:56 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":78,"action":"arrow_right_release","duration":2.055}'
  zigbee2mqtt:info 10/4/2019, 12:22:56 PM MQTT publish: topic 'zigbee2mqtt/remote_buro', payload '{"battery":47,"linkquality":78,"action":""}'
```

## How to reproduce it (minimal and precise)

Hold & release the brightness buttons

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190425

Koenkk, 2019-10-05:
I've checked again but this remote does not provide a duration for brightness.

sti0, 2019-10-06:
Is it possible that we measure the time between hold and release and output the duration by ourselfs?

Koenkk, 2019-10-07:
Yes that possible however you will get into issues when there is a delay on the second message (because now you will measure the duration between 2 zigbee messages while actually you want to measure the duration between the presses).

sti0, 2019-10-08:
That's right. Under that conditions I think its better to have no duration as a false duration. Measurement between to messages could be implemented by the user if needed.

closed 2019-10-08 by sti0