Koenkk/zigbee2mqtt #2079: No device gets connected

nasi-be, 2019-10-04:
Hi,

I have the problem, that no device gets connected to zigbee2mqtt and the cc2531.

I first run zigbee2mqtt with the cc2531 (2018....-firmware) and could connect 2 hue bulbs.
But because of a problem with reachability I tried to put a cc2530 router between the bulbs and the cc2531.

This doesn't solve the reachability, so I tried a lot (firmware-update of cc2530, zigbee2mqtt-dev, firmware-update of the hue bulbs) but at the end my problem was a defective power supply of the cc2530 router.

After switching the power supply the cc2530 gets connected to the cc2531, but I couldn`t establish a connection to the bulb. (Now also the bulb hanging near to the router)

My last idea was to update the firmware of the cc2531.
Now no device gets connected but I didn't receive an error message. Removing the devices out of the configuration.yaml and the database doesn't change anything.

Here's my configuration:
`homeassistant: true`
`permit_join: true`
`mqtt:`
`  base_topic: zigbee2mqtt`
`  server: 'mqtt://xxx'`
`  user: xxx`
`  password: xxx`
`serial:`
`  port: /dev/ttyACM0`
`advanced:`
`  baudrate: 115200`
`  last_seen: 'ISO_8601'`
`  pan_id: 0x1a63`

and here the logfile:

info  2019-10-04T14:16:00: Logging to directory: '/app/data/log/2019-10-04.14-15-59'
info  2019-10-04T14:16:00: Starting zigbee2mqtt version 1.6.0 (commit #732a97b)
info  2019-10-04T14:16:00: Starting zigbee-herdsman...
info  2019-10-04T14:16:05: zigbee-herdsman started
info  2019-10-04T14:16:05: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
info  2019-10-04T14:16:05: Currently 0 devices are joined:
warn  2019-10-04T14:16:05: `permit_join` set to  `true` in configuration.yaml.
warn  2019-10-04T14:16:05: Allowing new devices to join.
warn  2019-10-04T14:16:05: Set `permit_join` to `false` once you joined all devices.
info  2019-10-04T14:16:05: Zigbee: allowing new devices to join.
info  2019-10-04T14:16:05: Connecting to MQTT server at mqtt://mqttbroker
info  2019-10-04T14:16:05: Connected to MQTT server
info  2019-10-04T14:16:05: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2019-10-04T14:16:05: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"732a97b","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
info  2019-10-04T14:16:10: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0018e1a5e9","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1570198570765}]'
info  2019-10-04T14:16:10: Starting network scan (includeRoutes 'false')
info  2019-10-04T14:16:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1a5e9","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1570198570773}]}'
info  2019-10-04T14:16:10: Network scan finished
info  2019-10-04T14:16:10: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00124b0018e1a5e9","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null}],"links":[]}'
info  2019-10-04T14:18:40: Switching log level to 'debug'
info  2019-10-04T14:18:40: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"732a97b","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
debug 2019-10-04T14:21:00: Saving state to file /app/data/state.json
debug 2019-10-04T14:21:10: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''
debug 2019-10-04T14:21:10: Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'raw'
info  2019-10-04T14:21:10: Starting network scan (includeRoutes 'false')
info  2019-10-04T14:21:10: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0018e1a5e9","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1570198870765}]'
debug 2019-10-04T14:21:10: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"ieeeAddr":"0x00124b0018e1a5e9","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1570198870765}]'
info  2019-10-04T14:21:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1a5e9","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1570198870767}]}'
debug 2019-10-04T14:21:10: LQI succeeded for 'Coordinator'
info  2019-10-04T14:21:10: Network scan finished
info  2019-10-04T14:21:10: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/raw', payload '{"nodes":[{"ieeeAddr":"0x00124b0018e1a5e9","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":null}],"links":[]}'
debug 2019-10-04T14:22:47: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'Info'
info  2019-10-04T14:22:47: Switching log level to 'info'

I'm at a loss.
Does anyone have a hint, how to resolve the problems?

Best Regards
Thomas

Koenkk, 2019-10-05:
Do you have any other zigbee devices turned on? It helps to power off all zigbee devices before trying to pair.

nasi-be, 2019-10-06:
I only have two hue bulbs which I try to connect and 1 battery powered xiaomi aquara motion sensor, which is unfortunately out of range. This is the reason I try to insert the cc2530 as router.
(zigbee2mqtt and the cc2531 is installed in a server rack in the basement and the motion sensor is installed at the stairway to the first floor.)
When I try to pair the router or one hue bulb only this device is powered on. The other hue bulb is powered off and the the motion sensor is out of range. (It's not that easy to remove the battery, because the sensor is difficult to access)

closed 2019-12-12 by stale[bot]