Koenkk/zigbee2mqtt #2081: Securifi Peanut energy reports all zero

thardie, 2019-10-04:
Installed zigbee2mqtt from github yesterday (2019-10-03) and downloaded cc253X firmware and flashed at the same time. Trying to get a Securifi Peanut to report energy usage. I can turn it on an off through MQTT. When I turn it on and off on the plug itself, updates come through fine. The problem is, all energy data is zero. Voltage, Current, Watts, etc.

Here's a full debug log of starting up, turning the device off then on from the plug itself.

I followed the instructions for manually adding the PP-WHT-US to the database file to get it this far. Suggestions?

https://pastebin.com/9JJj1EYk

Koenkk, 2019-10-04:
It seems that the plugs only report power after a firmware upgrade of the device itself (https://github.com/Koenkk/zigbee-shepherd-converters/pull/513). Closing this as we cannot fix this from the zigbee2mqtt side.

closed 2019-10-04 by Koenkk

thardie, 2019-10-04:
Oh, wow. I really thought I'd searched for issues like this. Sorry for wasting your time, and thanks!

Koenkk, 2019-10-04:
no problem, added it to the docs now!