Koenkk/zigbee2mqtt #2082: Support for Ikea TRADFRI bulb E27 WW clear 250lm

sb1089, 2019-10-04:
I've got a new Ikea Tradfri bulb, but it's not yet supported.
Can you add support for it?

https://www.ikea.com/nl/nl/p/tradfri-led-lamp-e27-250-lumen-draadloos-dimbaar-warm-schijnsel-globe-bruin-helder-glas-70455676/

zigbee2mqtt:info 10/4/2019, 6:13:15 PM New device 'TRADFRI bulb E27 WW clear 250lm' with address 0x14b457fffe4fa9fe connected!
  zigbee2mqtt:info 10/4/2019, 6:13:15 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x14b457fffe4fa9fe","meta":{"modelID":"TRADFRI bulb E27 WW clear 250lm"}}'
  zigbee2mqtt:warn 10/4/2019, 6:13:15 PM Device with modelID 'TRADFRI bulb E27 WW clear 250lm' is not supported.

If I have to provide more information, let me know! :)

Koenkk, 2019-10-04:
This device is already supported in the latest dev branch.

closed 2019-10-04 by Koenkk

sb1089, 2019-10-04:
Oh really? That's great! 👍

magiva, 2019-11-08:
awesome, when will the dev branch be released for hassio ?

Koenkk, 2019-11-08:
@magiva it is already (labeled as edge)

magiva, 2019-11-09:
Sorry, i meant when will it be released from dev into an official release

pipetboy, 2020-02-11:
Is there any news here? I just bought one of these, seeing that it was supported in zigbee2MQTT. Now it looks as if I need to switch to the EDGE version?
(Or can I install both Edge AND keep running the GREAT normal Hassio version?)

BTW... Thanks for your work!

magiva, 2020-02-11:
theyre available in the live version now, no need for the edge
but...it needs a delay setting as when you turn off a few lights at once,
sometimes 1 or 2 is left on

pipetboy, 2020-02-14:
Sorry... My bad. I missed a few updates. It's working great now! Thank you so much for your fast response.