Koenkk/zigbee2mqtt #2083: Iris contact sensor reporting as false

mar005, 2019-10-04:
I am having this issue right here where I just started with version 0.1.6 and the only contact sensor I have paired is constantly showing false on contact.

_Originally posted by @mar005 in https://github.com/Koenkk/zigbee2mqtt/issues/1710#issuecomment-538577999_

mar005, 2019-10-04:
`- info: MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000e261001/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/0x000d6f000e261001","json_attributes_topic":"zigbee2mqtt/0x000d6f000e261001","name":"0x000d6f000e261001_contact","unique_id":"0x000d6f000e261001_contact_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000e261001"],"name":"0x000d6f000e261001","sw_version":"Zigbee2mqtt 1.6.0","model":"Contact sensor (3320-L)","manufacturer":"Iris"},"availability_topic":"zigbee2mqtt/bridge/state"}'
10/4/2019, 5:14:25 PM - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f000e261001/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f000e261001","json_attributes_topic":"zigbee2mqtt/0x000d6f000e261001","name":"0x000d6f000e261001_linkquality","unique_id":"0x000d6f000e261001_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000e261001"],"name":"0x000d6f000e261001","sw_version":"Zigbee2mqtt 1.6.0","model":"Contact sensor (3320-L)","manufacturer":"Iris"},"availability_topic":"zigbee2mqtt/bridge/state"}'
10/4/2019, 5:14:30 PM - warn: Failed to configure 0x000d6f000e261001 (0x000d6f000e261001) ('Error: AF data request fails, status code: 240. MAC transaction expired.') (attempt #1)
10/4/2019, 5:14:30 PM - warn: This can be ignored if the device is working properly
10/4/2019, 5:14:34 PM - warn: Failed to configure 0x000d6f000e261001 (0x000d6f000e261001) ('Error: request unsuccess:112') (attempt #2)
10/4/2019, 5:14:34 PM - warn: This can be ignored if the device is working properly
10/4/2019, 5:14:46 PM - warn: Failed to configure 0x000d6f000e261001 (0x000d6f000e261001) ('Error: request unsuccess:112') (attempt #3)
10/4/2019, 5:14:46 PM - warn: This can be ignored if the device is working properly
10/4/2019, 5:14:48 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":65}'
10/4/2019, 5:14:48 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":65}'
10/4/2019, 5:14:48 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":57}'
10/4/2019, 5:14:48 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:50 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":55}'
10/4/2019, 5:14:50 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:51 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:51 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:52 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:53 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:53 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:55 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:55 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:56 PM - info: Successfully reenabled joining
10/4/2019, 5:14:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:56 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:14:58 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":60}'
10/4/2019, 5:110/4/2019, 5:16:29 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":57}'
10/4/2019, 5:16:29 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":49}'
10/4/2019, 5:16:30 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":57}'
10/4/2019, 5:16:33 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":47}'
10/4/2019, 5:16:34 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000e261001', payload '{"contact":false,"linkquality":44}'
`

mar005, 2019-10-04:
I just paired another iris contact sensor and same issue it only reports false

Koenkk, 2019-10-05:
Can you provide the debug logging when closing and opening the contact (please indicate what log lines are caused open/close).

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

closed 2019-12-11 by stale[bot]