Koenkk/zigbee2mqtt #2085: CC2531 - Standby with no messages for few seconds

marcutzo84, 2019-10-05:
Hi all,

I'm getting a strange problem with my zigbee environment. Here my setup

Ubuntu server with HassIO in docker container
Zigbee2Mqtt 1.6.0 from addon in Hassio
CC2531 dongle with CC2531_DEFAULT_20190608.zip (coordinator)
I have 17 devices, all Xiaomi, some temperature sensors, two water leak sensors, two door sensors and two switch button (the round one Mijia).

Here the issue:
when I push the switch button it seems to work correctly. I see in home assistant the status change and the automation starts. I try agai with some click and/or double click and after 3 or 4 times zigbee2mqtt stop to show messages on the log. After few seconds (10s) again with click and double click on the button and the log show again status changes, messages back to work again. after few second again, and again....
I got same behavior with the door sensor: sometimes I don't get the status changes on zigbee2mqtt. Open and close the door few times and the messages come back to work. after a while again stop getting messages.

Hope someone could help on this, I tried to move the sensors near the dongle to get a strong signal but the issue still continue.

Thanks all

Koenkk, 2019-10-07:
Is your CC2531 connected via an USB extension cable? If not I would highly recommend this to avoid interference (as this looks like an interference issue)

marcutzo84, 2019-10-08:
Thanks Koen, I will have a try next days to see what happen. Honestly the current place is near a lot of radio devices (hue bridge, wifi accesspoint, bluetooth integrated card). I'm using a notebook to host my home server with the CC2531 stick connect on it, near the ap and other stuff reported above.

I'll keep you posted

marcutzo84, 2019-10-12:
I found the problem. Is related to another USB device connected to the same server. A USB storage with a Y cable was the root couse of my problem.

Thanks for the help.

closed 2019-10-12 by marcutzo84