Koenkk/zigbee2mqtt #2086: Can't pair anymore !! Nothing happens

D3ltorohd, 2019-10-05:
Hello,
since some days i have a problem to pair devices with my cc2531 stick.
It runs on a NUC, on there Debian and my end sw is FHEM. 

I have more than 15 devices, so i bought a inrr so120 to increase the amount of devices and to improve the range. So i delet one aqara sensor to have only 14 devices. So i tried to pair the innr plug with the stick, but it will not work. I cant see anything in the log, there is nothing happen and i have no new device shown in fhem (autocreate is on)

So i tried to repair my deleted sensor again, because with this it works in the past. So now i cant get it paired. I have one other sensor they was never paired, i tried to pair it without any success.

In the past all works fine and i can pair all 15 devices. But now, nothing works. I have restart fhem, restart z2m service, i reboot my system, i plug the stick off and later plug in again.

So whats happend now. I updated z2m to 1.6 the fw from the stick is from april 2019.

> {"version":"1.6.0","commit":"e26ad2a","coordinator":20190425,"log_level":"info","permit_join":true}

So any idea what's happend ? This one paired works great, i see it directly when i open or close a window. So i think the stick should work.

The only thing i was changed since i paired my other sensors, is to add an antenna to the stick.

Greet's

Koenkk, 2019-10-05:
Your stick is probably still on the limit of 15 devices (removal of battery powered devices mostly fails because they are sleeping). 

I would recommend to reflash the stick, start zigbee2mqtt and immediately after that pair the Innr SP 120; the other devices should re-join automatically after that.

eddso, 2019-10-06:
I have same problem. I read a lot in forums but nothing is clear enough to understand. Why we need reflash the Stick. Why already paired devices will join again, after reflash Stick, without pairing? Last but not least how to remove a paired device to get e.g. 14 devices again? Please someone, redirect me to an answer somewhere.

Koenkk, 2019-10-07:
- You need to reflash the stick to clear the device tables to let other devices join
- Because the network key doesn't change
- Depends; sometimes it is not possible to correctly remove a device (because they don't respond to leave request, e.g. Xiaomi battery powered devices). Then the only solution is to reflash.

D3ltorohd, 2019-10-11:
Ok thanks koenkk,
for now i have upgrade my FW to the latest 1.2 home default. After this i can directly pair the innr and 2 new sensors. So far so good, it works now again. But all the other sensors how was paired, did not react anymore ? What now can i do something, or musst i delet all from the config.yaml and repair them again ?

Koenkk, 2019-10-12:
No need to remove, just try to re-pair them.

D3ltorohd, 2019-10-12:
No chance, nothing happend. Try to re-pair one sensor. Nothing happens.

hwijers, 2019-12-13:
I now also have this problem with a CC2652 (fw: 20191106) and version 1.7.1+dev d05a3a0. After some time, a Hue motion detector or dimmer will fail. Pairing again does not help. I have to forcefully remove the device with **`mosquitto_pub -h x.x.x.x -t zigbee2mqtt/bridge/config/force_remove -r -m 'xxxxxxxxxxxxxxxx'`** and then the pairing and the device will work again.

Koenkk, 2019-12-14:
@hwijers I also have it with a Hue motion outdoor sensor, it works fine and then stops working after a few days. Probably because it switches parents zigbee2mqtt cannot reach it anymore, after some time the device disconnects. Currently investigating this but this takes long as it only happens every few days.

closed 2020-02-19 by stale[bot]