Koenkk/zigbee2mqtt #2088: I need you to install Zigbee2MQTT

Dim3333, 2019-10-06:
Hello,

I have a lot of problems to install Zigbee2MQTT on my NUC (Ubuntu Server) :

1/ I flashed my CC2531 with the last "router-cc253x-std.hex"(here : https://github.com/Koenkk/Z-Stack-firmw ... ter/router : OK

2/ When I plug my CC2531 on my NUC. The command (ls -l /dev/serial/by-id) show me only my RFXCOM, I don't see CC2531. :shock:

3/ I follow this procedure (https://github.com/Koenkk/zigbee2mqtt.i ... ee2mqtt.md) but :

3.1 I don't undestand the command "sudo chown -R pi:pi /opt/zigbee2mqtt", Could you explain me ? Replace pi:pi by what ?

3.2 I have an error during install npm : Error: Can't find Python executable "python", you can set the PYTHON env variable. What is the problem ?

Thanks for your support.

Kryzek, 2019-10-07:
2 ) You could try ```dmesg``` on command line. If you stick is functional you should see something like
```
usb 1-1.3: new full-speed USB device number 8 using dwc_otg
usb 1-1.3: New USB device found, idVendor=0451, idProduct=16a8
usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.3: Product: TI CC2531 USB CDC
usb 1-1.3: Manufacturer: Texas Instruments
usb 1-1.3: SerialNumber: __0X0012481014DDDDF4
cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```
in the log after plugging in the usb-stick. If you don't see something similar to these lines your stick/flash might not be functional.

If you see these lines, then you can use ttyACM0 (this might differ on your environment) on configuration.yaml file as follows:
```
serial:
  port: /dev/ttyACM0
```
3.1 ) If you are logging in to the system with lets say with username ```dim``` your command for chown would be ```sudo chown -R dim:dim /opt/zigbee2mqtt```.

3.2 ) Just a wild guess, maybe you don't have python environment installed... to install try
```
sudo apt-get update
sudo apt-get install python
```

Dim3333, 2019-10-07:
Hello Kryzek,

Thanks for your help.

2) After `dmseg` on command line, I don't see CC2531 or something similar. Could you send a link to download the good firmware ?
After flashing my CC2531 stick, the red LED is become blinking. Normal ?

3.1) OK. Done. I understood.

3.2) OK. Done. NPM is installed.

Thanks again.

Dim3333, 2019-10-08:
I solved the problem with the command     `sudo chown [USER] [PORT]` .
But one problem again .... 
`
npm ERR! path /home/dd/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/home/dd/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent
`

Dim3333, 2019-10-08:
I executed update.sh in /opt/zigbee2mqtt and the problem is solved.
But now the last problem (i'm sure ....) : 
After `sudo systemctl start zigbee2mqtt` and `systemctl status zigbee2mqtt.service`, I get

`● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-10-08 23:22:57 CEST; 28s ago
  Process: 1483 ExecStart=/usr/bin/npm start (code=exited, status=217/USER)
 Main PID: 1483 (code=exited, status=217/USER)

oct. 08 23:22:57 ubuntu systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling res
oct. 08 23:22:57 ubuntu systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is
oct. 08 23:22:57 ubuntu systemd[1]: Stopped zigbee2mqtt.
oct. 08 23:22:57 ubuntu systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
oct. 08 23:22:57 ubuntu systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
oct. 08 23:22:57 ubuntu systemd[1]: Failed to start zigbee2mqtt.
`

matchett808-gh, 2019-10-11:
Can you successfully run zb2mq via `npm start` (from within the install diretory?

Dim3333, 2019-10-11:
Yes but always the same problem : 

`dd@ubuntu:~$ npm start
npm ERR! path /home/dd/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/home/dd/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent
`

matchett808-gh, 2019-10-11:
You're not in the right folder, cd into the right folder, run `npm i`, then run `npm start`

Dim3333, 2019-10-14:
Same problem :
`npm i`
`npm WARN saveError ENOENT: no such file or directory, open '/usr/bin/package.jso                                                       n'
npm WARN saveError EACCES: permission denied, open '/usr/bin/package-lock.json.3                                                       063024340'
npm WARN enoent ENOENT: no such file or directory, open '/usr/bin/package.json'
npm WARN bin No description
npm WARN bin No repository field.
npm WARN bin No README data
npm WARN bin No license field.

up to date in 1.115s
`

'npm start`
`npm ERR! path /usr/bin/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/usr/bin/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/dd/.npm/_logs/2019-10-14T20_33_43_746Z-debug.log
`

Dim3333, 2019-10-14:
I reboot and command 'sudo npm start' and it's OK.

closed 2019-12-20 by stale[bot]