Koenkk/zigbee2mqtt #2089: Hiding individual devices in a group from HA integration?

jwilling, 2019-10-06:
Hi. First of all, thanks for all of your effort on this project. It's been a lot of fun to play with and works quite well!

I'm experimenting with the groups feature right now. I've created a group for three bulbs, and added the group to HA as the documentation describes through a light platform. This works well — I can control the group and each individual light through HA. The light entities themselves show up automatically due to auto discovery.

The thing is, I'd like to only see the group in HA, not each individual bulb. I know there's ways to hide entities in HA, but I'd prefer it to be done on the Z2M side. I don't see a way to do this with the current configuration options. I only see an option to blacklist entities, which isn't what I want. Am I missing something that exists? If not, could this be a feature request?

BudBundi, 2019-10-06:
So you want the configuration `homeassistant: false` on device level, I think it isn't possible at the moment

jwilling, 2019-10-07:
Yes that'd be one way to do it. And thanks for confirming!

Koenkk, 2019-10-07:
Implemented, see https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/configuration/device_specific_configuration.md for docs. Thanks @jwilling and @BudBundi

closed 2019-10-07 by Koenkk

jwilling, 2019-10-07:
You're actually a legend. Thank you Koen!

K-RAD, 2023-02-05:
> Implemented, see https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/configuration/device_specific_configuration.md for docs. Thanks @jwilling and @BudBundi

I can't seem to find the option to hide the device, and that documentation link is not available anymore

BudBundi, 2023-02-05:
> > Implemented, see https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/configuration/device_specific_configuration.md for docs. Thanks @jwilling and @BudBundi
> 
> I can't seem to find the option to hide the device, and that documentation link is not available anymore

You can find it here: [https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html)

under the device, add 'homeassistant: null' to disable the discovery for this device

NPutting, 2023-07-31:
Bit stale but is there a way to do this from the GUI now?

lanwin, 2024-07-07:
Any plans to bring this to ui? Would be very helpful.

verdel, 2024-07-09:
@Koenkk, do I understand correctly that the `homeassistant: null` option can only be used with devices? This option cannot be used with groups to hide them from HomeAssistant?

bzumik1, 2024-07-13:
> Any plans to bring this to ui? Would be very helpful.

I would also appreciate to have this option available in UA. Maybe there could be also button in group view which would hide all devices in the group.