Koenkk/zigbee2mqtt #2091: Update Error latest dev

zibous, 2019-10-07:
When I try to update to the last dev version I get the following error messages and can no longer use zigbee2Mqtt:

```
Stopping zigbee2mqtt...
Creating backup of configuration...
Updating...
Already up-to-date.
Installing dependencies...

> @serialport/bindings@8.0.1 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.14.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.14.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'

> serialport@8.0.1 postinstall /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> core-js-pure@3.1.4 postinstall /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/core-js-pure
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 2091 packages from 890 contributors and audited 877635 packages in 724.23s
found 2 vulnerabilities (1 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
Restore configuration...
Starting zigbee2mqtt...
Done!
```

Error after start
```
Oct 07 14:41:32 rb3 node[19003]: Error: Cannot find module 'zigbee-herdsman/dist/deprecated/ziee'
Oct 07 14:41:32 rb3 node[19003]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
Oct 07 14:41:32 rb3 node[19003]:     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
Oct 07 14:41:32 rb3 node[19003]:     at Module.require (internal/modules/cjs/loader.js:637:17)
Oct 07 14:41:32 rb3 node[19003]:     at require (internal/modules/cjs/helpers.js:22:18)
Oct 07 14:41:32 rb3 node[19003]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
Oct 07 14:41:32 rb3 node[19003]:     at Module._compile (internal/modules/cjs/loader.js:689:30)
Oct 07 14:41:32 rb3 node[19003]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
Oct 07 14:41:32 rb3 node[19003]:     at Module.load (internal/modules/cjs/loader.js:599:32)
Oct 07 14:41:32 rb3 node[19003]:     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
Oct 07 14:41:32 rb3 node[19003]:     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
Oct 07 14:41:32 rb3 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 07 14:41:32 rb3 systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Oct 07 14:41:32 rb3 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 07 14:41:32 rb3 systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Oct 07 14:41:32 rb3 systemd[1]: Stopped zigbee2mqtt.
Oct 07 14:41:32 rb3 systemd[1]: Started zigbee2mqtt.
```

npm audit fix does not help either
Is there a solution for this

Koenkk, 2019-10-07:
What is the output of:

```
git status
git branch
```

zibous, 2019-10-07:
@Koenkk 

```
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   data/configuration.yaml

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	data/archiv/
	data/data/

no changes added to commit (use "git add" and/or "git commit -a")

git branch
* master
```

Update script
```
#!/bin/bash -e

echo "Stopping zigbee2mqtt..."
sudo systemctl stop zigbee2mqtt

echo "Creating backup of configuration..."
cp -R data data-backup

echo "Updating..."
git checkout HEAD -- npm-shrinkwrap.json
git pull

echo "Installing dependencies..."
rm -rf node_modules
npm install

echo "Restore configuration..."
cp -R data-backup/* data
rm -rf data-backup

echo "Starting zigbee2mqtt..."
sudo systemctl start zigbee2mqtt

echo "Done!"
```

zibous, 2019-10-08:
@Koenkk 

Sorry now it works.

```
git clone --branch dev https://github.com/Koenkk/zigbee2mqtt.git  /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm install

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 1344 packages from 849 contributors and audited 877113 packages in 863.573s
found 0 vulnerabilities

node index.js

... running

```

closed 2019-10-08 by zibous