Koenkk/zigbee2mqtt #2094: Missing climate entity for Eurotronic SPZB0001

sti0, 2019-10-07:
# Bug Report
## What happened
Added a new Eurotronic SPZB0001 to zigbee2mqtt.

```
10/7/2019, 9:37:04 PM - info: New device 'SPZB0001' with address 0x00158d000192d0a1 connected!
10/7/2019, 9:37:04 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d000192d0a1/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d000192d0a1","json_attributes_topic":"zigbee2mqtt/0x00158d000192d0a1","name":"0x00158d000192d0a1_linkquality","unique_id":"0x00158d000192d0a1_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000192d0a1"],"name":"0x00158d000192d0a1","sw_version":"Zigbee2mqtt 1.6.0","model":"Spirit Zigbee wireless heater thermostat (SPZB0001)","manufacturer":"Eurotronic"},"availability_topic":"zigbee2mqtt/0x00158d000192d0a1/availability"}'
  zigbee2mqtt:info 10/7/2019, 9:39:42 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000192d0a1', payload '{"current_heating_setpoint":21,"eurotronic_system_mode":1,"local_temperature":20.87,"occupied_heating_setpoint":21,"unoccupied_heating_setpoint":16}'
  zigbee2mqtt:info 10/7/2019, 9:39:49 PM Successfully configured 0x00158d000192d0a1 (0x00158d000192d0a1)
```

Only a linkquality sensor is exposed to Home Assistant.

## What did you expect to happen
Exposing a climate component.

## How to reproduce it (minimal and precise)

Join a Eurotronic SPZB0001 to z2m

## Debug Info
zigbee2mqtt version: 1.6.0
CC253X firmware version: 20190425

sti0, 2019-10-07:
Battery information is also missing. The ioBroker implementation got the battery information for this device. That means it must be send (https://forum.iobroker.net/topic/20754/gel%C3%B6st-eurotronic-heizk%C3%B6rperthermostat-spirit-zigbee-einstellungen)

Koenkk, 2019-10-07:
fixed in latest dev branch, can you confirm?

sti0, 2019-10-07:
That was fast. Thanks.

Edge version on hassio starts up with

`zigbee2mqtt:info  2019-10-07T20:30:15: Starting zigbee2mqtt version 1.6.0 (commit #3e75108741a0d7328b278a8c26daaf8a5adf0934)`

Changes are not included in this commit.
I think it needs some time to get the newest commit. Is it right? Maybe I give it another try tomorrow.

Koenkk, 2019-10-08:
Should be up now.

sti0, 2019-10-08:
Still got the old commit. Even after a new installation of the zigbee2mqtt_edge addon.

`zigbee2mqtt:info  2019-10-08T16:55:42: Starting zigbee2mqtt version 1.6.0 (commit #3e75108741a0d7328b278a8c26daaf8a5adf0934)`

sawfear, 2019-10-08:
I´m getting the newest commit on the edge addon and it seems now to be reading more stats (battery and temperature aswell as linkquality) But unfortunately the temperature can not be set automatically with Home Assistant Automation. Do you have any idea why? I tried with this config:

- id: '1570554579605'
  alias: Test
  trigger:
  - at: '19:25'
    platform: time
  action:
  - data:
      entitiy_id: climate.WC
      temperature: 22
    service: climate.set_temperature

The Home Assistant log shows this entry when firing the automation:

2019-10-08 20:35:33 ERROR (MainThread) [homeassistant.components.automation] Error while executing automation automation.test. Invalid data for call_service at pos 1: extra keys not allowed @ data['entitiy_id']

Your help is much appreciated!
Thanks

stefanroelofs, 2019-10-08:
>     entitiy_id: climate.WC

Typo. Should be: entity_id

sti0, 2019-10-08:
@sawfear quick offtopic question: I bought two of the thermostats. Both are very noisy on open/close. I think not useful in sleeping rooms. Could you confirm this?

stefanroelofs, 2019-10-08:
I bought 1 this week. They do make noise, but I don't think it is too bad.
I think that when someone is sleeping they won't have to actuate much, so its no problem for me.

sawfear, 2019-10-08:
> 
> 
> @sawfear quick offtopic question: I bought two of the thermostats. Both are very noisy on open/close. I think not useful in sleeping rooms. Could you confirm this?

Mine isn´t too noisy also.. I guess it is a little bit louder than other thermostates, but not that you will wake up, when it´s adjusting itself.. Unless you have a really light sleep.

sti0, 2019-10-08:
I want to install it at my daughters room. It should heat a half hour before wake up. With that noise she will wake up immediately... I contacted the manufacture maybe a product failure. Right now I got a AVM Fritz 301 DECT on this radiator and this one is really quiet. (but could not be controlled instantly combined with window sensors).

sti0, 2019-10-08:
Ok my fault. As learned by #63 a change in zigbee-shepherd/herdsman shows no new commit on zigbee2mqtt... (sounds logical). 

Climate entity + battery level is now pushed to HA. 

Think we're done here :) Thanks.

closed 2019-10-09 by Koenkk

sti0, 2019-10-09:
@Koenkk 
Is it possible to change HA state according to the zigbee2mqtt/[device]/set payload {"eurotronic_system_mode": XX} as mentioned in https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html ? Actually the state is always unknown.  
Changing the actions at the climate entity have no impact.
[image]

Its possible to set the device to off by setting {"eurotronic_system_mode": 32}. Is it possible to set this through the HA frontend (Action = off)? Even if the device is in normal heat or boost mode the state should be "heat".

Koenkk, 2019-10-10:
@sti0 dor this you should adapt the toZigbee converters to resond to the correct attribute.

Koenkk, 2019-10-10:
In toZigbee.js

sti0, 2019-10-10:
Ok thanks. I will look into it and maybe come back to you if I have further questions :)